realsenseai/librealsense #2684: About multi cameras HW sync

kunyao, 2018-11-06:
I am not sure if HW sync can be done with both depth and RGB. I read it so in the white book, however, when I implement with both pipeline and frame_queue, I find HW sync only takes place in depth sensors, RGB sensor still works in the original manners(fixed timestamp interval, no drift caused by the master trigger).

Are there any camera options need to be turned on for this? 

BTW I use D415.

dorodnic, 2018-11-07:
Hi @kunyao 
Please check Auto-Exposure Priority control under RGB > Controls. Should be disabled.
For the D415, Depth to RGB sync only kicks in when the two are at same FPS. Disabling Auto-exposure priority ensures auto-exposure will not reduce FPS. Also, what firmware version are you using?

kunyao, 2018-11-08:
Thanks, @dorodnic.
I tried, but setting RS2_OPTION_ENABLE_AUTO_EXPOSURE to 0 doesn't help with RGB sync. Still, no drift is seen, and also it introduces new issue by causing frame level delay(~33ms for 30fps)). Sometime it can catch up after serveral frames, but most of the time it won't, just accumulate the 33ms difference (which make it very hard to notice the drift).
This case doesn't happen when I turn on the auto exposure. Is there any trick to set exposure time? I try a lot, but no satifying answers(BTW what's the unit of RGB exposure time, it's not compatible with the Depth right?)
I am using firmware: 05.10.06.00

dorodnic, 2018-11-09:
Sorry, I should have been more specific, this is the option - `RS2_OPTION_AUTO_EXPOSURE_PRIORITY`.
Something like:
```cpp
rs2::pipeline p;
auto prof = p.start();
auto sensor = prof.get_device().query_sensors()[1];
sensor.set_option(RS2_OPTION_AUTO_EXPOSURE_PRIORITY,  0.f);
```

kunyao, 2018-11-09:
Interesting.. I successfully detect HW RGB sync(stable drift) by unpluging and repluging the devices!!! And it only happens in this case (I have to do repluging each time, hardware reset doesn't work). @dorodnic Thanks, I never know there is a **RS2_OPTION_AUTO_EXPOSURE_PRIORITY** option. That makes more sense.  

BTW It's my fault of those frame level lag as I write two get_frame together.

RealSense-Customer-Engineering, 2018-12-07:
[Realsense Customer Engineering Team Comment]
No response in a month, submit issue as solved.

closed 2018-12-07 by RealSense-Customer-Engineering