realsenseai/librealsense #2686: Matlab: Access to advance mode

kklaw85, 2018-11-07:
Hi,
Does anyone knows how to access to Advance mode APIs from Matlab wrapper? From what I see, the existing Matlab wrapper does not contains any Advance mode commands.

RealSense-Customer-Engineering, 2019-01-08:
[Realsense Customer Engineering Team Comment]
Hi kklaw85,

For Matlab wrapper, advanced mode is not supported yet.
You may check back later and subscribe to RS for updates.

Thank you!

lramati, 2020-03-04:
Partial support is being added by #5970

RealSenseCustomerSupport, 2020-03-12:
----------------------------------------------

@kklaw85 Could you please check if PR#5970 https://github.com/IntelRealSense/librealsense/pull/5970 can meet your need? Anything else needed please let us know. Thanks!

GenusJqj, 2020-03-12:
Thank you！ I will have a try as soon as I return back to school, for I'm in home without a camera now. If there is any problem  more i will put it on Github and contact you. Best wishes.

| |
jiqinjie_njaurobot
|
|
jiqinjie_njaurobot@163.com
|
签名由网易邮箱大师定制

RealSenseCustomerSupport, 2020-03-26:
----------------------------------------------

@GenusJqj Thanks for the reply! Maybe you can check the code snippet in PR https://github.com/IntelRealSense/librealsense/pull/5970 before you get the camera. Any other questions please let us know. Thanks!

RealSenseCustomerSupport, 2020-04-08:
----------------------------------------------

@GenusJqj Did you get chance to check and any update? Thanks!

GenusJqj, 2020-04-08:
Thank you for  your concerns, I have read advanced_mode.m . However, I can't return back to school until a mouth later, due to COVID-2019 .
Regards.

| |
jiqinjie_njaurobot
|
|
jiqinjie_njaurobot@163.com
|
签名由网易邮箱大师定制

RealSenseCustomerSupport, 2020-04-22:
----------------------------------------------

@GenusJqj Understood.Take care. Can we close this ticket now? You can create another new one later if you find current implementation can't meet your requirement. Thanks!

GenusJqj, 2020-04-22:
Yes，thank you! Take care.
Regards

| |
jiqinjie_njaurobot
|
|
jiqinjie_njaurobot@163.com
|
签名由网易邮箱大师定制

closed 2020-04-23 by RealSenseCustomerSupport