realsenseai/librealsense #2688: D415 depth image origin

noamzafrir, 2018-11-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | (05.10.06.00) | 
| Operating System & Version |   Win (8.1/10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.16.1 }                          | 
| Language                            |  C                          | 
| Segment			|  others}                          | 

### Issue Description
Hello,

What is the origin (x=0, y=0, z=0) of the depth image in D415?
In the SR300 the origin is the depth lens, but the SR300 has only one depth lens.
The D415 has two depth lenses. Is the origin is the exact middle of the two depth lenses?

Thank you.

MartyG-RealSense, 2018-11-07:
The 400 Series has a pair of IR imagers that are used to create the depth image.

The baseline is between the left and right imagers.  However, RealSense stereo cameras tend to rely on the left imager to generate the image.   The dynamic calibration guide for the 400 Series says: "Assume left camera is the reference camera and is located at world origin."

noamzafrir, 2018-11-07:
Thank you, Marty.
That is very helpful.
You can close this issue.

closed 2018-11-07 by RealSense-Customer-Engineering