realsenseai/librealsense #2693: segmentation in reading bag files

ywjia, 2018-11-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | 5.10.3.0 | 
| Operating System & Version |   Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.15                                         | 
| Platform                                 | PC|
| SDK Version                            |  2.16.0 | 
| Language                            |  C++| 
| Segment			|  VR | 

I have a sequence of bag files captured from realsenseviewer, named as test_####.bag. Each of the bag files contains ~1s of data with color frames and depth frames, with frame rates at 30fps for both (therefore there should be ~30 frames of color and depth frames in each bag file). I use the following code to read the bag files. 

int main(int argc, char * argv[])
{
   int start_shot = atoi(argv[1]); // = 0
   int end_shot = atoi(argv[2]); // = 22
   for(int s=start_shot; s<=end_shot; s++)
   {
        char bag_file[256];  sprintf(bag_file, "test_%04d.bag", s);
        if (!fileExists(bag_file)) continue;
        rs2::config cfg;
        cfg.enable_device_from_file(bag_file);
        rs2::pipeline pipe;
        rs2::pipeline_profile profile = pipe.start(cfg);
        auto color_stream = profile.get_stream(RS2_STREAM_COLOR).as<rs2::video_stream_profile>();
        auto depth_stream = profile.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>();
        int color_width = color_stream.width();
        int color_height = color_stream.height();
        int depth_width = depth_stream.width();
        int depth_height = depth_stream.height();
        float depth_scale = get_depth_scale(profile.get_device());
        rs2_intrinsics c_intrin = color_stream.get_intrinsics();
        rs2::frameset frameset;
        bool frameset_read = true;
        for(int i=0; i<10; i++) { // ignore the first 10 frameset
                if(pipe.try_wait_for_frames(&frameset, 5000)==false) {
                        std::cout << "try_wait_for_frames timeout" << std::endl;
                        frameset_read = false;
                        break;
                }
        }
        if (!frameset_read) continue;
        if(pipe.try_wait_for_frames(&frameset, 5000)==false) {
                std::cout << "try_wait_for_frames timeout" << std::endl;
                continue;
        }

        // process the frameset, which is commented out

       reture 0;
}

Sometime the program hangs in the middle of bags files (not always on the same one), and sometimes I encounter the following error:

"terminate called after throwing an instance of 'std::system_error'
  what():  Invalid argument
./run_test: line 28: 10163 Aborted                 (core dumped) ./run_test 0 22"

I really cannot see how that can happen. Any advice is appreciated.
Particularly:
(1) Is the way of using "try_wait_for_frames()" in the above code correct?
(2) Is it necessary to call pipe.end(), corresponding to pipe.start(), at the end of each loop?

Thanks.

ywjia, 2018-11-08:
I just installed the latest SDK, and after re-compiling my code, I got the following error message:

/home/user/librealsense-master/src/media/playback/playback_device.cpp:495: librealsense::playback_device::do_loop(T)::<lambda(dispatcher::cancellable_timer)> [with T = librealsense::playback_device::try_looping()::<lambda()>]: Assertion `m_is_started == false' failed.
./run_test: line 18: 28743 Aborted                 (core dumped) 

All the bag files can be played in realsense-viewer without any issue.

Thanks.

ywjia, 2018-11-08:
Using the latest development branch, the program hangs after running through bag files.
No error message was shown, just hanging there.

ywjia, 2018-11-08:
Actually, in another test with the latest release,  the same issue of hanging occurs again.

RealSense-Customer-Engineering, 2019-01-12:
[Realsense Customer Engineering Team Comment]
Hi ywjia,

Wonder if there is any update?

Some data point here: I used your code, but did a simply dumping some intrinsics info in the processing part. Also keep a loop of repeating the bag file operation for 100 times in a roll.
With such kind of loop test, I was able to execute for multiple times (>500 times total), and no issue found yet.

This may help debug the issue, if still need to.

Thanks!

RealSense-Customer-Engineering, 2019-01-12:
[Realsense Customer Engineering Team Comment]
BTW, I did with a D415 off an Intel NUC w/ Ubuntu 16.04.

Thanks!

ywjia, 2019-01-14:
Thanks, but I tested again (using firmware 05.10.13 ) and the issue is still there. 

Sometimes it worked fine. But if I run many times, sometimes I got "Segmentation fault (core dumped)", sometime I got "librealsense::playback_device::do_loop(T)::<lambda(dispatcher::cancellable_timer)> [with T = librealsense::playback_device::try_looping()::<lambda()>]: Assertion `m_is_started == false' failed.
".

How many bag files did are there in each loop, and roughly how many frames in each bag file, please?

RealSenseCustomerSupport, 2019-01-21:
Hi ywjia,

I mimicked your case: bag file for 1 second streaming, and tested 100 times for each loop.

Thanks,
Richard

RealSenseCustomerSupport, 2019-01-28:
Hi ywjia,

Wonder if there is any update?

Thank you!

ywjia, 2019-01-28:
I still have the issue. Is there any way I can sent the bag files to you to have a test, please? 
The bag files are a total of ~800Mbytes. Thanks.

RealSenseCustomerSupport, 2019-02-08:
----------------------------------------------

Hi ywjia,

Please check back later with Realsense future releases and see it will help fix your issue.
This one will be closed for now.

Thanks,
Richard

closed 2019-02-15 by RealSenseCustomerSupport

MeruloHD, 2022-08-03:
Hello! Did you ever find a solution to this problem? I have the exact same.