realsenseai/librealsense #2697: Realsense D415 with OpenNI2 "device not found". How to resolve it ?

yyxr75, 2018-11-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (5.10.6) | 
| Operating System & Version |   {Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  4.4.0-138-generic                                       | 
| Platform                                 | PC |
| SDK Version                            |  { build / 2.16.3 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {others }                          | 

### Issue Description

  The problem is, I followed the tutorial from librealsense/wrappers/openni2/README.md which tell me how to develope realsense devices on OpenNI2 framework, but things went wrong as the error gives: "device not found".
  I wonder whether my operating procedure was wrong or right. My steps goes like below:
  1.Download and install OpenNI2 SDK
  2.Run OpenNI2/Packaginng/ReleaseVersion.py to create a package named "OpenNI-Linux-x64-2.2"
  3.Copy "librs2driver.so"+"librealsense2.so"+"librealsense2.so.2"+"librealsense2.so.2.16.3" to "OpenNI-Linux-x64-2.2/Samples/Bin/OpenNI2/Drivers/"
  4.Then as the tutorial says "Launch any OpenNI2 example located at OPENNI2_DIR/Samples/Bin/", I choose "SimpleViewer". But result turned out wrong like this title described above. So
WHAT SHOULD I DO?

dorodnic, 2018-11-08:
How about realsense-veiwer?

yyxr75, 2018-11-08:
it works well, all the bin file of realsense works well

yyxr75, 2018-11-08:
it works very well.

dorodnic, 2018-11-08:
Hmm.. So far we only tested the OpenNi wrapper on Windows, so we will need to try to reproduce this. 
However, based on #2651 I believe @Daichou was able to make it work on Linux, perhaps he can comment?

Daichou, 2018-11-08:
It work well on my computer. (Ubuntu 18.04 + D435)
I don't use libopenni2-dev package. I just download openni2 from [official site](https://structure.io/openni) and run install.sh to install usb device rule. Following, I copy necessary diriver library (librs2driver.so"+"librealsense2.so"+"librealsense2.so.2"+"librealsense2.so.2.16.3) to downloaded openni2 directory (OpenNI-Linux-x64-2.2/Samples/Bin/OpenNI2/Drivers/) and run SimpleViewer. SimpleViewer window will pop out.

yyxr75, 2018-11-09:
Thank u all,
I will keep trying and give feedback as long as I make any progress

yyxr75, 2018-11-09:
YES, PROBLEM DONE
MUCH THANKS EVERYONE
I just made it, I follow the @Daichou 's guide and Down the “[OpenNI 2.2.0.33 Beta (x64) 3.0mb, zipped]” from [the OpenNI 2 SDK Binaries & Docs](https://structure.io/openni), and then 
RUN INSTALL.SH
This is the step I forget which bring me troubles. Because INSTALL is going to add the environment variables: OPENNI2_INCLUDE and OPENNI2_REDIST to the specific file.(I think so)
Whatever the reason is, thank you so much!
One drawing below:
[image]

closed 2018-11-09 by yyxr75

SFM2020, 2018-12-01:
I cant use openni with my D435 either. I ran sudo ./install.sh to install openni2,but I cant find "librs2driver.so" ,thats why ?

yyxr75, 2018-12-03:
you may try my procedure to download a openni2 firstly and compile it for the .so files

SFM2020, 2018-12-03:
> you may try my procedure to download a openni2 firstly and compile it for the .so files

I did your steps,but i cant find librs2driver.so in my pc

yyxr75, 2018-12-03:
have you review the page describing the [wrapper openni2](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openni2), which is given by the librealsense2 team. Did you follow the steps described here? 
These lib files are compiled from this.

yyxr75, 2018-12-03:
have you review the page describing the [wrapper openni2](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openni2), which is given by the librealsense2 team. Did you follow the steps described here?
These lib files are compiled from this.
remember to recompile your realsense-sdk to turn on the openni2

yyxr75, 2018-12-03:
have you review the page describing the [wrapper openni2](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openni2), which is given by the librealsense2 team. Did you follow the steps described here?
These lib files are compiled from this.
remember to recompile your realsense-sdk after turning on the openni2
use ccmake to find the switch of openni2

SFM2020, 2018-12-04:
> have you review the page describing the [wrapper openni2](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openni2), which is given by the librealsense2 team. Did you follow the [...]

thank you

cucrui, 2019-01-07:
> It work well on my computer. (Ubuntu 18.04 + D435)
> I don't use libopenni2-dev package. I just download openni2 from [official site](https://structure.io/openni) and run install.sh to install usb [...]

> It work well on my computer. (Ubuntu 18.04 + D435)
> I don't use libopenni2-dev package. I just download openni2 from [official site](https://structure.io/openni) and run install.sh to install usb [...]

sorry I met a problem that I cant find the file "librs2driver.so"

cucrui, 2019-01-07:
> thank you

have u found file "librs2driver.so"?

cucrui, 2019-01-08:
> have you review the page describing the [wrapper openni2](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openni2), which is given by the librealsense2 team. Did you follow the [...]

hey dude, I just follow your procedure and when i compiled the librealsense-2.17.1 I can just find files "librealsense2.so librealsense2.so.2 librealsense2.so.2.17.1" in "BUILD" folder, no file named "librs2driver.so" what should i do. Thx for ur reply!

yyxr75, 2019-01-08:
> hey dude, I just follow your procedure and when i compiled the librealsense-2.17.1 I can just find files "librealsense2.so librealsense2.so.2 librealsense2.so.2.17.1" in "BUILD" folder, no file [...]

but man,  that was how I did to make the lib files. Maybe u could try again or something. By the way, the .so files is under the linux system

mfoo431, 2019-01-11:
Greetings, @yyxr75! I'm having the same issue as @cucrui. librs2driver.so is nowhere to be found. I've been digging around for days, and I'm getting pretty desperate. Is there any chance you could upload your copy of the file somewhere? It'll save me from a lot of headache! Thanks!

cucrui, 2019-01-11:
> Greetings, @yyxr75! I'm having the same issue as @cucrui. librs2driver.so is nowhere to be found. I've been digging around for days, and I'm getting pretty desperate. Is there any chance you could upload your copy of the file somewhere? It'll save me from a lot of headache! Thanks!

Hey, dude. I figured out how to solve problem. In folder "Wrapper" , you can find folder named "openni2", cd in and compile it, and you can see in the CMakeLists.txt file, the project named "rs2driver"，so after compile it, the file "librs2driver.so" came out.That's it.

mfoo431, 2019-01-12:
IT WORKED! You're a lifesaver, @cucrui. Thank you so much for your help!

yyxr75, 2019-01-14:
Congratulation guys!

pillaya21, 2019-03-20:
> Hey, dude. I figured out how to solve problem. In folder "Wrapper" , you can find folder named "openni2", cd in and compile it, and you can see in the CMakeLists.txt file, the project named [...]

@cucrui hey I am also facing the same issue. Can you explain how to compile it?

cucrui, 2019-03-20:
> @cucrui hey I am also facing the same issue. Can you explain how to compile it?

just cmake and make, i if u dont meet any problem, u will see that file

pillaya21, 2019-03-20:
@cucrui  Yes, I tried that when I tried to make I got this error, thats why I thought I wasn't compiling it right
 odroid@odroid:~/librealsense/wrappers/openni2$ make
[ 14%] Building CXX object CMakeFiles/rs2driver.dir/src/Rs2Base.cpp.o
In file included from /home/odroid/librealsense/wrappers/openni2/src/Rs2Base.cpp:1:0:
/home/odroid/librealsense/wrappers/openni2/src/Rs2Base.h:3:33: fatal error: Driver/OniDriverAPI.h: No such file or directory
 #include <Driver/OniDriverAPI.h>
                                 ^
compilation terminated.
CMakeFiles/rs2driver.dir/build.make:62: recipe for target 'CMakeFiles/rs2driver.dir/src/Rs2Base.cpp.o' failed
make[2]: *** [CMakeFiles/rs2driver.dir/src/Rs2Base.cpp.o] Error 1
CMakeFiles/Makefile2:72: recipe for target 'CMakeFiles/rs2driver.dir/all' failed
make[1]: *** [CMakeFiles/rs2driver.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

I checked the openni2 installation and found the required file at 
/usr/include/OpenNI-Linux-Arm-2.2/Include/Driver
I also added the required variables to .bashrc
export OPENNI2_INCLUDE=/usr/include/OpenNI-Linux-Arm-2.2/Include
export OPENNI2_REDIST=/usr/include/OpenNI-Linux-Arm-2.2/Redist
I still got the same error

mfoo431, 2019-03-20:
@pillaya21 You might have to switch up a few things, but I hope these instructions help!

1) Navigate to librealsense-master/.

2) Create a folder here called build and navigate inside.

3) Open a terminal here and type in cmake ../.

4) In the same terminal, type in make.

5) Navigate to OpenNI-Linux-x64-2.2/Include/.

6) Copy everything here and paste it into librealsense-master/wrappers/openni2/src/.

7) Navigate to librealsense-master/wrappers/openni2/.

8) Create a folder here called build and navigate inside.

9) Open a terminal here and type in cmake ../.

10) In the same terminal, type in make.

11) There should now be a new folder called _out, and inside should be librs2driver.so.

12) Copy librs2driver.so and paste it into librealsense-master/build/.

13) All four driver files should now be in this folder.

nyxrobotics, 2019-08-08:
I have got D415's depth image on SimpleViewer following this thread, but I can't see image on NiViewer.
Here is the message.
> One or more of the following nodes could not be enumerated:
> Device: PrimeSense/SensorKinect/5.1.2.1: The device is not connected!
> Press any key to continue . . .

Do you have any idea ....?

cucrui, 2019-08-09:
> I have got D415's depth image on SimpleViewer following this thread, but I can't see image on NiViewer.
> Here is the message.
>
>
> Do you have any idea ....?

Try to follow mfoo431's comments, and have you get librs2driver.so file?

nyxrobotics, 2019-08-09:
Yes. I was able to run  SimpleViewer with librs2driver.so.

cucrui, 2019-08-09:
> Yes. I was able to run SimpleViewer with librs2driver.so.

Sorry, I never meet this problem. Waitting for others reply.