realsenseai/librealsense #2701: Suggestion: rs-convert.cpp should be a console function instead of a CMD function

HippoEug, 2018-11-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version | Win 10 | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC |
| SDK Version                            | 2.16.1 | 
| Language                            | C++  | 
| Segment			| others                   | 

# Suggestion
Hello! I am trying to streamline the process of getting (x, y, z) data stored in a .CSV file. Part of it involves converting the `.bag` into a `.csv` file using `rs-convert.cpp`.

[image]

I think this is a hassle to use the CMD everytime just to convert, and should instead be built into console for example (see below).

# Example
```
Specify the location of .bag file      : C:\Users\admin\Documents\a.bag
Specify the location to store .csv file: C:\Users\admin\Documents\datafiles
```

# Implementation
I do not think this particular suggestion/change should be complicated, as the actual conversion process code is:
```
auto pipe = make_shared<rs2::pipeline>();
rs2::config cfg;
cfg.enable_device_from_file(inputFilename.getValue());
pipe->start(cfg);

auto device = pipe->get_active_profile().get_device();
rs2::playback playback = device.as<rs2::playback>();
playback.set_real_time(false);

auto duration = playback.get_duration();
int progress = 0;
auto frameNumber = 0ULL;

while (true) {
    auto frameset = pipe->wait_for_frames();

    int posP = static_cast<int>(playback.get_position() * 100. / duration.count());

    if (posP > progress) {
        progress = posP;
        cout << posP << "%" << "\r" << flush;
    }

    if (frameset[0].get_frame_number() < frameNumber) {
        break;
    }

    frameNumber = frameset[0].get_frame_number();

    for_each(converters.begin(), converters.end(),
        [&frameset] (shared_ptr<rs2::tools::converter::converter_base>& converter) {
            converter->convert(frameset);
        });

    for_each(converters.begin(), converters.end(),
        [] (shared_ptr<rs2::tools::converter::converter_base>& converter) {
            converter->wait();
        });
}

cout << endl;

for_each(converters.begin(), converters.end(),
    [] (shared_ptr<rs2::tools::converter::converter_base>& converter) {
        cout << converter->get_statistics() << endl;
    });

return EXIT_SUCCESS;
```

dorodnic, 2018-11-09:
I think both use cases are valid, since parameter passing allows simpler batching. 
But when you run `rs-convert` without arguments, it can prompt for user input. 

This can be a great opportunity to give [contributing](https://github.com/IntelRealSense/librealsense/blob/master/CONTRIBUTING.md) a try ;)

HippoEug, 2018-11-09:
You're right. I first gotta figure out Cmake and Git better before I can attempt. Good for my learning too, I'll give it a shot. Thank you :)

closed 2018-11-09 by HippoEug