realsenseai/librealsense #2702: Windows UWP Driver not complete - Mapping Depth to RGB

teichgraf, 2018-11-09:
We are using the UWP Windows.Media.Capture APIs to access the RealSense D415. We need to map the depth data onto the color for rendering a point cloud.

We are doing a manual registration / mapping based on camera intrinsics and extrinsics. We can get the camera intrinsics from the UWP APIs but the coordinate mapping transform (extrinsics) are not provided.
This call would normally provide a 4x4 matrix for doing the transformation and it works fine with Kinect v2 but for the RealSense this returns a null value. 

var matDepthToRgb = depthSource.Info.CoordinateSystem.TryGetTransformTo(colorSource.Info.CoordinateSystem)

Got the latest driver and Windows 10 17134 but still the CoordinateSystem provided by any source regardless if marked as depth or color is NULL.

depthSource.Info.CoordinateSystem == null 
colorSource.Info.CoordinateSystem  == null

I know there is a point cloud sample in the native RealSense SDK that is using

environment::get_instance().get_extrinsics_graph()

But we need to use the UWP Windows.Media.Capture APIs and don't see those extrinsics values exposed. Not via the coordinate system transformation nor another API. 
We don't use the CoordinateMapper as we need to run the mapping in a custom shader and really just need the above mentioned matrix. This is a bug in the driver that should surface those parameters to Windows.Media.Capture but it's broken since months.

I have the latest Windows 10 UWP driver and D415 firmware installed as of today.
Winver 1803 / 17134 / RS4

dorodnic, 2018-11-09:
If using librealsense is an option, we have [sensor-control](https://github.com/IntelRealSense/librealsense/tree/master/examples/sensor-control) example that among other things offers extrinsics.
@RealSense-Customer-Engineering - can we check the UWP driver and submit a bug to the relevant team if there is no fix?

teichgraf, 2018-11-09:
Thanks for the quick response Sergey! 👍 
Unfortunately we can't use the native librealsense as our app has to be an UWP app.

teichgraf, 2018-11-09:
Yes, that is my discussion on the Intel forums but since there's no outcome since months I was trying another path here. 

The MRTK (Mixed Reality Toolkit) is not related at all to this issue. The MRTK is a toolkit for Unity development targeting the HoloLens. How would that at all fix a driver issue or be a workaround...

We need a working driver that is what is needed and it seems not to be open source, so no way to fix it ourselves. :(

teichgraf, 2018-11-09:
The non-Unity MRTK is mainly companion apps and tools. I know it quite well and it's not helping us here. Appreciate you are taking the efforts to post Marty and didn't want to sound rude. 🙌

MartyG-RealSense, 2018-11-09:
It's ok, I understand your frustrations.  

Could you give some information please about why using UWP is so important to your project?  When UWP first came out, its main advantage was that UWP apps would work on Windows 10, Xbox and Windows Phone.   Now that Windows Phone is dead, the advantages gained from using it instead of developing a Windows-only app may be limited.

teichgraf, 2018-11-09:
Our app needs to run on Windows 10: Desktop, HoloLens and XBox, so yes UWP. 
Also the universal layer with an unified API that Windows.Media.Capture offers is a big plus (if the driver is implemented fully).

dorodnic, 2018-11-14:
I was thinking about this problem... I believe Media Capture API allows the user to read from camera XU (need to look up how, couldn't find it online, but I think I saw someone do it). 
If this is true, one could send hardware command like in [ds5-device.cpp:271-275](https://github.com/IntelRealSense/librealsense/blob/master/src/ds5/ds5-device.cpp#L271-L275) and then parse the resulting table similar to how its done in librealsense.

ashikns, 2018-12-04:
I tried this in a C# sample. Short answer: it didn't work. It seems that the UWP driver simply does not provide access to the data. When trying to access it an exception is thrown saying "The property set does not exist on the device". Here is the code that I tried: https://gist.github.com/ashikns/21b93ebe94454fd1d85ed33415fe21e2. I also discovered another guy on github who has done the same but in a much more comprehensive manner: https://github.com/jubransh/SynchronicMediaCapture. The sample from this github link doesn't run without some fixes, but the gist of it is that even if you get it to run it'll still throw the same error as I mentioned before.

All experiments were done on a Intel D415 camera with firmware 5.10.6, OS is Windows 10 1809.

dorodnic, 2018-12-06:
Hi @ashikns 
I'm sorry it didn't work for you. 
With the default UWP driver I get the same "The property does not exist" error. However, with [D400 Series UWP Driver for Windows 10](https://downloadcenter.intel.com/download/27799/Intel-RealSense-D400-Series-Universal-Windows-Platform-UWP-Driver-for-Windows-10) installed I do manage to get data back. 
(The only change from the GIST is `var selectedGroup = groups.First(x => x.DisplayName.Contains("Depth"));` instead of `groups.First();`)

@teichgraf - could you please clarify if using the Intel UWP Driver is an option for you?

ashikns, 2018-12-10:
@dorodnic Confirming that the updated usb driver works 👍 Thanks for your help

teichgraf, 2018-12-13:
Thank you so much @dorodnic and @ashikns 👏

RealSense-Customer-Engineering, 2018-12-27:
[Realsense Customer Engineering Team Comment]
Hi @teichgraf,

closed 2018-12-27 by RealSense-Customer-Engineering

Skavesen, 2022-12-23:
does anyone have an example of a camera working with a uwp app?

MartyG-RealSense, 2022-12-23:
Hi @Skavesen  An app called **Camera Stream Correlation** has previously been suggested for use with RealSense, as described at https://github.com/IntelRealSense/librealsense/issues/2722

RealSense UWP support was not used by most RealSense users however and updates of the RealSense UWP driver have ceased.

Skavesen, 2022-12-23:
And if I need a UWP app, because there is a RealSense library, but I get an error during initialization. Any advice on fixing it? Windows 10 19044, camera d415, sdk 2.0 from github (downloaded today), camera driver 6.1.160.22, usb 3.0. Everything is according to the rules. But there is an error during initialization
 var pipeline = new Pipeline(); 
System.TypeInitializationException: 'The type initializer for 'Intel.RealSense.Context' threw an exception.'
PlatformNotSupportedException: Custom marshalers are not supported

MartyG-RealSense, 2022-12-23:
@Skavesen  The RealSense UWP driver is not required to use a RealSense 400 Series camera with Windows and is only needed in very rare circumstances where the camera needs to be made compatible with a UWP-supporting application such as the Microsoft Hololens headset.

If you do need to interface with a UWP application then the **PlatformNotSupportedException: Custom marshalers are not supported** error message has been reported previously at https://github.com/IntelRealSense/librealsense/issues/5774

There is not any information available regarding a solution and my recommendation would be to not attempt to use a RealSense camera with UWP applications.