realsenseai/librealsense #2703: Error Mac Nodejs wrapper example "Could not locate the bindings file. "

cadrinif, 2018-11-09:
When i run node nodejs-capture.js in example folder i have this error : 
Documents/Tests/realsense/librealsense-master/wrappers/nodejs/node_modules/bindings/bindings.js:91
  throw err
  ^

Error: Could not locate the bindings file. Tried:
/Documents/Tests/realsense/librealsense-master/wrappers/nodejs/build/node_librealsense.node

I have Mac Sierra, node -v (v8.11.1), npm -v (5.6.0).
The camera is D415.

RealSense-Customer-Engineering, 2019-01-10:
[Realsense Customer Engineering Team Comment]
Please make sure you have run "npm install" in wrappers/nodejs/examples folder before launch the example.

https://github.com/IntelRealSense/librealsense/tree/v2.17.1/wrappers/nodejs#2-run-examples

RealSense-Customer-Engineering, 2019-01-16:
[Realsense Customer Engineering Team Comment]
@cadrinif  Any update about this issue? Did you get through? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2019-01-21:
@cadrinif Any other questions about this issue? If no other questions, we'll close the ticket in a few days. Thanks!

closed 2019-02-04 by RealSenseCustomerSupport

flors, 2021-02-24:
Hi! I've got the same error... after running npm install