realsenseai/librealsense #2705: MATLAB Live Streaming and More Examples

apoorva2398, 2018-11-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version | Win 10  | 
| Platform                                 | PC|
| SDK Version                            |  2.16.1                         | 
| Language                            |  MATLAB                        |  

### Issue Description
I am currently using this code (modified depth_example.m) to live stream depth:

xxxxxxxxxxxxxxx
function depth_example()
    % Make Pipeline object to manage streaming
    pipe = realsense.pipeline();
    % Make Colorizer object to prettify depth output
    colorizer = realsense.colorizer();
    
    % Start streaming on an arbitrary camera with default settings
    profile = pipe.start();
    
    % Get streaming device's name
    dev = profile.get_device();
    name = dev.get_info(realsense.camera_info.name);
    
    % Get frames. We discard the first couple to allow
    % the camera time to settle
    while true
        fs = pipe.wait_for_frames();
        % Select depth frame
        depth = fs.get_depth_frame();
        % Colorize depth frame
        color = colorizer.colorize(depth);
        
        % Get actual data and convert into a format imshow can use
        % (Color data arrives as [R, G, B, R, G, B, ...] vector)
        data = color.get_data();
        channels = vec2mat(data, 3);
        img(:,:,1) = vec2mat(channels(:,1), color.get_width());
        img(:,:,2) = vec2mat(channels(:,2), color.get_width());
        img(:,:,3) = vec2mat(channels(:,3), color.get_width());
        
        % Display image
        imshow(img);
        title(sprintf("Colorized depth frame from %s", name));
    end
    
    % Stop streaming
    pipe.stop();
end
xxxxxxxxxxxxxxxxxxxxxxx

However, imshow is not recommended to be used in a loop. Is there a more efficient way to live stream in MATLAB? Also, is it possible to add more examples for common cases like changing camera settings and different streams like the ones for C++?

LouMagsClark, 2018-11-11:
This is what I have from modifying the depth_example:

`function stream_example() 

    %Make Pipeline object to manage streaming
    pipe = realsense.pipeline();
    %Make Colorizer object to prettify depth output
    colorizer = realsense.colorizer();
    %Start streaming on an arbitrary camera with default settings
    profile = pipe.start()
    
    %Set the timeframe for streaming in seconds
    tic
        while toc < 5
            
    %Wait for the frames
    fs = pipe.wait_for_frames();
    %Select depth frame
    depth = fs.get_depth_frame();
    %Colourize the depth frame 
    color = colorizer.colorize(depth);
    %Get actual data and convert into a format imshow can use
    data = color.get_data();
    %(Color data arrives as [R, G, B, R, G, B, ...] vector)
    img = permute(reshape(data',[3,color.get_width(),color.get_height()]),[3 2 1]);
    %Display the image
    imshow(img)`

I'm currently working on getting the x,y,z data whilst streaming. In terms of camera settings, I currently set them in the SDK before streaming in Matlab, although I'm not sure if there is an update for setting the processing options from issue #2520

apoorva2398, 2018-11-11:
Yes, that's similar to what I have.

Do you know how to stream the infrared and RGB frames as well?

lramati, 2018-11-12:
@apoorva2398 you can check out the [config ](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/config.m) class, the enable stream API is what you're looking for. The [C++ version](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_pipeline.hpp#L136) might be a bit more legible, and its identical to the Matlab one.

apoorva2398, 2018-11-12:
I keep getting an error at the third line when I run

`pipe = realsense.pipeline();`
`config = realsense.config();`
`config.enable_stream(realsense.stream.depth, 640, 480, realsense.format.z16, 30);`

The error is
`Error using realsense.stream`
`Cannot find a member of the 'realsense.stream' enumeration class that corresponds to each element of the given input argument.`

`Error in realsense.config/enable_stream (line 86)
            args = num2cell(int64([varargin{:}]));`

`Error in Sampling_IR1_and_Depth (line 5)
config.enable_stream(realsense.stream.depth, 640, 480, realsense.format.z16, 30)`

I tried stepping into the code, but I couldn't debug it. Could you please tell me what is wrong?

lramati, 2018-11-12:
I think its an issue with the square brackets on the line in the error, where its trying to convert all the args into realsense.stream objects to fit in the list. My knowledge of Matlab isn't great so there are some errors of this kind in the m files of the wrapper. Replacing the line with a for loop that accomplishes the same thing should solve your issue

apoorva2398, 2018-11-12:
Thank you! The updated config.m file works correctly.

closed 2018-11-12 by apoorva2398

jattiogbe, 2019-03-14:
Hi @apoorva2398 and @LouMagsClark ,

I am a new user of D415 and Matlab, I wanted to know if you managed to stream the infrared and RGB frames ?

DieterHeestermans, 2019-03-15:
@jattiogbe 
Here is the code for streaming RGB frames:

%Make Pipeline object to manage streaming
pipe = realsense.pipeline();
%Start streaming on an arbitrary camera with default settings
profile = pipe.start()

%Set the timeframe for streaming in seconds
tic
    while toc < inf
        
%Wait for the frames
fs = pipe.wait_for_frames();
%Select color frame
color = fs.get_color_frame();
%Get actual data and convert into a format imshow can use
data = color.get_data();
%(Color data arrives as [R, G, B, R, G, B, ...] vector)
img = permute(reshape(data',[3,color.get_width(),color.get_height()]),[3 2 1]);
%Display the image
imshow(img)
end

Kind regards

jattiogbe, 2019-03-15:
Great, Thank you !
I will try this next week, edit this post and let you know how it went. 

Best regards