realsenseai/librealsense #2706: C# Export vertices coordinates

yohanb78, 2018-11-11:
Hello,

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.10.03.00 | 
| Operating System & Version |   {Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.16.1                          | 
| Language                            |  C#                   | 
| Segment			|  others                           | 

I'm trying to export a data snapshot of the vertices in a json format.
Below, a sample of my code inspired by the C# capture example :

```
//...
using (var frames = pipeline.WaitForFrames())
                        {
                            var depthFrame = frames.DepthFrame.DisposeWith(frames);
                            PointCloud pc = new PointCloud();
                            var points = (Points) pc.Process(depthFrame).DisposeWith(frames);
                            var colorizedDepth = colorizer.Process(depthFrame).DisposeWith(frames);
                            Dispatcher.Invoke(DispatcherPriority.Render, updateDepth, colorizedDepth);
                            **Dispatcher.Invoke(DispatcherPriority.Render, savePoints, points);**
                            
                        }
//...

static Action<Points> savePointClouds()
        {
            return new Action<Points>(points =>
            {
            using (points)
            {
                        Points.Vertex vertex = Marshal.PtrToStructure<Points.Vertex>(points.VertexData);
                        for (int i = 0; i < points.Count; i++)
                        {
                            var str = string.Format("x:{0}, y:{1}, z:{2}", vertex.x, vertex.y, vertex.z);
                        }
                }
            });
        }
```

My comprehension of the SDK tells me that variable vertex should be an Array of struct Vertex.
But I have failed to extract it after many code attempts

If someone has an idea, it will be helpful.

Thanks a lot

HippoEug, 2018-11-11:
Hi there, I think I was trying to a very similar thing to yours but in C++, difference also being I was trying to save the data in .CSV format. [Here](https://github.com/IntelRealSense/librealsense/issues/2584#issuecomment-434147781) was my issue, but from what I know this doesn't sound possible.

ogoshen, 2018-11-11:
> My comprehension of the SDK tells me that variable vertex should be an Array of struct Vertex.

That's right, you should use something like this:
```cs
var vertices = new Points.Vertex[points.Count];
points.CopyTo(vertices);
```

yohanb78, 2018-11-11:
Thanks to you,

@HippoEug : efore I posted my issue I read yours but it was hard to me to correlate the methods in c++ with c# methods.

@ogoshen : Why I didnt' find the copyTo method in the sdk doc... :( I guess that it s because I didn't succeed in generating the doc for the wrapper C# when I compiled to the realsense dll.

Is there a place where I can find the C# doc, or I have to generate it ?
I will close this issue after I had an answer about this 👍 
Best,

RealSenseCustomerSupport, 2019-02-19:
----------------------------------------------

Hi @yohanb78,

Still need any support for this topic?
It looks the C# API doc needs to be improved.

yohanb78, 2019-04-08:
Hello,

Yes you're right. I will close this issue, I have now moved my devlopment around the nodejs wrapper in order to be coherent with the stack of the rest of my apps.

Regards

closed 2019-04-08 by yohanb78