realsenseai/librealsense #2707: Inter Cam HW Sync causes RGB/D drift

kiran1228, 2018-11-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.10.6 | 
| Operating System & Version |  Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.15.0-36 (with patch for metadata)       | 
| Platform                                 | PC |
| SDK Version                            |  2.16.4    | 
| Language                            |  C++                        | 

### Issue Description
We have an application that is running multiple cameras and are trying to achieve both RGB-D sync per camera and also inter camera sync between devices. We are using the callback method to fill a `rs2::frame_queue` for the depth and color separately then sync them ourselves by reading the `RS2_FRAME_METADATA_FRAME_TIMESTAMP` from the frame metadata. The `RS2_OPTION_AUTO_EXPOSURE_PRIORITY` option is set to 0.0 as described in other threads to achieve RGB-D sync. The depth camera is set to manual exposure mode.

I have found that when setting the `RS2_OPTION_INTER_CAM_SYNC_MODE` to 0.0 (no sync) the difference between the depth and color frame time stamps (depth_ts - color_ts) are steadily ~300 usec. With `RS2_OPTION_INTER_CAM_SYNC_MODE` to 2.0 (multicam sync) the frame time stamps will drift (color frames return faster). The total amount of time for drift depends on how many frames are queued and how often we synchronize the frames. If we check for sync every frame and the queue is a max of 1 frame then the difference between the depth and color frame time stamps (depth_ts - color_ts) drift between -15000 to 15000usec. 

Is this a behavior what is expected with these settings or is there some settings that are needed to have both syncs work together.

RealSense-Customer-Engineering, 2018-11-14:
[Realsense Customer Engineering Team Comment]
So you're trying to make multiple D415 cameras be sync'd together?

1) How does the Sync cable look like ? how long is it? Please keep in mind, the Slave camera is very sensitive to ESD which causes Slave camera frequently interrupted. Sometimes RGB camera may stop streaming.

2) When you run Sync mode, please disable "AE Priority" under RGB camera.

3) Always start Master Depth camera first, then start Slave camera (Depth and RGB).

Thanks

kiran1228, 2018-11-14:
1 & 3. The sync cables we are using are several meters long but we are employing the ethernet method from a central sync generator. 
2. "AE Priority" for the RGB camera is set to 0.0.

I have also noticed that when we turn on the `RS2_OPTION_INTER_CAM_SYNC_MODE` to 2.0 and there is no sync signal we still get the same frame drifting. I read is a seperate issue if there is no sync signal for the depth camera it will timeout and trigger itself. Is it also expected for sync issues in this case?

RealSense-Customer-Engineering, 2018-11-14:
[Realsense Customer Engineering Team Comment]
OK, I assume the cable is well built and immune to ESD.

If Slave camera doesn't get the Sync pulse every frame, it will just run freely upon its own clock (own frame rate) which means it will start drifting.

You must set one camera to be the Master camera, and all others to be the Slave cameras. Always set Master first and run Master first, then set Slave and run Slave.

kiran1228, 2018-11-14:
I get how this effects the sync between d415 units but why does that effect how well the rgb and depth are synced within a single unit?

RealSense-Customer-Engineering, 2018-11-14:
[Realsense Customer Engineering Team Comment]
Within a single unit, the firmware might set RGB camera as Master and Depth camera as Slave. In this case, the ESD won't kill Slave camera (Depth) because Depth camera can be auto reset if frames drop too many (but RGB camera doesn't have such feature.)

To check which one is master, Depth or RGB, you can click icon "i" which realtime display the frame rate. Usually on Depth camera it shows 30.01fps, and RGB shows 29.97fps. If Depth is configured as a Master, both Depth and RGB will show 30.01fps once they are sync'd together, otherwise, they both shows 29.97fps if RGB is configured as Master.

btw, I assume your PC is not a slow one.

Thanks

kiran1228, 2018-11-15:
So I'm testing this with a camera with FW 5.10.6.0 which is the latest one on the realsense website. I'm seeing that the framerates for both cameras are 29.76fps. So this implies that the RGB camera is configured as the master of the pair of cameras, correct?

So for me to achieve multicamera sync while keeping the RGB-D sync I would need to switch the master-slave relationship of the RGB and D cameras such that the D camera is master and the RGB is slave. This has the risk of the RGB dropping frames which could be kill the camera. 

If we don't change the relationship Is there a way to have the external sync pulse to trigger the color camera?

RealSense-Customer-Engineering, 2018-11-15:
[Realsense Customer Engineering Team Comment]
Good, that means the RGB-D is sync'd, but this only works for Single camera or on Master camera.

For multiple cameras sync application, all Slave cameras including Depth and RGB are running as Slave camera, so the ESD issue still has the impact.

Using external sync trigger is fine as long as you can filter out the ESD, but please note that you will have to tweak the sync trigger very accurately to satisfy the Slave cameras. For example, from my experiment, the D415 actual frame rate for 30fps is 30.01091Hz, you will need to generate same trigger pulse.
I don't have the data about the range of trigger period at this moment. I remembered the delta should be less than 1 line (or maybe a couple).

Thanks

kiran1228, 2018-11-16:
> Using external sync trigger is fine as long as you can filter out the ESD, but please note that you will have to tweak the sync trigger very accurately to satisfy the Slave cameras. For example, [...]
I don't have the data about the range of trigger period at this moment. I remembered the delta should be less than 1 line (or maybe a couple).

So based on your statement above I'm not seeing how it is possible to achieve both RBG-D sync and intercam sync at the same time. If we need to tune the trigger to fire at the actual framerate of the depth camera then will the color camera also by synced to this signal? Or do we need to set the trigger to the actual framerate of the color camera which is around 29.761hz.

As a more global question, have you or any applications you know of reliably gotten multicam RGB-D sync working. It seems like this system is very prone to errors and I am wondering if this is practical.

RealSense-Customer-Engineering, 2018-11-16:
[Realsense Customer Engineering Team Comment]
From image sensors point view, all image sensors (including master/slave and Depth/RGB) are sync'd at the same time (delta<10us) of End of Exposure.

For D415, the Depth sensor and RGB sensor are same chips, but using different settings.

If you set all cameras to Slave cameras (Inter Cam Sync Mode=2.000; and RGB AE Priority=Disabled), you can use an external independent trigger (1.8V, 50us~100us) to drive all the Slave cameras (pin#5 of the 9-pin connector). I recommend to make the trigger align to the Depth frame rate (30.01Hz). [Hint, use oscilloscope to help tweak the signal generator. (I haven't tried 29.761Hz on depth sensor)]

There are some famous customers using Multi-cam Sync feature, but I just can't disclose the names.

closed 2018-12-03 by RealSense-Customer-Engineering