realsenseai/librealsense #2712: patch-realsense-ubuntu-xenial.sh: No such file or directory

Junzeng-x14, 2018-11-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-38-generic  | 
| Platform                                 | PC |

### Issue Description
I follow the instructions of installation as [Full Build](https://github.com/Interbotix/turtlebot2i/wiki/Full-Build-Instructions)

`sudo apt-get install libglfw3-dev`

`cd ~`

`git clone https://github.com/IntelRealSense/librealsense.git`

`cd librealsense`

`mkdir build && cd build`

`cmake ../`

`make && sudo make install`

`cd ..`

`sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/`

`sudo udevadm control --reload-rules && udevadm trigger`

`./scripts/patch-realsense-ubuntu-xenial.sh`

Everything works until the last line, it's said that `./scripts/patch-realsense-ubuntu-xenial.sh: No such file or directory`. I thought it might come from patch skip.

```
--------------------------
|diff --git a/drivers/media/usb/uvc/Makefile b/drivers/media/usb/uvc/Makefile
|index c26d12fdb8f4..d86cf22155d1 100644
|--- a/drivers/media/usb/uvc/Makefile
|+++ b/drivers/media/usb/uvc/Makefile
--------------------------
File to patch: 
Skip this patch? [y] 
Skipping patch.
1 out of 1 hunk ignored
```

No idea about how to solve this problem with my current version of kernel. Thank for help in advance!

dorodnic, 2018-11-12:
Hi @Junzeng-x14 
The instructions are somewhat outdated, however, in this context, I think it is absolutely safe to skip the patch and continue with the installation process.

Junzeng-x14, 2018-11-12:
Yes, I did skip the patch information and installation process, but however the file `patch-realsense-ubuntu-xenial.sh` cannot be find in the folder `scripts`.

dorodnic, 2018-11-12:
`patch-realsense-ubuntu-xenial.sh` is the kernel patch application script for RealSense cameras, but you no longer need a kernel patch to work with the SR300 with kernel 4.15, so there is no real reason to run it. 
The file is still [available](https://github.com/IntelRealSense/librealsense/blob/legacy/scripts/patch-realsense-ubuntu-xenial.sh), but the instructions should be updated with `git checkout legacy`, and this should only be done for olders kernels.

RealSense-Customer-Engineering, 2018-12-07:
[Realsense Customer Engineering Team Comment]
No response in a month.  Close issue as solved.

closed 2018-12-07 by RealSense-Customer-Engineering