realsenseai/librealsense #2713: Pipeline.Start() Error in PointCloudProcessingBlocks Scene

GameHuiTea, 2018-11-13:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10) | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.0 }                          | 
| Language                            |  {unity }                          | 
| Segment			|  {AR }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
When i run PointCloudProcessingBlocks Scene. I meet this error.

Exception: Couldn't resolve requests
rs2_pipeline_start_with_config(pipe:0000000042D68FD0, config:0000000042D68D30)

what should i do?

dorodnic, 2018-11-13:
There is a lot of information missing, but I'd guess that the camera was detected as a USB2 device and hence cannot provide the configuration required in the Unity scene. 
Please share more about the setup and the error and we will do our best to assist.

GameHuiTea, 2018-11-14:
Okay. I'll do my best and give you the information. Thank you very much. 
first I using D415 model. and
Unity package(Download realsense.unitypackage) and Git hub have determined that Unity folder in "Clone or Download" is different What should I use?
present, I'm using project by Unity package.
Can play scene except AlignmentSample scene and Point Cloud Depth and Colorcene.

if play AlignmentSample scene. project crash. I don't know about this.

I think Unity doc info match "Clone or Download" project. I'm Still not Trying.
this way is correct? Do I need to do this through a clone or download project?

GameHuiTea, 2018-11-14:
I don't touch any settings. All setting is default.
I've found a more detailed cause of the error.

The pointCloudProcessingBlocks scene failed because the RSDevice parameter Profiles contains values that do not support the device.
I changed 
width = 1280 -> 640
height = 720 -> 480
This is how we solved the problem.

and crash problem of AlignmentSample scene.
if RsAlign script enable. project crash. i don't have idea about this. help me TAT
default Parameters is 
order: 5
fork: false
Holes Fill: 0

thank you for reading

GameHuiTea, 2018-11-14:
I'd like to know more about the stream index.

GameHuiTea, 2018-11-14:
thank you for reply.
Okay. this is USB picture.
[image]

I think this is USB2.0! I changed thank you.
but AlignmentSample scene still crash maybe different problem. TAT

steam index pictures
[image]
[image]

I have same think about serail number
but I don't know what serial number do.

HippoEug, 2018-11-14:
I can see you are using the USB Cable that came with the Intel Realsense camera. But have you confirmed the black ports on the back of your Desktop PC is USB 3.0? Most USB 3.0 ports are blue instead of black due to the standard, but there are exceptions of course. Have you confirmed the camera is indeed connected to your PC as USB3?

[image]

Have you tried using Intel Realsense Viewer Application? Are you able to stream both depth and color at the same time? Here is the newly published [installer](https://github.com/IntelRealSense/librealsense/releases/download/v2.16.2/Intel.RealSense.SDK-2.16.2.304.exe) if you have not downloaded somehow.

Also, the installer does install some example programs using C++ too, you can run them if you have Visual Studio installed. 
Here are the things installed after installing the SDK using the link above. 

[image]

Running the `Examples for Intel RealSense SDK 2.0` results in:

[image]

I can also run the align function but with C++ here:

[image]
[image]

If you can indeed run the align example on C++, it must be some configuration with Unity then. I do not use Unity, so I am unable to help you in that area.

dorodnic, 2018-11-14:
Hi @GameHuiTea 
Thank you for providing more information. I think the camera is still connected to USB2 and this might be what causing the problem. The cable in your picture is USB3, but the port is USB2 I believe. 
USB3 ports are usually blue and marked with Super-Speed label, like this:
[image]
It is possible that the Unity scene has trouble with this. Alternatively, there might be something in the Scene itself, perhaps it requires higher shader spec. I hope @matkatz or @ogoshen would know more.

GameHuiTea, 2018-11-15:
I solved all the problems. Thank you very much. The only thing left is to solve my curiosity. I wonder what the steam index and the requested serial number mean!

HippoEug, 2018-11-15:
> I solved all the problems

The align function doesn't crash anymore? Because of USB2.0?

> I wonder what the stream index and the requested serial number mean

I am not sure about this at all, maybe someone from Intel or uses Unity can better answer your question. But from guessing, I think the program supports multiple Depth Cameras to be used simultaneously to cover a 3D scene. 

[image]
[image]

The serial number is to probably give you an option to uniquely identify which camera you want to configure/capture or something. 

About Stream Index, there are a few functions that passes it as a parameter I think. Here are 2 examples:

#### 1
```
void rs::device::get_stream_mode (stream stream,
    int index,
    int& width,
    int& height,
    format& format,
    int& framerate
) const
```
#### 2
```
rs_device* rs_get_device(rs_context* context, 
    int index,
    rs_error** error
)
```

I think it is to identify which stream you're using, Depth Stream could be 0, Color Stream could be 1 and vice versa. It could also be an index of device to retrieve, maybe you have 4 cameras, and Index represented are 0, 1, 2, 3 for each camera?

HippoEug, 2018-11-15:
Great you managed to solve your issues. Hope someone from Intel confirms what Stream Index is!

RealSense-Customer-Engineering, 2018-11-19:
[Realsense Customer Engineering Team Comment]
@GameHuiTea For Stream Index, please refer to below description in  https://github.com/IntelRealSense/librealsense/blob/master/examples/sensor-control/api_how_to.h

            // The rs2_stream provides only types of data which are
            //  supported by the RealSense SDK
            // For example:
            //    * rs2_stream::RS2_STREAM_DEPTH describes a stream of depth images
            //    * rs2_stream::RS2_STREAM_COLOR describes a stream of color images
            //    * rs2_stream::RS2_STREAM_INFRARED describes a stream of infrared images

            // As mentioned, a sensor can have multiple streams.
            // In order to distinguish between streams with the same
            //  stream type we can use the following methods:

            // 1) Each stream type can have multiple occurances.
            //    All streams, of the same type, provided from a single
            //    device have distinct indices:
            int stream_index = stream_profile.stream_index();

GameHuiTea, 2018-11-19:
> [Realsense Customer Engineering Team Comment]
Thank you! Your answer is helpful to me!
lastly, what is required serial number meaning?

closed 2018-11-20 by GameHuiTea