realsenseai/librealsense #2714: selected configuration(fps,resolution) is not supported

123liluky, 2018-11-13:
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | 05.09.14.00 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC |
| Language                            |  python | 
While I change resolution from 640*480 to 1280*720 and FPS from 15 to 30, RGB Camera can not be started. But I started RGB Camera in 640*512 resolution and 30 FPS before. Where is the problem?
[image]

dorodnic, 2018-11-13:
@123liluky, as @HippoEug suggested the camera seem to be detected as a USB2.1 device.
To see the full list of supported resolutions and FPS values, you can run `rs-enumerate-devices`.

123liluky, 2018-11-14:
Thank you. I have changed to USB3 and the problem is solved.

closed 2018-11-14 by 123liluky