realsenseai/librealsense #2715: Depth data shows up as green in programs other than realsense-viewer

PancakeSparkle, 2018-11-13:
Camera Model: SR300
Firmware Version: 3.21.0.0
Operating System & Version: Ubuntu 18.10
Kernel Version: 4.18.0                                       
Platform: PC
SDK Version: 2.16.4                       
              

### Issue Description
Trying to view data from the depth camera in other programs than realsense-viewer results in a green screen
[image]
[image]

dorodnic, 2018-11-13:
You should have two video devices when connecting the depth camera - one for depth information and one for color. If you select the second one, you should be able to use it as a regular web-cam. The depth device provides depth information that cannot be processed by regular cv algorithms. We hope that using librealsense more developers will start taking advantage of depth but RealSense also supports several other industry standards, such as OpenNi2 and Microsoft's new Video Capture API.

RealSense-Customer-Engineering, 2018-11-15:
[Realsense Customer Engineering Team Comment]
Librealsense does provide the ability to stream the infrared stream of SR300.  Should be available in the API stack.  I believe it's outlined in the src/ivcam folder.

dorodnic, 2018-11-20:
@PancakeSparkle - the reason it is problematic, is that for SR300 the infrared image is not exposed as a separate video device, but rather via special FOURCC formats of the depth device.

closed 2018-11-20 by RealSense-Customer-Engineering