realsenseai/librealsense #2718: Announcement: RealSense D435i camera available for pre-order

MartyG-RealSense, 2018-11-13:
Hi everyone,

The RealSense D435i camera model with added Inertial Measurement Unit (IMU) is now available for pre-order from Intel's online Click store for shipping in the week of December 3 2018.

https://communities.intel.com/thread/145631

UnaNancyOwen, 2018-11-15:
@dorodnic @MartyG-RealSense Can you distribution sample data (*.bag) that including IMU data of RealSense D435i on [this page](https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md)? (or Do you have that plan?) Thanks!

MartyG-RealSense, 2018-11-15:
The IMU data is time-stamped to align with depth data, according to the store listing for the D435i.  I don't have any information about how that data may be written into a bag though.  Dorodnic should be able to help you with that question.

HippoEug, 2018-11-15:
Hopefully with this D435i, more development would be made towards a full 360° point-cloud object 3D reconstruction included in the SDK!

dorodnic, 2018-11-15:
> @dorodnic @MartyG-RealSense Can you distribution sample data (*.bag) that including IMU data of RealSense D435i on this page? (or Do you have that plan?) Thanks!

We should publish something by early next week.

I'd like to note that on launch librealsense is only going to support D435i on Linux and Windows 10. We are still working on integrating more platforms. That said, the IMU is exposed as standard compliant HID device with accelerometer and gyro sensors. 

We are also working on adding dedicated documentation and tutorials.

kartur, 2018-11-16:
> documentation and tutorials

Can you say a little bit more about this? Will tutorials be available when the D435i ships or later?
Will tutorials include a SLAM for example?

MartyG-RealSense, 2018-11-16:
It was stated in Intel's multiple camera webinar in September 2018 that Intel were not planning on releasing their own SLAM solution, and strongly encourage other companies to provide a solution if they wish.   It was said, "We will be happy to post whatever you have on our website, and you can even sell it there in the future".

There is a precedent though for individual Intel employees contributing tutorials for non-Intel SLAM solutions, such as the ORB-SLAM2 tutorial that was published by Meitav Kleinfeld for Intel Euclid (a candy bar sized PC with a RealSense ZR300 camera inside).

https://www.euclidcommunity.intel.com/static/tutorials/pdf/orbslam/orbslam2_tutorial.pdf

dorodnic, 2018-11-17:
Hello @kartur 
SLAM will not be part of the SDK. 
Using this SDK, you will be able to fetch IMU data synchronously or asynchronously from many languages (and eventually on all platforms). 
We will try to provide guides for integration with free open-source libraries (in addition to ORB-SLAM, [google cartographer](https://github.com/googlecartographer) seem promising) and evaluate and promote commercial solutions (at [software-for-intel-realsense](https://realsense.intel.com/software-for-intel-realsense/))

kartur, 2018-11-19:
@MartyG-RealSense 
@dorodnic 
Thanks for the answers, a guide for ORB-SLAM2 would be a nice thing IMO.

MartyG-RealSense, 2018-11-19:
The Euclid's  ORB-SLAM2 example above is mostly based around ROS instructions.  So I would speculate it may work with a 400 Series camera too if the roslaunch instruction was changed to suit a D415 or D435 instead of the ZR300.

MartyG-RealSense, 2018-11-19:
A member of Intel Support said that the ZR300's IMU is the Bosch BMI055, and provided a link to a data sheet document for it.

https://www.mouser.com/ds/2/783/BST-BMI055-DS000-08-786482.pdf

The store listing for the D435i says it uses the BMI055 too.

The information for the T265 Tracking Module will probably be added to an upcoming edition of the 400 Series camera data sheet, since the data sheet used to cover its predecessor the T150 Tracking Module before it was discontinued.

skohlbr, 2018-11-19:
AFAIK there was an issue with the IMU and frame data not being synchronized on the ZR300. I take this is improved in the D435i?

MartyG-RealSense, 2018-11-19:
On the D435i, IMU data is timestamped to be aligned with depth data as needed.

Thisismydigitalself, 2018-11-20:
In the process of learning my way into 3D scanning i bumped into this new 435i sensor and I was thinking of purchasing one but I don't know what is the accuracy one should expect from such sensor(s) in regards to the actual/accurate size of an object being scanned and then imported to CAD. will i get a 1:1 size or i will need to re-scale / calibrate size in CAD? I also like to know if having an IMU might one day help in getting better/faster scans or this feature is there for other purposes and the previous model will fit my needs better for less $$.

I am ready to pre-order this week but first I need to know it to fits my needs.
Thank you all in advace. this 3D scanning is fun.

MartyG-RealSense, 2018-11-20:
Intel's documentation says about accuracy that "The depth accuracy of the D400 series cameras is less than one percent of the distance from the object.  So if the camera is 1 meter from the object, the expected accuracy is between 2.5mm to 5mm."

The extra set of data that the IMU provides allows for better 'dense reconstruction' during a scan.   The primary factors that affect accuracy though include distance of the object from the camera, calibration, lighting and the scene.  So whilst you may get a higher quality model thanks to the enhanced dense reconstruction, I do not know if you would get an advantage in terms of increased accuracy.

In regard to importing scans into another package such as 3D modeling software, others who have had scale problems when importing files such as point cloud .ply have fixed it by changing the depth scale in the program that they are importing the scan into.

Thisismydigitalself, 2018-11-20:
Thank you very much Marty for the information you've provided. This is much appreciated.
there is no problem scaling an imported model to CAD, it's just me wondering how much will be needed cause if this sensor is not accurate down to the mm than this obviously defeats the whole purpose.
I am more than eager to learn everything that i can in regard to 435i and importing scaned model to CAD (I use Fusion360) but i failed to find much inforamtion on-line. I would be grateful if you could please direct me to where i should be looking. thank you.

MartyG-RealSense, 2018-11-20:
If accuracy is vital to you and you are going to be scanning at close range, you may be able to get additional accuracy by changing the depth unit value of the camera.  The 400 Series is configured to be able to 'see' up to 65 meters (and depth-sense up to 10 meters), but the trade-off is that at close range the accuracy may be less than the earlier RealSense SR300 model.  By changing the depth unit value, you can get the 400 Series cameras to mimic the close-range performance of the SR300.

The SR300's default scale is 1/32th of a millimeter and the 400 Series' default scale is 1 millimeter,   The documentation for the 400 Series' 'Projection' system of coordinate calculation states that "the depth scale can be modified by calling rs2_set_option() with RS2_OPTION_DEPTH_UNITS, which specifies the number of micrometers per one increment of depth. 1000 would indicate millimeter scale",  

"10000 would indicate centimeter scale, while 31 would roughly approximate the SR300's 1/32th of a millimeter scale".

MartyG-RealSense, 2018-11-20:
The easiest route to importing into Fusion 360 may be to create an .obj format 3D model with the camera.  This can be done with commercial software, though if you do not wish to  buy more software, there is a way to do it for free that requires a bit more labor.

1.  Take a 'point cloud' scan with the RealSense camera to create a 'ply' point cloud file.
2.  Import the ply file into a free 3D modeling software such as MeshLab or Blender and convert the ply into a solid mesh, then export it as an .obj and import the obj into Fusion 360.

You could save a lot of labor by using the earlier SR300 camera model though.  The 2016 RealSense SDK has a free program called 3DScan that can create an obj scan directly, and then you should be able to import that straight into Fusion 360.

MartyG-RealSense, 2018-11-20:
I apologize for causing confusion.  I like to present a range of different options for consideration so that users can find the solution that is best suited to their needs.

The 400 Series hardware is very capable indeed, and more powerful than previous generations of camera.  The 'Stereoscopic' technology that it uses though (a pair of infrared sensors) sometimes produces results that people believe is not as good as cameras such as Kinect 1 and the SR300 (which use a different camera imaging technology called 'structured light').

 Yes, people do understandably want to buy the product that will have the longest lifetime before the next generation.  A roadmap for the 400 Series that was released to the public in 2017 projected a 3 year life cycle from 2017-2019, with the next generation of RealSense camera being released in 2020.  That is just the sales lifetime though.  All models of RealSense camera have technical advice provided for them for years after they are no longer sold, though software and driver support for them may end.

As time goes by though, the 400 Series is becoming better and better as it receives continuous, regular updates, and the open source nature of its SDK software means that it can continue to receive support from its community over the long term even when Intel's official software update support ends. 

If long-term support is important to you then the 400 Series is a natural choice, since the next generation in 2020 is likely to have a smooth upgrade path between the 400 Series and its successor.

In conclusion, it is certainly possible to achieve your project goals with the 400 Series.  It may be easiest if you can afford to invest in additional software as well as the Fusion 360 you already have.  It is still possible to create the .obj model for free, though it will take a bit more time and labor.

Thisismydigitalself, 2018-11-20:
Many thanks Marty. Very interesting information for someone like me who just dived into this 3d scanning world just couple of days ago out of necessity following the need to design special harnesses for wounded wild animals and I want to give realsense a chance to prove itself worthy. just for the record, you're not the one to confuse me, on the contrary. it's the internet in conjunction with new product's speculations and contradicting information that makes it so hard for noob like me to decide and trust me to read MANY pages on Realsense and 3D scanning before posting here. the 435i is on pre-sale without any real information and comparision reviews done by Intel but then again - I am a noobie to all of this exciting 3D scanning so no complains whatsoever. can't wait to start playing and working with this technology.

I guess it's gonna be the 435i. i have faith in the open-source community.
Cheers.

MartyG-RealSense, 2018-11-20:
Thank you very much for your kind words.  :)

The D435i is identical to the standard D435 aside from the addition of the IMU.  So any information you find about D435 will apply to D435i too.

I can understand your particular application with the harnesses.  I have watched a lot of TV shows about veterinary surgery.  I also gave advice on a RealSense project where the heads of children with disabilities were being 3D scanned to create custom molds for helmets for them (standard store bought helmets would not fit their head shape).

https://communities.intel.com/message/533342#533342

Thisismydigitalself, 2018-11-20:
Cool, thanks for the linked thread. will read it through tomorrow.
Btw, Out of so many countries Intel store does not ship to mine, errr..

MartyG-RealSense, 2018-11-20:
What country are you in, if I may ask?

You may be able to buy from an Intel Approved retailer such as Mouser.

https://www.mouser.co.uk/ProductDetail/Intel/82635D435IDK5P/?qs=%2fha2pyFaduhZX4eNlY6xD%252bUZRZrFYuPskY21RPa96oI%3d

Thisismydigitalself, 2018-11-21:
Nietzche said "Out of chaos, comes order". I sure hope so.
Intel does not make it easy for some of us to buy the D435i.
- Intel does not list my countly.
- Framos is showing image of the D435i but takes order for D435
- Mouser minimum order is 5 units
- Stemmer imaging list only the D435.

So it seem i wont be ordering any time soon. 

[image]

[image]

[image]

MartyG-RealSense, 2018-11-21:
There are occasionally cases of countries (such as Dubai and the UAE) where Intel do not ship directly to and so local suppliers / retailers are the only recourse.  So I do understand the frustration with that.

It may be related to certification of the product.  For example, Israel is not on the standard shipping list, so the Click store creates special product listings just for Israel delivery.  Another curiosity is that the store's Up Board and NUC bundle deals cannot be shipped to the United Kingdom, even though the cameras on their own can.

You can search Intel's database of Intel Approved Distributors for a local retailer.  You may be able to obtain a standard D435 from them.  The standard model is likely to be fine for your application, as you would get the most advantage from the D435i if your project was related to 'SLAM' (mapping a room and its obstacles in real-time as a mobile robot equipped with a RealSense camera drives around).

https://www.intel.co.uk/content/www/uk/en/partner/where-to-buy/overview.html

Thisismydigitalself, 2018-11-22:
Thx, i guess I'll wait as i hate being forced to buy what's available and not what I want even tho having IMU is not a required feature right now but might come handy for enhancing manual 360° point-cloud 3D scanning. Let's hope that Intel's click store will offer the D435i to more countries sooner than later.

Thisismydigitalself, 2018-11-25:
@ev-mp 
Thank you very much, some noob's questions if i may:
Why is it that i don't see correlation between RGB and Depth streams? looks like RGB was captured @ 640x480. makes no sense to me. How can i see the combined 3D model of these scans? when I switch to 3D view i see only Depth stream but not the RGB even tho RGB is turned ON.

[image]

Cheers.

UnaNancyOwen, 2018-11-25:
@ev-mp Thanks! 👍

agrunnet, 2018-11-25:
in your screen shot, in the upper left there are drop-down menus. You can select what source (texture) you want to shown on the point cloud. You should see option for color. You should probably also select the same resolution for the color and the depth.

Thisismydigitalself, 2018-11-25:
Thank you, this fixed my 3D view but still, why RGB was captured @ 640x480 (4:3) and Depth @ 1280x720 (16:9). sorry if this is a silly question. i'm a noob.
[image]

agrunnet, 2018-11-25:
I agree that it should have been captured at higher resolution. But the RBG is a separate camera, so it can be captured at any resolution. Also, when you apply post processing with to the depth with subsampling of 2 for example, then the depth is reduced to 640x480, and they match. 
So check out the post-processing knob right above the RGB camera. Check out our white paper on Best-Known-Practices and you will see that we recommend at least 2x down-sampling for most applications.

Thisismydigitalself, 2018-11-25:
Quote from the White Paper: 
**_"The D435 camera is based on the D430 module and **uses a separate RGB camera** attached to a separate stiffener. This choice was based on feedback from previous generations of cameras and modules where some customers wanted the flexibility to choose whatever RGB camera they wanted."_**

Now i am totally confused. i was under the impression that the RGB i see in these bag files are actually from the sensor itself. I guess I need to read some more about the 435. i might be missing the big picture here.

MartyG-RealSense, 2018-11-25:
The D415 model has its RGB sensor integrated on its circuit board.  It is the only circuit board in the 400 Series range that has an integrated RGB sensor.  All the other models (400, 410, 420, 430) do not have an integrated RGB sensor.

So how Intel provides RGB for the cased USB D435 camera model is to attach an RGB sensor separately to a D430 board.

Thisismydigitalself, 2018-11-25:
Thank you Marty. Now I know that I know nothing :). i need more information. Please direct me to more information on how should one scan 360 deg 3D scan with texture using 435 with a seperate rgb camera.  I want to learn how/what warps the 2D texure map on the 3D scaned model when these two are shifted and not perfectly aligned to each other.

Thank you. your help is much aprecieted.

MartyG-RealSense, 2018-11-25:
You are very welcome.  :)

If you want an outdoor camera from the old generation, the RealSense R200 can be used outdoors.  It is used on flying drones.

https://click.intel.com/intel-aero-platform-for-uavs-vision-accessory-kit.html

With the modern 400 Series cameras, the easiest way to do 360 degree scans with a single camera would be to use commercial 3D scanning software.  If you are scanning an object, it could be put on a turntable.  If you are scanning something larger such as the human body, you could walk the camera round the person until you have captured all the data that you need.

Items 3, 10 and 11 in the list of RealSense resources linked to below show 3D scanning software that works with the 400 Series.

https://click.intel.com/intelr-realsensetm-developer-kit-r200-951798.html

https://communities.intel.com/message/564054#564054

If you are interested in the old R200 though, it can be bought still from the Intel store.

dorodnic, 2018-11-26:
To clarify - the RGB was captured in lower resolution to reduce the size of the download for the community.
Our thinking was - D435 part of D435i didn't change, so we didn't want to focus on that aspect of the product.

kartur, 2018-11-26:
Thanks for the samples. I noticed the gyro stream being very choppy. Feels like 5 FPS albeit showing 200 (or 400?) FPS.

I noticed it runs better when I disable the RGB/Depth-Image.

Thus I was wondering. is there already some kind of timestamp alignment going on? From the shown timestamps it doesn't seem like it. Guess it's still work in progress?

[image]

MartyG-RealSense, 2018-11-26:
The SDK supports filtered point clouds via post-processing.

https://communities.intel.com/message/550709#550709

barnjamin, 2018-11-26:
@MartyG-RealSense I'm sorry I meant for the IMU data since its typically fairly noisy

MartyG-RealSense, 2018-11-26:
I don't have any information about that.  One of the Intel guys such as Dorodnic (above) should be able to answer that one.

dorodnic, 2018-11-26:
Hi @kartur 
When playing in the RealSense Viewer it will try to synchronise between all streams. This can be turned off by switching to 3D view tab and disabling the lock icon in the up-right corner. 
@barnjamin - at the moment we are not providing any processing of the IMU data, only the raw data from hardware.

dorodnic, 2018-11-26:
Also, the upcoming 2.17.0 SDK release is going to address several software issues related to the D435i.

barnjamin, 2018-11-26:
Any plans to add something like that?  Or really a synchronized stream of filtered transformations between subsequent video frames derived from IMU data to feed into a reconstruction system.

dorodnic, 2018-11-26:
Hi @barnjamin 
No such plans for now. D435i addresses demand from the community for synchronised depth and IMU data, to be used with 3rd-party SLAM solutions. 
We might offer full-blown 6-dof tracking at some point in the future, but it is currently outside the scope of this product.

MartyG-RealSense, 2018-11-26:
Intel certainly has 6DOF development experience, particularly from the cancelled Project Alloy headset in 2017 (which had a RealSense ZR300 camera in it).  I believe the VR department of Intel was responsible for Alloy rather than the RealSense department?

https://www.hardwarezone.com.sg/feature-intels-project-alloy-first-high-quality-untethered-vr-headset-mixed-reality-experience

I was looking forward to Alloy, so it'd be nice if 6DOF tracking does make a comeback in some form someday, even if it is through RealSense support of third-party 6DOF hardware.  *smiles*

MartyG-RealSense, 2018-11-28:
MeshLab or Blender should be able to convert a pre-recorded .ply point cloud model into a solid mesh.  I believe other 3D modeling software platforms such as Maya and 3DS Max could do it as well.  If you are aiming for a free / low cost budget, MeshLab or Blender (being free) would be the obvious choices of software to consider. 

Of those two, MeshLab is likely to provide the best and easiest conversion of a RealSense-produced .ply, based on user feedback that I have heard.

If you want to go deeper into the subject, the author of a point cloud program called Pyntcloud has recommended looking at the paper linked to below about the subject of converting a point cloud to a mesh.

https://hal.inria.fr/hal-01017700/document

Thisismydigitalself, 2018-11-28:
Thank you, I was asking about capturing the ply. are you saying the "viewer" can also capture?

MartyG-RealSense, 2018-11-28:
It is easy to capture a point cloud ply on the RealSense Viewer.  If you look in the corner of the Viewer window, there are '2D' and '3D' buttons.  If you click on the '3D' button then you enter the point cloud mode of the Viewer.  You can then use the record function to record a point cloud as a ply file.

HippoEug, 2018-11-29:
>  about capturing the ply. are you saying the "viewer" can also capture a 360 deg walk around an object type of scan?

As of now, SDK2.0 does not do a 360 degree walk around capture to fully make a 3D model. You can only capture multiple `.ply` files and either stitch it yourself (ie in Meshlab) or use a Third Party program to do a 360 degree scan. 

The old SDK1.0 used to have this function

> record function to record a point cloud as a ply file.

Yes, but when you play it back it will only show as a recording of a 3D video, not a reconstruction of the object!

Thisismydigitalself, 2018-11-29:
> or use a Third Party program to do a 360 degree scan.
Thank you @HippoEug , my question still stands: which 3rd party software, free or cost-effective, supports D400's cameras? 

It cannot be more obvious, -  i am a newbie :)
All that i know about 3D/360 model capturing is what i read and watched on YT this last week or so. i must say the information out there is so limited that i wonder either this technology is new, Intel is not that good in consumers depth camera offering, or its marketing lacks big time and this product is sololy for developers. all the videos i watched are people **_playing_** with the technology and my finding is that only "Cappasity" with it's cloud base solution is the only solution out there to fully support this camera and my need with this camera. cloud base sw is not my cup of tea. i am willing to pay nicely but in return i want a professional working product. I wonder if my request is too far fetched and i am asking too much and what i need is to go for these multi $$ closed products that can be found on Amazon and such. If Realsense is only for developers then Intel should mention this in any store it is selling this product. the fact that only legacy products are for consumers like myself makes me think that I might be looking in the wrong place for what i am after. These D400 cameras looks great - no complain whatsoever! I was only describing my feeling during this last week where i was trying hard to understand if it will work for me. @MartyG-RealSense told me right at the beggining that this camera might no be for me and he was so right and I understand that now. i still hope i could use these, but as of now i don't know how.

Thank you all for your help.

MartyG-RealSense, 2018-11-29:
Intel are continuously improving their tutorial support for RealSense.  I believe they may have a central location for these tutorials at some point, perhaps similar to the 'Learn' hub-page of the Unity game creation engine's website.

https://unity3d.com/learn

Over at the Intel Support site, Intel are also merging their various support sites together, including their RealSense support forum, into a single location from December 9 2018 to make support easier.

https://forums.intel.com/s/?language=en_US

RealSense overall is not a new product, as it has been around since 2014.  It is fair to say though that it is a developer-focused product, and so a certain amount of pre-existing knowledge about aspects such as Visual Studio is assumed.  This can make RealSense less accessible for beginner hobbyists.   This was perhaps more clearly signaled with the older RealSense cameras that were called Development Kits, whereas the current generation of cameras are marketed as Depth Cameras.

Intel have definitely made progress in making RealSense a 'plug and play' product that can be set up quickly, compared to previous generations of RealSense camera.  Undoubtedly there is still more that can, and will, be done though to improve this accessibility further.

HippoEug, 2018-11-29:
Yes, some people have managed to do full **REAL TIME 3D scanning with the D400 cameras** either using **paid 3rd party applications (afaik: 29 Nov 2018) or applications written by themselves**. As of now, you can do it for free using MatLab and stitch multiple `.ply` files yourself to re-create a 3D object.

Easy3DScan by Cappasity currently does not support D400 Cameras afaik, only the older Realsense SR300s etc. In fact, you could reconstruct a full 3D object with the older cameras and SDK1.0. But again, I don't really see a point in buying a older camera at this point just to do 3D scanning..

Please give RecFusion a try. I think it does exactly what you want, **REAL TIME 3D SCANNING**. It is free to capture (demo version), but to export the full 3D object it is $100USD if I am not wrong. Here is some [personal preview](https://github.com/IntelRealSense/librealsense/issues/2488#issuecomment-438156072). You can go to their site and [download](https://communities.intel.com/external-link.jspa?url=https%3A%2F%2Fwww.recfusion.net%2Findex.php%2Fen%2Fdownload) the single scanning version.

I too am waiting for a function to do a full 3D object scan included in the SDK. Or at least a SDK from another company that supports it (Open3D?)

Thisismydigitalself, 2018-11-29:
Thank you so much @HippoEug  & @MartyG-RealSense for taking the time for helping me and others.
i was not familiar with these two scan tools as i probably did not know what to ask google to look for me so, - I have more reading to do and I like it. I wonder if i should jump to conclusion and ask myself why, if realsense sensors are available since 2014, so few companies supports these? they do support the older versions, but much fewer supports these D400 (not to mention intel's Easy 3D Scan not supporting these). RecFusion looks good but only if pricing as mentioned on thier website is a typo :)  I am preparing a set of questions for them and hopfully their answers will help both parties to close a deal. bottom line is - i feel it's too early for me (even though and because 2014 is now long gone) to jump on this bandwagon. this will change the moment the smokescreen clears.

[image]

MartyG-RealSense, 2018-11-29:
Third-party tool support for the 400 Series started off slowly in the first half of 2018 but began picking up quickly after June.  There are a few different 3D model scanning solutions now, recently-added face tracking in the Nuitrack SDK software (with face recognition coming too), and several rugged tablets with RealSense 400 Series camera options.

The 400 Series still has a while left in its life-cycle (projected to be a three-year one from 2017-2019) and then the next generation of RealSense camera is projected to be introduced in 2020, according to Intel's publicly presented roadmap in 2017.  It is probable that the 2020 generation will be supported by third-parties faster after launch than the 400 Series was initially, because the open-source development ecosystem that was new with the 400 Series is well established now.

Thisismydigitalself, 2018-12-06:
@HippoEug 
Have you tried RecFusion with D415?..
RecFusion looks good but I gave it a try with a D415 and it does not work for me. either my PC is too weak ( Win10/64/SSD/16Gb/Nvidia 750TI/USB 3.1/ or I am doing something wrong. all looks good till i try to reconstruct but the 3D view scrawls like 0.1HZ or so and the result is bad. i also tried recording sequance for post render to no avail.  i sent them a mail. awaits their reply.

shi-gan, 2018-12-07:
Can we use D435i for [VINS-MONO](https://github.com/HKUST-Aerial-Robotics/VINS-Mono)?Will IMU exceed 100Hz?In d435i_walk_around.bag,I found the IMU's data is near 60Hz.

MartyG-RealSense, 2018-12-07:
Details of the Hz rates of the IMU in the D435i can be found in the table below.

[image]

Thisismydigitalself, 2018-12-11:
Hi, I'm In search for either a 3D mesh model of D415 sensor or a hand grip for both holding it Vertically and/or Horizontally. I would also like to know if there is a git or facebook page for the D415.

Thank you.

MartyG-RealSense, 2018-12-11:
CAD models for all 400 Series cameras can be downloaded from the link below:

https://realsense.intel.com/intel-realsense-downloads/#cad

The TinkerCAD site has a mount blueprint for the D415.

https://www.tinkercad.com/things/lAyhkwykfVP-realsense-d415-bumblebee-camera-mount

This GitHub site is the main GitHub for RealSense support.  The other support site, run by Intel Support, is at the link below:

https://forums.intel.com/s/topic/0TO0P00000018NTWAY/intel-realsense-technology?language=en_US

MartyG-RealSense, 2018-12-11:
I'm very glad I could be of help.  Good luck with your scans!

UnaNancyOwen, 2018-12-14:
> We are also working on adding dedicated documentation and tutorials.

Hi @dorodnic,
RealSense D435i arrived today and tried it!
I will touch it for a while, and send feedback. Thanks, 👍 
https://github.com/UnaNancyOwen/RealSense2Sample/tree/master/sample/Motion

By the way, Where is documents and tutorials for IMU?

[14 more replies]

closed 2018-12-28 by RealSense-Customer-Engineering