realsenseai/librealsense #2719: CMake can't generate a library librealsense2

Medo91code, 2018-11-13:
Hi

I must make a project that can detect objects used Intel Realsense D435 .

"the Operating System is windows 8.1 x64 "

I decided that i use Visual Studio 2017 , C++ , Opencv for this project. i can work with Opencv in VS2017 but i can't use Realsense D435 library (librealsense2) with opencv and VS2017 , because CMake can't generate this library . I followed the these Steps "Getting Started" (in this link : https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv#windows ) .

but when i did step 10 , it shows to me these Errors:

CMake Error at C:/Program Files (x86)/CMake/share/cmake-3.13/Modules/ExternalProject.cmake:2329 (message): error: could not find git for clone of libusb Call Stack (most recent call first): C:/Program Files (x86)/CMake/share/cmake-3.13/Modules/ExternalProject.cmake:3105 (_ep_add_download_command) CMakeLists.txt:807 (ExternalProject_Add)

how can i fix that ?

can you please tell me what can i do step by step ?

Thank you very much

HippoEug, 2018-11-14:
Which version of Realsense SDK are you using? I think the new [installer](https://github.com/IntelRealSense/librealsense/releases/download/v2.16.2/Intel.RealSense.SDK-2.16.2.304.exe) works pretty well without the need to use CMake. 

[Source](https://github.com/IntelRealSense/librealsense/issues/2632#issuecomment-433807151)

dorodnic, 2018-11-14:
Hi @Medo91code 
Did you manage to overcome the problem? 
For the record, the issue is related to `BUILD_WITH_TM2` flag. Enabling it requires `git` to be installed on the computer. You can safely disable the flag (for now).

dorodnic, 2018-11-27:
@Medo91code - I assume this was resolved. If you have any additional question please feel free to open additional issues.

closed 2018-11-27 by dorodnic

spaltiel, 2018-12-12:
I have similar issue and I could not resolve it yet.

got the following error (see also attached image):
error: could not find git for clone of libusb
also 
"Performing Test COMPILER_SUPPORTS_CXX0X - Failed"

Do you have any suggestions how to be able to build realsense library without errors?

I use VS2015.

regards,
Amir

[image]

honghande, 2018-12-27:
I have similar issue and I could not resolve it yet.

I use windows 10 64bit.
Visual studio 2015.
latest cmake.

cmake fails to generate the code.

got the following errors (see also attached image):
CMake Error at D:/intel/cmake-3.13.2-win64-x64/share/cmake-3.13/Modules/ExternalProject.cmake:2329 (message):
  error: could not find git for clone of libusb
Call Stack (most recent call first):
  D:/intel/cmake-3.13.2-win64-x64/share/cmake-3.13/Modules/ExternalProject.cmake:3105 (_ep_add_download_command)
  CMake/external_libusb.cmake:4 (ExternalProject_Add)
  CMake/windows_config.cmake:41 (include)
  CMakeLists.txt:16 (os_set_flags)

Do you have any suggestions how to be able to build realsense library without errors?

kind regards,
Amir

galadash, 2019-03-20:
Hi, I encountered this issue as well, and toggling  BUILD_WITH_TM2 did not work for me either.
As many answers and posts tell us that "to support the TM2 (T265) device we need `libusb`" and "you don't really need it in librealsense for now", I assume we can try and work around the issue.

I looked in the files mentioned in the error, and found out that in `CMake/windows_config.cmake`, line 41, was as follows:
`    include(CMake/external_libusb.cmake)`

I read this as a hardcoded way to incorporate the libusb, which we don't need.
I "wrapped" this in an if-statement similar to the one on line 33 of the main `CMakeLists.txt`:

```
    if(BUILD_WITH_TM2)
        include(CMake/external_libusb.cmake)
    endif()
```
This made it possible for me to generate the appropriate files, and continue with the steps [mentioned here](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#windows).

I have been able to generate a pyrealsense2.pyd file, and I have been able to run the examples afterwards... So far everything seems OK.

@dorodnic , can you let me know if I might have "broken" some functionalities by doing this, or whether this might have been a bug? If so, should I create a new issue?

@spaltiel @honghande , can you check if this also solves your issues?

Good luck!

spaltiel, 2019-03-20:
Hi,

If I remember correctly we have needed to do the following:
1. go ahead and install Git - https://git-scm.com/download/win 
Make sure that it will get installed on C:\Program Files 
2. you may need to uncheck BUILD_WITH_TM2 from CMAKE GUI.
now it should work.

BTW It also took us very long time to solve this cmake issues and to be able to generate solutions.
Above was recommended by intel support - It indeed help us.
After doing above we were able to generate C#, and python versions.

Hope it will help you.

Good luck,
A.I.