realsenseai/librealsense #2720: Librealsense on Android: rs2::sensor::set_option(RS2_OPTION_EMITTER_ENABLED, 1) causes exception

xtrawurst, 2018-11-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 Series | 
| Firmware Version                   | any | 
| Operating System & Version |  Android  | 
| SDK Version                            |  2.16.0 | 
| Segment			|  Tablets | 

### Issue Description
Title says most. The method call first takes a very long time, then throws an exception:
`rs2::invalid_value_error, set_xu(id=) failed! `

RealSense-Customer-Engineering, 2018-11-14:
[Realsense Customer Engineering Team Comment]
@xtrawurst What android device are you testing? And please provide the logcat and dmesg logs for the issue. Thanks!

xtrawurst, 2018-11-14:
@RealSense-Customer-Engineering  I'm testing on a Samsung Galaxy Tab S3. How can I obtain the dmesg logs?

xtrawurst, 2018-11-14:
@RealSense-Customer-Engineering 
I've dug a little deeper. The call sequence is
1. `librealsense::uvc_xu_option::set()` (option.h)
2. `libuvc_uvc_device::set_xu()` (libuvc.cpp)
3. `uvc_set_ctrl()` (ctrl.cpp)
4. `libusb_control_transfer()` (libusb sync.c)

In `libusb_control_transfer()` the `transfer->status` becomes `LIBUSB_TRANSFER_STALL`, resulting in a return value of `LIBUSB_ERROR_PIPE` (-9), which then causes the exception to be thrown in `librealsense::uvc_xu_option::set()` (option.h).

RealSense-Customer-Engineering, 2018-11-15:
[Realsense Customer Engineering Team Comment]
@xtrawurst,

is your device rooted? or can your root your android device at first?

1)obtain the dmesg logs:
in the bash:
$   adb root
$   adb shell

in the device's command line interface:

# dmesg > /sdcard/dmesg.txt
# exit

in the bash:

$ adb pull /sdcard/dmesg.txt  .

2) could you built librealsense for android with following command:

cmake .. -DANDROID_ABI=armeabi-v7a -DCMAKE_TOOLCHAIN_FILE=/opt/android-ndk-r16b/build/cmake/android.toolchain.cmake -DBUILD_SHARED_LIBS=false -DFORCE_LIBUVC=off

pls reference to the https://github.com/IntelRealSense/librealsense/issues/2617 accordingly?

Thanks. look forward to your reply.

RealSense-Customer-Engineering, 2018-11-15:
[Realsense Customer Engineering Team Comment]
@xtrawurst,

we recommend that using V4L2_BACKEND rather then LIBUVC_BACKEND.

pls reference the following PR in the community:

https://github.com/IntelRealSense/librealsense/pull/2489

xtrawurst, 2018-11-15:
@RealSense-Customer-Engineering 
To run on V4L2_BACKEND, as you suggested, we would need a patched Android system, no?

RealSense-Customer-Engineering, 2018-11-15:
[Realsense Customer Engineering Team Comment]
@xtrawurst,

yes. pls reference the following PR in the community:
https://github.com/IntelRealSense/librealsense/pull/2489

RealSense-Customer-Engineering, 2018-11-20:
[Realsense Customer Engineering Team Comment]
@xtrawurst,

any progress with this issue?  look forward to your response.

xtrawurst, 2018-11-20:
@RealSense-Customer-Engineering 
We cannot access dmesg on our device and unfortunately the V4L2 backend is not an option either. For now we simply do not make the method call as the emitter is on by default. It just takes away the option of turning it off.

It would be good if you could try to reproduce the issue on your end.

RealSense-Customer-Engineering, 2018-11-20:
[Realsense Customer Engineering Team Comment]
Thanks for your update.

we understood what you concerned.

pls reference to the following issue:

https://github.com/IntelRealSense/librealsense/issues/2746

hope the coming Android backend will meet your requirement.

RealSense-Customer-Engineering, 2018-11-23:
[Realsense Customer Engineering Team Comment]
@xtrawurst,

any progress with this issue?

might be you could reference the following PR in the community:

https://github.com/IntelRealSense/librealsense/pull/2765

we could find there is setEmitter(...) API in this PR.

RealSense-Customer-Engineering, 2018-11-26:
[Realsense Customer Engineering Team Comment]
@xtrawurst,

there is a libirsjni.so and irsa.jar in the PR2765.

and we validated the PR2765 and it works well as expected on RK3288/RK3399 Android board(rooted).

RealSense-Customer-Engineering, 2018-12-04:
[Realsense Customer Engineering Team Comment]
@xtrawurst, any progress with this issue? we'd like to hearing a positive message, Thanks.

xtrawurst, 2018-12-04:
@RealSense-Customer-Engineering 
At this point we are just waiting for the upcoming official support of librealsense for un-rooted Android.

RealSenseCustomerSupport, 2019-02-21:
----------------------------------------------

@xtrawurst Please have a try with https://github.com/IntelRealSense/librealsense/pull/3083 . And please noted it was just merged to development branch and not officially publish yet.

RealSenseCustomerSupport, 2019-02-25:
----------------------------------------------

@xtrawurst Any good news with https://github.com/IntelRealSense/librealsense/pull/3083 ? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

@xtrawurst Unroot android solution now included in 2.19 librealsense. Please have a try with that and looking forward to your update.Thanks!

RealSenseCustomerSupport, 2019-03-11:
----------------------------------------------

@xtrawurst Any update of the testing with 2.19 librealsense? Thanks!

xtrawurst, 2019-03-11:
@RealSense-Customer-Engineering 
Yes: https://github.com/IntelRealSense/librealsense/issues/3385

RealSenseCustomerSupport, 2019-03-21:
----------------------------------------------

@xtrawurst Is it possible for you to try build RealSense AAR from source?

RealSenseCustomerSupport, 2019-03-26:
----------------------------------------------

@xtrawurst Any update?

RealSenseCustomerSupport, 2019-04-04:
----------------------------------------------

@xtrawurst Did you get through the issue? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-04-26:
----------------------------------------------

@xtrawurst Haven't got update for long time. If no other questions, will close the ticket accordingly. Thanks!

closed 2019-05-03 by RealSenseCustomerSupport

xtrawurst, 2019-06-14:
@RealSense-Customer-Engineering This is showing up again when I tried the latest librealsense for (unrooted) Android.

The tested version was 2740bbc.

Tested on a Samsung Galaxy S8 (SM-G950F).

When setting emitter enabled this happens on the log:
```
E librs   : SetXU result:-1
... lots of the same ...
... and finally ...
W librs   : set_xu(id=) failed! Last Error: Broken pipe
```

and then an `rs2::invalid_value_error` exception: `set_xu(id=) failed! Last Error: Broken pipe`

xtrawurst, 2019-06-15:
False alarm. Was using a broken D435.