realsenseai/librealsense #2721: D400 Series on Android: Some librealsense method calls sometimes take a very long time.

xtrawurst, 2018-11-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 Series | 
| Firmware Version                   | any | 
| Operating System & Version |  Android | 
| SDK Version                            | 2.16.0 | 
| Segment			| Tablets | 

### Issue Description
Concretely the methods that can take a very long time (10s of seconds) to execute are:
1. `video_stream_profile::get_intrinsics()`
2. `rs400::advanced_mode::load_json()`

In the case of `load_json()` more often than not the result is an exception and the JSON config is actually not loaded.

RealSense-Customer-Engineering, 2018-11-14:
[Realsense Customer Engineering Team Comment]
@xtrawurst Could you please provide the logcat and dmesg logs for the issue? And which FW version are you using? Thanks!

xtrawurst, 2018-11-14:
@RealSense-Customer-Engineering Using FW 05.10.06. Will post logs shortly.

xtrawurst, 2018-11-14:
W.r.t. the failing loading of the JSON config, the point of failure is in the method `librealsense::ds5_advanced_mode_base::set_laser_power()` (advanced_mode.cpp). An exception is thrown somewhere in the call tree below this method.

RealSense-Customer-Engineering, 2018-11-15:
[Realsense Customer Engineering Team Comment]
Is there any message in dmesg/logcat/exception about this issue?

Would you try to run v4l2 backend for librealsense?

xtrawurst, 2018-11-15:
@RealSense-Customer-Engineering 
> Is there any message in dmesg/logcat/exception about this issue?
I can't access dmesg. I scanned the logcat for any message but couldn't find anything.

> Would you try to run v4l2 backend for librealsense?
Unfortunately this is not an option as it would require a patched Android system.

RealSense-Customer-Engineering, 2018-11-20:
[Realsense Customer Engineering Team Comment]
In my opinion, The v4l2 backend is the best solution at this moment.
I guess, the libuvc isn't stable on your Android system.

In the future, maybe we can find a new solution on Android.

xtrawurst, 2018-11-20:
As the V4L2 backend requires extensive target device modifications this isn't a scalable solution for us.

As for the JSON loading taking a very long time, @dorodnic 's answer on this thread may provide some insight:
https://github.com/IntelRealSense/librealsense/issues/2456

Could it be that something similar is going on in the case of `video_stream_profile::get_intrinsics()`?

RealSense-Customer-Engineering, 2018-11-26:
[Realsense Customer Engineering Team Comment]
Hi @xtrawurst,
On my Android devices, the function get_intrinsics cost about 51ms, the function load_json cost about 310 ms.
I can't reproduce this issue on my device. It seems there is a timeout issue somewhere.

I don't think this 10 seconds issue is the UVC extension unit latency issue.

xtrawurst, 2018-11-26:
@RealSense-Customer-Engineering 
Are you using the libUVC or the V4L backend?

RealSense-Customer-Engineering, 2018-11-26:
[Realsense Customer Engineering Team Comment]
V4L2 backend, right now.
As you know, there are many differences between Android and Android.
I‘ll try libuvc later.

RealSenseCustomerSupport, 2019-02-21:
----------------------------------------------

@xtrawurst Please have a try with https://github.com/IntelRealSense/librealsense/pull/3083 . And please noted it was just merged to development branch and not officially publish yet.

xtrawurst, 2019-02-21:
@RealSense-Customer-Engineering Thanks, will do.

RealSenseCustomerSupport, 2019-02-25:
----------------------------------------------

@xtrawurst Any good news?

xtrawurst, 2019-02-25:
@RealSense-Customer-Engineering 
We're currently a bit busy with other urgent projects but we'll get to it soon. I'll post any updates here.

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

@xtrawurst Unroot android solution now included in 2.19 librealsense. You can have a try with that. Thanks!

RealSenseCustomerSupport, 2019-03-11:
----------------------------------------------

@xtrawurst Any update of testing with 2.19 librealsense?

xtrawurst, 2019-03-11:
@RealSense-Customer-Engineering 
Yes: https://github.com/IntelRealSense/librealsense/issues/3385

RealSenseCustomerSupport, 2019-03-21:
----------------------------------------------

@xtrawurst Is it possible for you to try build RealSense AAR from source? For https://github.com/IntelRealSense/librealsense/issues/3385 we're looking at it also.

RealSenseCustomerSupport, 2019-03-26:
----------------------------------------------

@xtrawurst Any update?

closed 2019-04-08 by RealSenseCustomerSupport