realsenseai/librealsense #2722: windows 10 UWP 怎么使用realsense(D435)?

laolaolulu, 2018-11-14:
How do windows 10 UWP use realsense (D435)?

The driver has been installed. Is there an example of UWP using realsense?

MartyG-RealSense, 2018-11-14:
Microsoft has a GitHub page with a list of UWP sample programs.

https://github.com/Microsoft/Windows-universal-samples/tree/master/Samples

A member of Intel's support team has previously highlighted a UWP sample (not specific to RealSense cameras) called CameraStreamCorrelation, which "shows how to use spatially correlated color and depth cameras and the depth frames to map image pixels from one camera to another using a [DepthCorrelatedCoordinateMapper]".

https://github.com/Microsoft/Windows-universal-samples/tree/master/Samples/CameraStreamCorrelation

laolaolulu, 2018-11-19:
https://github.com/Microsoft/Windows-universal-samples/tree/master/Samples/CameraStreamCorrelation
Can this example be rewritten to c#?

laolaolulu, 2018-11-27:
https://github.com/Microsoft/Windows-universal-samples/tree/master/Samples/CameraStreamCorrelation
cannot execute

[image]

RealSense-Customer-Engineering, 2018-12-07:
[Realsense Customer Engineering Team Comment]
Hello @laolaolulu,

Regarding the issue you are seeing with the UWP app Camera Stream Correlation.  You are seeing an exemption when Visual Studio is attempting compilation, simply:

1 & 2)  Remove the check mark from the box.

3)  Press the play button (Continue) above, and app should work accordingly.

See attached image with detailed view of above instructions.

Another UWP app you can try is the Camera Frames app:  https://github.com/Microsoft/Windows-universal-samples/tree/master/Samples/CameraFrames

Please let us know if you have any other issues.
Thank you!

RealSense-Customer-Engineering, 2018-12-18:
[Realsense Customer Engineering Team Comment]
No response in over a week, submit issue as solved.

closed 2018-12-18 by RealSense-Customer-Engineering