realsenseai/librealsense #2727: RGB Sensor on pyrealsense2

acceptablesimple7, 2018-11-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Operating System & Version |   LInux Ubuntu 16| 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NVIDIA Jetson TX2 L4T 28.2.1 |
| Language                            |  Python  | 
| Segment			|  Robot                           | 

I am using the opencv_viewer_example.py that calls the pyrealsense2 for a project I am working on, however I would like to use the RGB sensor simultaneously and when I do so I get an error that the it is being used

    xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy

I then proceeded to duplicate the camera with V4l2loopback without failing to realize that I was going to run into the same issue because the pyrealsense2 module calls the camera regardless instead of trying to use the duplicates that I have set up. Any clues as to how to keep the module from using the RGB sensor or to make it use one my duplicates?

RealSense-Customer-Engineering, 2018-11-15:
[Realsense Customer Engineering Team Comment]
Hi @acceptablesimple7,

Do you try to use D415's RGB sensor or other RGB sensor on TX2 board?

acceptablesimple7, 2018-11-15:
I am trying to use the RGB sensor from the D415

acceptablesimple7, 2018-11-15:
Was actually able to fix the problem by configuring and enabling only the depth stream in the pipeline arguments

closed 2018-11-15 by acceptablesimple7