realsenseai/librealsense #2728: Tutorial 1 using RGB camera?

kqi914, 2018-11-14:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Operating System & Version |   {Linux (Ubuntu 16)}  | 
| Platform                                 | NVIDIA Jetson TX2  |
| Language                            |  python                          | 

### Issue Description
I wrote a python script using the pyrealsense2 module to gather the depth data based on [Tutorial 1](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-tutorial-1-depth.py).
For example purposes, lets assume I am using the python script from that link. To my understanding, I am not using the RGB camera in this program, so when I try to access the RGB camera to do some object recognition, the terminal says that I cant because it the device is busy.

However, in the SDK it is possible to open both, the RGB camera and the depth camera at the same time. Trying to clone the camera did not fix this issue. What I was thinking is that somehow, the Tutorial 1 script is calling the RGB camera, thus not allowing another process to use it. Anyone know of a way to fix this?

dorodnic, 2018-11-15:
I believe that starting the pipeline with default arguments will try to enable both depth and RGB streams, just in case you need it later. You can, however, explicitly state what streams you need by using the `rs.config` class:
```py
p = rs.pipeline()
c = rs.config()
c.enable_stream(rs.stream.depth)
p.start(c)
```
This should not be locking the RGB camera, if it still does, its a bug.

kqi914, 2018-11-15:
Thanks dorodnic, I will try. On a side note, is there any documentation for pyrealsense2? I could not find any documentation for it, only for pyrealsense. Or are they similar in a way that the documentation for pyrealsense can be used for pyrealsense2?

dorodnic, 2018-11-15:
Hi @kqi914 
Can you add the link to the documentation you are referring to? At the moment we only publish [C++ documentation](https://intelrealsense.github.io/librealsense/doxygen/annotated.html) and the python wrapper is more or less one-to-one translation, but we will be happy to expand the documentation.

kqi914, 2018-11-16:
Your solution worked, thanks!
Here are the two links I found for pyrealsense documentation:
[Pdf](https://media.readthedocs.org/pdf/pyrealsense/dev/pyrealsense.pdf)
[Website](https://pyrealsense.readthedocs.io/en/master/)

closed 2018-11-26 by RealSense-Customer-Engineering