realsenseai/librealsense #2730: The pixel size in different distances to calculate area

shahabe, 2018-11-15:
### Issue Description
I am using D415 RealSense camera and want to calculate the area of an object which I have the bounding box of it.

My problem is that I cannot find the pixel size according to its distance to camera ( trigonometry).
I assume that I can calculate the are by  `(pixelWidth * distance) * (pixelHeight * distance)` for each pixel in my bounding box.
When I use trigonometry i.e. [here](https://stackoverflow.com/a/45481222/6037506), the result is not correct.

How can I calculate the correct pixel size?

HippoEug, 2018-11-15:
Just a quick suggestion: Have you tried [rs-measure.cpp](https://github.com/IntelRealSense/librealsense/blob/master/examples/measure/rs-measure.cpp) to determine if the results are accurate? If you can get the values, divide it by the resolution? 

If it is up to your standard, you can probably just reuse the code.

shahabe, 2018-11-15:
@HippoEug Thank you for your reply.

Yes, I have used the rs_measure.cpp, but it gives the measurement of two points.
What I want is to calculate the pixel area based on the distance to the camera, which means that the closer pixels to camera has smaller area and the far pixels are bigger.

In this case I should put a surface (knowing its size) parallel to the camera (in one constant distance) and calculate the pixel area in that specific distance, and continue this process for different distances, which is very troublesome and may increase the inaccuracy.

The question is, how to calculate the pixel size for different distances based on the intrinsic parameters of the camera.

shahabe, 2018-11-28:
Unfortunately, the result is not accurate and depends on the size of the object and its distance.

How can I use the intrinsic matrix provided by the camera?
It has the f_x and f_y but the definition is:

> Focal length of the image plane, as a multiple of pixel width/height

Actually, I cannot understand what it means!

RealSense-Customer-Engineering, 2019-01-15:
[Realsense Customer Engineering Team Comment]
Hi shahabe,

To convert pixel (x',y') into a 3D coordinate (x,y,z):
https://github.com/IntelRealSense/librealsense/tree/master/examples/measure#converting-between-pixels-and-points-in-3d
Which should help you to obtain needed coordinates for finding a surface/volume of an object...
Closing this, Please reopen if you have further question. Thank you.

closed 2019-01-15 by RealSense-Customer-Engineering