realsenseai/librealsense #2733: MATLAB: Colorize image matrix instead of depth_frame

apoorva2398, 2018-11-15:
How can I colorize a depth image matrix (say, from a recorded tiff or bag file) in MATLAB? The colorize method only takes in a depth_frame. Is there a way to convert an (depth) image to a depth_frame in that case?

lramati, 2018-11-15:
You can read a bag file recorded with the library via the playback
mechanism, which will expose recorded depth frames as proper depth frames
to your program so that a regular colorizer processing block can be used.
While it's not in Matlab, we have a C++ example of this
(examples/record-playback). If you're not reading from a recorded bag file,
a software-device mechanism exists, but has not yet been ported to Matlab.

apoorva2398, 2018-11-15:
Any idea when the software-device mechanism will be available in MATLAB? We have a few datasets in TIFF which we would like to use in MATLAB for now.

lramati, 2018-11-15:
I don't know when the software device will come to Matlab. That said, it
shouldn't be too difficult to write something in C++ to save the TIFF
images via software-device + recorder into bags that Matlab can open. You
can also check the python branch of my fork for an as of yet untested port
of the software-device functionality to Python