realsenseai/librealsense #2734: C# align crushes when using Stream.Depth

papakaliati, 2018-11-16:
| Camera Model                        | D435  | 
| Firmware Version                    | 05.10.06.00 | 
| Operating System & Version  |  Win 10 | 
| SDK Version                            |  2.16.3 ( tested also with 2.16.0   | 
| Language                                |  C#  |

### Issue Description

When using Align with Stream.Depth a **System.AccessViolationException: 'Attempted to read or write protected memory. This is often an indication that other memory is corrupt.' is thrown.**

When using Stream.Color, or any other mode it works without issue, but that's a huge compromise, since you are loosing a big part of the depth stream. 
Results are the same with every version since 2.16.0 I have tried.

papakaliati, 2018-11-16:
Yes, though i still haven't figured out if its possible to view them aligned on the Realsense viewer app.  Additionally I can use the c# wrapper to access both the depth and streams.  And even when using the Align with Stream.Color, the depth image aligns with the color image without any issue. 

The single issue, that cause the application to crush, is when using align with Stream.Depth. But then again that's the use case I am after.

Maltese-Falcon, 2018-11-21:
You have to set color profile with the stream that has alpha channel: for instance, rgba8 or bgra8

bobdavies2000, 2018-12-19:
The support for aligning depth and color under C# is broken.  It works fine in C++.  If you got something working with C#, consider yourself lucky.  I have tried several different ways with 415, 435, and 435i cameras.  The minimalist approach to recreate the problem: just put an Align object in any of the C# examples and change the WaitForFrames() in the main loop to be align_to.Process(pipeline.WaitForFrames()).  Don't even bother trying to use the frames.  It will explode memory use and them time out.  

Here is my hardware configuration (the newest one):

Using device 0, an Intel RealSense D435I
    Serial number: 838212072924
    Firmware version: 05.10.03.00

I downloaded and rebuilt the latest librealsense this morning to make sure this isn't an old problem.  I configure librealsense with C# bindings and OpenCV.  Defaults everywhere else.

I would like to stick with C# so I would appreciate it a lot if this can be fixed.  

Bob Davies

BTW: there are some other easily-avoided problems: 

1) why does librealsense only work in 64-bit?  It probably should work in 32-bit as well.  I don't mind but I was stumped when I first tried a 32-bit application.
2) the OpenCV examples will not compile with the latest version of OpenCV.  Simple changes.  I can give them to you if you need.

ogoshen, 2018-12-31:
@papakaliati just looking at the error message, sounds like you're invoking the aligner on incompatible input. Which is weird because aligning the other way works... maybe one of the frames is missing or disposed too soon?

I've created this [gist](https://gist.github.com/ogoshen/58eb3b74b229d8a47d8c6f2bebc099be) that aligns depth to color, use it to replace the code in the **cs-tutorial-2-capture** sample.

@bobdavies2000 in the procedure you're describing frames won't be disposed properly. The expected behavior in such case, which I can reproduce, is that streaming stops once the frame queue is full (default is 16 frames). No memory leak or excessive usage observed though.

bobdavies2000, 2019-01-11:
I was able to circumvent my problems with align by taking the default
settings exclusively.  This allows the align process to align depth to
color properly.  Here is all the code needed to get the next set of frames:

extern "C" __declspec(dllexport)
int *RS_GetNextFrameSet()  try
{
rs2_error* e = 0;
static rs2::pipeline pipe;
static rs2::config cfg;
static rs2::pipeline_profile profile = pipe.start();
static float depth_scale = get_depth_scale(profile.get_device());
static rs2::align align(RS2_STREAM_COLOR);

static auto sensor = profile.get_device().first<rs2::depth_sensor>();
// Using the align object, we block the application until a frameset is
available
auto frameset = align.process(pipe.wait_for_frames());
{
depthFrame = frameset.get_depth_frame();
depthRGBFrame = c.process(depthFrame);
colorFrame = frameset.get_color_frame();
}
return (int *)colorFrame.get_data();
}

The colorFrame, depthFrame, and depthRGBFrame are defined elsewhere so that
additional interfaces to C# can be used to optionally provide the depth and
depthRGB frames.  This does not allow me to change the resolution to any
setting other than the default.  I have tested D415, D435 and D435i cameras
and it works for all 3.  The only difference is that the default resolution
for D435i is 1280x720 while the others are 640x480.  This is all that I
need right now.

Thanks for the follow up and prompting me to respond to ogoshen's post.

Bob Davies

>

bobdavies2000, 2019-01-12:
I reviewed the changes that made align work in the cs-tutorial-2 example
and the key change was:

using (var original = pipeline.WaitForFrames())
using (var frames = aligner.Process(original))

Where I had written this as:

using (var frames = aligner.Process(pipeline.WaitForFrames() ))

The latter will compile fine and is functionally identical in my mind but
it fails.  Why?

Bob

>

ogoshen, 2019-01-13:
@bobdavies2000 as I've mentioned in my previous comment and is noted [here](https://github.com/IntelRealSense/librealsense/tree/development/wrappers/csharp), you have to explicitly dispose of frames properly.
In the latter case, the implicit `original` frameset isn't disposed and will only be released once the GC kicks in, which can take a long time, by then the internal frame queue would be full.

You can think of it like trying to open 30 files per seconds for reading and not disposing of the FileStreams, eventually the OS will throw an exception for running out of file handles.

I'll be taking a second look at [GC.AddMemoryPressure](https://docs.microsoft.com/en-us/dotnet/api/system.gc.addmemorypressure?view=netframework-4.7.2) which might at least alleviate the problem slightly by helping with the GC scheduling (but isn't supported on all platforms, like Unity\Mono)

RealSenseCustomerSupport, 2019-01-21:
Thanks to ogoshen for the sharing.

bobdavies2000, if nothing else is needed, this will be closed.

Thanks!

closed 2019-02-04 by RealSenseCustomerSupport