realsenseai/librealsense #2735: Unable to import pyrealsense

Teknox1996, 2018-11-16:
|Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version | Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    |  4.15.0-39 
| Platform                                 | Intel Nuc |
| SDK Version                            |  2.16.3 | 
| Language                            | python | 
| Segment			| Robot | 

### Issue Description
Hi,
I'm using ROS Melodic and want to convert 2d coordinates to 3d coordinates. For this reason I want
to use pyrealsense2 but if I try to import realsense on python there is no module named realsense. I also try "pip install pyrealsense"  and get a error message:

`Collecting pyrealsense
  Using cached https://files.pythonhosted.org/packages/6b/08/bdf4ad92bffd0c53d403aa41af012d75315fb14896537c1e04e1c4396f72/pyrealsense-2.2.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-gUWlDA/pyrealsense/setup.py", line 55, in <module>
        language="c++",)])
      File "/home/robot/.local/lib/python2.7/site-packages/Cython/Build/Dependencies.py", line 947, in cythonize
        ctx = c_options.create_context()
      File "/home/robot/.local/lib/python2.7/site-packages/Cython/Compiler/Main.py", line 595, in create_context
        self.cplus, self.language_level, options=self)
      File "/home/robot/.local/lib/python2.7/site-packages/Cython/Compiler/Main.py", line 78, in __init__
        from . import Builtin, CythonScope
      File "/home/robot/.local/lib/python2.7/site-packages/Cython/Compiler/CythonScope.py", line 5, in <module>
        from .UtilityCode import CythonUtilityCode
      File "/home/robot/.local/lib/python2.7/site-packages/Cython/Compiler/UtilityCode.py", line 3, in <module>
        from .TreeFragment import parse_from_strings, StringParseContext
      File "/home/robot/.local/lib/python2.7/site-packages/Cython/Compiler/TreeFragment.py", line 17, in <module>
        from .Visitor import VisitorTransform
      File "Cython/Compiler/Visitor.py", line 17, in init Cython.Compiler.Visitor
      File "/home/robot/.local/lib/python2.7/site-packages/Cython/Compiler/ExprNodes.py", line 46, in <module>
        from .Pythran import (to_pythran, is_pythran_supported_type, is_pythran_supported_operation_type,
    ImportError: cannot import name pythran_is_numpy_func_supported
    
    ----------------------------------------
Command "python setup.py `

When I try "pip install realsense2" it works but I cant also "import realsense2".

Can anyone help me or know a better opportunity to convert 2d coordinates in 3d coordinate?

Teknox1996, 2018-11-16:
I downloaded pyrealsense2 with "pip install pyrealsense2" and get "Requirement already satisfied: pyrealsense2 in /home/robot/.local/lib/python2.7/site-packages (2.16.2.306)" but I when I try to import pyrealsense or pyrealsense2 it dont work

lramati, 2018-11-16:
Can you share the full error?

Teknox1996, 2018-11-16:
I installed cython 28.5 instead  of cython 29 and installed pythran again And now I'm coming on but I get a new error message afterwards:
`Collecting pyrealsense
  Using cached https://files.pythonhosted.org/packages/6b/08/bdf4ad92bffd0c53d403aa41af012d75315fb14896537c1e04e1c4396f72/pyrealsense-2.2.tar.gz
Requirement already satisfied: numpy in /home/robot/.local/lib/python2.7/site-packages (from pyrealsense) (1.15.3)
Requirement already satisfied: cython in /home/robot/.local/lib/python2.7/site-packages (from pyrealsense) (0.28.5)
Requirement already satisfied: pycparser in /home/robot/.local/lib/python2.7/site-packages (from pyrealsense) (2.19)
Requirement already satisfied: six in /home/robot/.local/lib/python2.7/site-packages (from pyrealsense) (1.11.0)
Building wheels for collected packages: pyrealsense
  Running setup.py bdist_wheel for pyrealsense ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-_uutiy/pyrealsense/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-JAYrJJ --python-tag cp27:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-2.7
  creating build/lib.linux-x86_64-2.7/pyrealsense
  copying pyrealsense/extlib.py -> build/lib.linux-x86_64-2.7/pyrealsense
  copying pyrealsense/stream.py -> build/lib.linux-x86_64-2.7/pyrealsense
  copying pyrealsense/extstruct.py -> build/lib.linux-x86_64-2.7/pyrealsense
  copying pyrealsense/offline.py -> build/lib.linux-x86_64-2.7/pyrealsense
  copying pyrealsense/constants.py -> build/lib.linux-x86_64-2.7/pyrealsense
  copying pyrealsense/utils.py -> build/lib.linux-x86_64-2.7/pyrealsense
  copying pyrealsense/core.py -> build/lib.linux-x86_64-2.7/pyrealsense
  copying pyrealsense/__init__.py -> build/lib.linux-x86_64-2.7/pyrealsense
  running build_ext
  building 'pyrealsense.rsutilwrapper' extension
  C compiler: x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-3hk45v/python2.7-2.7.15~rc1=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC
  
  creating build/temp.linux-x86_64-2.7
  creating build/temp.linux-x86_64-2.7/pyrealsense
  compile options: '-Ipyrealsense -I/home/robot/.local/lib/python2.7/site-packages/numpy/core/include -I/usr/local/include/librealsense -I/usr/include/python2.7 -c'
  x86_64-linux-gnu-gcc: pyrealsense/rsutilwrapper.cppx86_64-linux-gnu-gcc: pyrealsense/rsutilwrapperc.cpp
  
  cc1plus: warning: command line option ???-Wstrict-prototypes??? is valid for C/ObjC but not for C++
  pyrealsense/rsutilwrapperc.cpp:3:10: fatal error: rs.h: No such file or directory
   #include "rs.h"
            ^~~~~~
  compilation terminated.
  cc1plus: warning: command line option ???-Wstrict-prototypes??? is valid for C/ObjC but not for C++
  pyrealsense/rsutilwrapperc.cpp:3:10: fatal error: rs.h: No such file or directory
   #include "rs.h"
            ^~~~~~
  compilation terminated.
  error: Command "x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-3hk45v/python2.7-2.7.15~rc1=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -Ipyrealsense -I/home/robot/.local/lib/python2.7/site-packages/numpy/core/include -I/usr/local/include/librealsense -I/usr/include/python2.7 -c pyrealsense/rsutilwrapperc.cpp -o build/temp.linux-x86_64-2.7/pyrealsense/rsutilwrapperc.o" failed with exit status 1
  cc1plus: warning: command line option ???-Wstrict-prototypes??? is valid for C/ObjC but not for C++
  In file included from /home/robot/.local/lib/python2.7/site-packages/numpy/core/include/numpy/ndarraytypes.h:1821:0,
                   from /home/robot/.local/lib/python2.7/site-packages/numpy/core/include/numpy/ndarrayobject.h:18,
                   from /home/robot/.local/lib/python2.7/site-packages/numpy/core/include/numpy/arrayobject.h:4,
                   from pyrealsense/rsutilwrapper.cpp:529:
  /home/robot/.local/lib/python2.7/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
   #warning "Using deprecated NumPy API, disable it by " \
    ^~~~~~~
  pyrealsense/rsutilwrapper.cpp:532:10: fatal error: rs.h: No such file or directory
   #include "rs.h"
            ^~~~~~
  compilation terminated.
  cc1plus: warning: command line option ???-Wstrict-prototypes??? is valid for C/ObjC but not for C++
  In file included from /home/robot/.local/lib/python2.7/site-packages/numpy/core/include/numpy/ndarraytypes.h:1821:0,
                   from /home/robot/.local/lib/python2.7/site-packages/numpy/core/include/numpy/ndarrayobject.h:18,
                   from /home/robot/.local/lib/python2.7/site-packages/numpy/core/include/numpy/arrayobject.h:4,
                   from pyrealsense/rsutilwrapper.cpp:529:
  /home/robot/.local/lib/python2.7/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
   #warning "Using deprecated NumPy API, disable it by " \
    ^~~~~~~
[... cut ...]
    compilation terminated.
    
    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-_uutiy/pyrealsense/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-itU6NJ/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-install-_uutiy/pyrealsense/
`

lramati, 2018-11-16:
You need to install pyrealsense2, not pyrealsense

Teknox1996, 2018-11-16:
I already installed pyrealsense2

RealSense-Customer-Engineering, 2018-11-21:
[Realsense Customer Engineering Team Comment]
Hi Teknox1996,

Please make sure that you are using the right python if you have multiple versions.

Thanks!

closed 2018-12-17 by RealSense-Customer-Engineering

i-chaochen, 2019-07-29:
Hi @RealSense-Customer-Engineering 
```
ImportError: No module named pyrealsense2

```
I have the same issue on the import pyrealsense2. My machine is jetson nano, so I have to install it from the source, and I tried ``` cmake ../ -DBUILD_PYTHON_BINDINGS=TRUE -DPYTHON_EXECUTABLE=/usr/bin/python ``` to bind my python. But still cannot import pyrealsense2. And I noticed from the readme.me this does not support SDK 2.0  yet, does it mean I have to degrade the version to v1.12.1?

```
>[`pyrealsense`](https://github.com/toinsson/pyrealsense) AKA [`pyrealsense/2.0`](https://pypi.python.org/pypi/pyrealsense/2.0) is a community supported Python wrapper for **librealsense v1.12.1**, This wrapper does not support newer versions and **does not work with the RealSense SDK 2.0**.
```

lramati, 2019-08-20:
Please open a new issue so that we can help you get pyrealsense2 imported.
However, you will be pleased to know that SDK2.0 *IS* supported. Compiling from source builds pyrealsense2 v2.XX.X, while the separate pyrealsense v2.0 project also available through PyPi is an out-of-date community project for legacy librealsense.