realsenseai/librealsense #2736: Question about multi cam calibration through picking up points on color frame

LLDavid, 2018-11-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D415 } | 
| Firmware Version                   |5.10.03| 
| Operating System & Version |  win10  | 
| Kernel Version (Linux Only)    |                            | 
| Platform                                 | PC  |
| SDK Version                            |                         | 
| Language                            |  python                        | 
| Segment			|                 | 

### Issue Description
Hi, I am writing  a script for 4 cam external calibration. 
First i used the opencv function to locate the corner points on the chessboard for each cam:
[image]
And project the pixels to the 3d points:<br/>
`depth_points = rs.rs2_deproject_pixel_to_point(depth_intr, pixel, depth_scale)`
` color_points = rs.rs2_transform_point_to_point(depth_to_color_extrin, depth_points)`
<br/>
Then run a simple ICP to get the RT matrix for each cam.
The RT matrix will be applied to points cloud from each cam:
<br/>
`pcd_1 = create_point_cloud_from_rgbd_image(rgbd_image_1, pinhole_camera_intrinsic_1)`
        `pcd_1.transform(RT_Matrix_1to0)`
    `    pcd_2 = create_point_cloud_from_rgbd_image(rgbd_image_2, pinhole_camera_intrinsic_2)`
        `pcd_2.transform(RT_Matrix_2to0)`
       ` pcd_3 = create_point_cloud_from_rgbd_image(rgbd_image_3, pinhole_camera_intrinsic_3)`
       ` pcd_3.transform(RT_Matrix_3to0)`
       ` pcd_4 = create_point_cloud_from_rgbd_image(rgbd_image_4, pinhole_camera_intrinsic_4)`
        `pcd_4.transform(RT_Matrix_4to0)`
<br/>
But the rendering result is still a mess:
[image]<br/>
Clearly some step is wrong.
Here are a few questions I have that might be related:

> 1. should I use the pixel on the 'aligned dept frame' or just 'depth frame' for deprojection?
> 2. the 3d points used for ICP should be the depth point or the color point? I am kind of confused here.

Anyone has any idea? Thanks in advance

RealSense-Customer-Engineering, 2018-11-26:
[Realsense Customer Engineering Team Comment]
I think you can do calibration using IR(left) image between two different cameras. This is a classical stereo calibration problem.
we can get the extrinsics between color and depth/IR from video_stream_profile::get_extrinsics_to.

LLDavid, 2018-11-30:
> [Realsense Customer Engineering Team Comment]
> I think you can do calibration using IR(left) image between two different cameras. This is a classical stereo calibration problem.
> we can get the extrinsics between color and depth/IR from video_stream_profile::get_extrinsics_to

Thanks for the reply. I think the problem is I did not use the left imager as the origin

closed 2018-11-30 by LLDavid