realsenseai/librealsense #2738: Android - Color frames timing out

Furasheru, 2018-11-17:
**Camera Model**:  D415 
**Firmware Version**: 05.10.06.00
**Operating System & Version**: Android 7.1.1
**Platform**: Android
**SDK Version**: 2.16.3
**Language**: C++
**Segment**: Smartphone

I compiled librealsense with Android NDK r18b according to [this article](https://github.com/IntelRealSense/librealsense/blob/master/doc/android/AndroidNativeSamples.md) and ran **rs-color** and **rs-depth** examples - both worked fine. Then I built my own binary based on **rs-color** code, and found that frames time out very often. This seems to be more of a problem for color frames, when I tried depth-only stream I got frames more consistently. I tried both C++ and C interfaces. Here's the source code I tried last:
```C++
#include <librealsense2/rs.h>
#include <librealsense2/h/rs_pipeline.h>
#include <librealsense2/h/rs_frame.h>

#include <cstdlib>
#include <cstdint>
#include <cstdio>
#include <cstring>
#include <stdexcept>

class RealsenseCamera
{
    /* Checks if error occured, wraps it into exception and throws.
    Set cleanup_ to true to cleanup before throwing (for constructors) */
    void handle_error(rs2_error* e, bool cleanup_ = false)
    {
        if (e)
        {
            char buff[256];
            std::memset(buff, 0, sizeof(buff));
            snprintf(buff, sizeof(buff)-1,
                "rs_error was raised when calling %s(%s): %s",
                rs2_get_failed_function(e),
                rs2_get_failed_args(e),
                rs2_get_error_message(e));
            if (cleanup_) cleanup();
            throw std::runtime_error(buff);
        }
    }

    void print_device_info(rs2_device* dev)
    {
        rs2_error* e = 0;
        printf("\nUsing device 0, an %s\n", rs2_get_device_info(dev, RS2_CAMERA_INFO_NAME, &e));
        handle_error(e, true);
        printf("    Serial number: %s\n", rs2_get_device_info(dev, RS2_CAMERA_INFO_SERIAL_NUMBER, &e));
        handle_error(e, true);
        printf("    Firmware version: %s\n\n", rs2_get_device_info(dev, RS2_CAMERA_INFO_FIRMWARE_VERSION, &e));
        handle_error(e, true);
    }

    void cleanup()
    {
        if (profile) rs2_delete_pipeline_profile(profile);
        if (cfg) rs2_delete_config(cfg);
        if (pipe) rs2_delete_pipeline(pipe);
        if (cam) rs2_delete_device(cam);
        if (devices) rs2_delete_device_list(devices);
        if (ctx) rs2_delete_context(ctx);
    }
public:
    RealsenseCamera()
    {
        rs2_error* e;

        rs2_log_to_console(rs2_log_severity::RS2_LOG_SEVERITY_DEBUG, &e);

        rs2_context* ctx = rs2_create_context(RS2_API_VERSION, &e);
        handle_error(e, true);

[... 124 lines cut ...]
 17/11 12:34:15,339 DEBUG [-454559456] (sync.cpp:258) SYNC (TS: (FN: I Depth I Infrared I Infrared ) I Color )--> Color 69 13140.200000

 17/11 12:34:15,339 DEBUG [-454559456] (sync.cpp:516) fps 15 Color 69 13140.200000
 17/11 12:34:15,339 DEBUG [-454559456] (sync.cpp:529) (TS: (FN: I Depth I Infrared I Infrared ) I Color )Color 69 13140.200000 fps 15 gap 66.666664 next_expected: 13206.866664
 17/11 12:34:15,340 DEBUG [-454559456] (sync.cpp:258) SYNC (TS: )--> Color 69 13140.200000

 17/11 12:34:15,340 DEBUG [-454559456] (sync.cpp:516) fps 15 Color 69 13140.200000
 17/11 12:34:15,340 DEBUG [-454559456] (sync.cpp:529) (TS: )Color 69 13140.200000 fps 15 gap 66.666664 next_expected: 13206.866664
 17/11 12:34:15,340 DEBUG [-454559456] (archive.cpp:287) CallbackFinished,Color,0,DispatchedAt,1542447255340.806885
 17/11 12:34:15,341 DEBUG [-454559456] (archive.cpp:294) Frame Callback [Color#0] overdue. (Duration: 1542447255340.806885ms, FPS: 15, Max Duration: 62ms)
 17/11 12:34:15,342 DEBUG [-454559456] (syncer-processing-block.cpp:28) SYNCED: Color 69, 13140.200000
 17/11 12:34:15,342 DEBUG [-454559456] (archive.cpp:501) CallbackStarted,Color,69,DispatchedAt,1542447255342.449951
Got composite frame (1 subframes)
 17/11 12:34:15,342 DEBUG [-454559456] (archive.cpp:287) CallbackFinished,Color,0,DispatchedAt,1542447255342.654053
 17/11 12:34:15,342 DEBUG [-396999372] (archive.cpp:287) CallbackFinished,Color,0,DispatchedAt,1542447255342.775879
 17/11 12:34:15,343 DEBUG [-396999372] (archive.cpp:294) Frame Callback [Color#0] overdue. (Duration: 1542447255342.775879ms, FPS: 15, Max Duration: 62ms)
 17/11 12:34:15,344 INFO [-396999372] (archive.cpp:354) The user was holding on to 1 frames after stream 0x0xe6fa1090 stopped
 17/11 12:34:17,272 DEBUG [-396999372] (archive.cpp:287) CallbackFinished,Color,69,DispatchedAt,1542447257272.753906
 17/11 12:34:17,273 DEBUG [-396999372] (archive.cpp:294) Frame Callback [Color#69] overdue. (Duration: 1934.420898ms, FPS: 15, Max Duration: 62ms)
 17/11 12:34:17,273 INFO [-396999372] (archive.cpp:364) All frames from stream 0x0xe6fa1090 are now released by the user
Device closed.
```
Out of 10 runs about 3 or 4 receive frames, while others time out. With depth frames it's more like 8 or 9 out of 10, but still not reliable enough.

Do you have any ideas what might be the issue? Thank you.

RealSense-Customer-Engineering, 2018-11-20:
[Realsense Customer Engineering Team Comment]
Hi Furasheru,

pls reference the following PR in the community:

https://github.com/IntelRealSense/librealsense/pull/2489

we recommend that using V4L2_BACKEND rather then LIBUVC_BACKEND with Android device. built librealsense for android with following command:

cmake .. -DANDROID_ABI=armeabi-v7a -DCMAKE_TOOLCHAIN_FILE=/opt/android-ndk-r16b/build/cmake/android.toolchain.cmake -DBUILD_SHARED_LIBS=false -DFORCE_LIBUVC=off

RealSense-Customer-Engineering, 2018-11-23:
[Realsense Customer Engineering Team Comment]
Hi Furasheru,

any progress with this issue?

might be you could reference the following PR in the community:

https://github.com/IntelRealSense/librealsense/pull/2765

RealSense-Customer-Engineering, 2018-12-04:
[Realsense Customer Engineering Team Comment]
Hi Furasheru,
any progress with this issue? we'd like to get a positive response from you. Thanks.

RealSense-Customer-Engineering, 2018-12-13:
[Realsense Customer Engineering Team Comment]
@Furasheru Do you still have issue? If not, we'll close the ticket in a few days. Thanks!

Furasheru, 2018-12-13:
We've done a lot of testing, and concluded that this was a hardware malfunction of a USB controller. We've since switched to another device and had no problems so far.

Thanks for your help.

RealSense-Customer-Engineering, 2018-12-14:
[Realsense Customer Engineering Team Comment]
Hi Furasheru,

did you try V4L2 backend?

We had been validated D415 on some Android based board and phone and it worked very well with librs V4L2 backend. so I don't think color frames timing out issue is caused by hardware malfunction.

there are some unobvious bug with the default libUVC backend in librealsense on Android system, I think that's the root cause of color frames timing out.  appreciate too much if you could try V4L2 backend.

Thanks.

adraffy, 2018-12-19:
How do I use the V4L2 backend?

I tried changing `set(FORCE_LIBUVC ON)` inside of `android_config.cmake` but Android Studio fails to compile with a bunch of errors and warnings.

RealSense-Customer-Engineering, 2018-12-20:
[Realsense Customer Engineering Team Comment]
@Furasheru As pointed in earlier reply, please refer to the PR below. Need other file changes than just turn off the LIBUVC. Please have a try. Thanks!

https://github.com/IntelRealSense/librealsense/pull/2489/files

RealSense-Customer-Engineering, 2019-01-03:
[Realsense Customer Engineering Team Comment]
@Furasheru How about the progress? Please update. Thanks!

RealSense-Customer-Engineering, 2019-01-08:
[Realsense Customer Engineering Team Comment]
@Furasheru Any update? Looking forward to your reply. Thanks!

closed 2019-01-09 by RealSense-Customer-Engineering