realsenseai/librealsense #2740: Projecting point to pixel on an infrared color image

noamzafrir, 2018-11-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | (05.10.06.00 ) | 
| Operating System & Version |   {Win (8.1/10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { 2.16.1 }                          | 
| Language                            |  {C }                          | 
| Segment			|  {others }                          | 

### Issue Description
Hello,

I am trying to project a point to a pixel on a color image generated from an infrared stream. The projection is not accurate (has some offset)
When I do this on a color image generated from an RGB image, it works perfectly.

I am doing the projection like this:

**Infrared:**
rs2::video_stream_profile ir_stream_profile = m_pipeline_profile.get_stream(RS2_STREAM_INFRARED).as<rs2::video_stream_profile>();
	m_ir_intrin = ir_stream_profile.get_intrinsics();
...
rs2_project_point_to_pixel(to_pixel, &m_ir_intrin, from_point);

**RGB:**
rs2::video_stream_profile rgb_stream_profile = m_pipeline_profile.get_stream(RS2_STREAM_COLOR).as<rs2::video_stream_profile>();
m_rgb_intrin = rgb_stream_profile.get_intrinsics();
...
rs2_project_point_to_pixel(to_pixel, &m_rgb_intrin, from_point);

When I am giving the point an offset of 27.5mm in the x-axis and then do the projection with the RGB intrinsic, the projection is perfect.
27.5mm is the center between the left and right imagers, and is probably the origin of the infrared image.
This is how I do it with the offset:
from_point[XX] = from_point[XX] - 0.0275;
rs2_project_point_to_pixel(to_pixel, &m_rgb_intrin, from_point);

The calibration file look fine to me (see attached).
[F739112060251F.txt](https://github.com/IntelRealSense/librealsense/files/2592917/F739112060251F.txt)

What could be the reason the projection with the IR intrinsic is not working?

Thank you,
Noam.

RealSense-Customer-Engineering, 2019-01-11:
[Realsense Customer Engineering Team Comment]
Hi moamzafrir,

Wonder if you made any progress.
There is a document regarding to projecting point to pixel. It would help if you have not read through yet. It provides some examples and explanations.
https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0

Thanks!

noamzafrir, 2019-01-13:
We have been able to overcome this issue.
Thank you for the helpful info in the attached link.,

RealSenseCustomerSupport, 2019-01-21:
Hi moamzafrir,

Glad to hear that it helped and you got the issue solved.

Thanks!

closed 2019-02-25 by dorodnic