realsenseai/librealsense #2741: D435 disconnection issues

jasoncatchpole, 2018-11-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version | Windows 10 | 
| Kernel Version (Linux Only)    | NA                                         | 
| Platform                                 | PC -  Intel Core i7-4770, Nvidia 2070 RTX |
| SDK Version                            |  2.16.3                          | 
| Language                            |  C++      | 
| Segment			|  Kiosk application                          | 

### Issue Description
Hi there,

We have been building a kiosk application for a large customer and are switching from Kinect V2 to Intel Realsense. In our application the application needs to run for at least a day (12 hours straight), and ideally we should be able to run a Kiosk over multiple days without having to restart it. We have been having issues with the realsense randomly disconnecting, and also getting into a state where we restart the machine and the camera is in a bad state (even the Realsense viewer application can't initialise it). We've gone through a phase of trying to solve these issues, some of which include the following:

1. Updating the camera firmware (we noted in the [camera firmware release notes](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense-D400-Series-Spec-Update.pdf) that this type of thing has been an issue in the past)
2. Updated SDK version
3. Disabled USB suspend and other USB related Windows power settings
4. Attempt to perform a hardware_reset() on the device if we think it has gotten into this bad state while we are attempting to initialise the device as described [here](https://github.com/IntelRealSense/librealsense/issues/2161). 

There have been quite a few posts along these same lines that we have looked at [here](https://github.com/IntelRealSense/librealsense/issues/1615), [here](https://communities.intel.com/thread/128605) and [here](https://communities.intel.com/message/560278#560278)

We've been using the stock Intel cable but nothing we have tried as of yet has helped us in sorting the issue. We are becoming concerned as the delivery date for this customer project looms.
Are there any additional logs we can look at to determine why the device is failing to be initialised?

HippoEug, 2018-11-19:
What are you using to cool the camera? Do you think it's a safety cut off feature kicking in? There is a similar [issue](https://github.com/IntelRealSense/librealsense/issues/2612) here.

jasoncatchpole, 2018-11-19:
Thanks for the reply. I don't think its heat related, we have the system operating in an air conditioned room. In fact in some of our fresh testing tweaking all the USB related settings we could find we've managed to get it running for a couple of days straight in our most recent test so that is looking promising. 

Now we are just down to an issue where the camera isn't detected when the system first boots up but I suspect this may be due to the application attempting to initialise the device perhaps before the OS has completed some of its operations. I'm investigating that now. I'll report back here and potentially close this issue if we resolve all our problems

closed 2018-12-09 by jasoncatchpole

Dontla, 2019-09-18:
Hey, guy! Have you solved this problem?

tispratik, 2019-11-01:
We found the issue. The issue is related to the specs of the USB cable. RealSense is rated at 2 Amps and most of the USB cables on the market are rated for 0.9 Amps. The moment we switched to a cable rated at 3 Amps, the issue went away. We have stress tested it with the new cables and we couldn't break the system, so far it's been robust. One more thing we did is to make sure to not have to use extension cables, so we have custom ordered 3 Amp USB C to USB A cables from Newnex for our needs.

Some good cables to look for: 
https://www.amazon.com/gp/product/B076CV7MPB
https://www.newnex.com/usb-type-c-cables-legacy.php (Look for power delivery 3 A)

Don't order general purpose cables like CableMatters or AmazonBasics, examples below:
https://www.amazon.com/gp/product/B00DMFB5OK
https://www.amazon.com/gp/product/B01GGKYR2O

Dontla, 2019-11-25:
Does this problem only exist in windows platform?