realsenseai/librealsense #2743: Unable to Stop Realsense Pipeline! [C++]

HippoEug, 2018-11-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version | Win 10 | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC |
| SDK Version                            | 2.16.1 | 
| Language                            | C++  | 
| Segment			| others                   | 

# Introduction
Hi! I am trying to convert/split a `.bag` file into `.csv` and `.png` for further data analytics. However, I am unable to stop the pipeline when the conversion is complete. I borrowed most of the conversion code from `rs-convert.exe`.

# Goal
For example, given a `.bag` file, in this case `record.bag`, I am able to produce `realsense_Color_7.png`, `realsense_Depth_5.png`, and most importantly `realsense_Depth_5.csv`.

[image]

# Issue
After conversion, I have confirmed the  `realsense_Color_7.png`, `realsense_Depth_5.png`, and `realsense_Depth_5.csv` opens just fine, but the pipeline refuses to close, resulting in a `Not Responding Program` since my conversion function is written as a blocking one.

#### Conversion Function Code:
```
if (convert_bag_button) {
	rs2::config convertConfig;
	std::vector<std::shared_ptr<rs2::tools::converter::converter_base>> convertToCSV;
	std::vector<std::shared_ptr<rs2::tools::converter::converter_base>> convertToPNG;

	convertToCSV.push_back(std::make_shared<rs2::tools::converter::converter_csv>("realsense"));
	convertToPNG.push_back(std::make_shared < rs2::tools::converter::converter_png>("realsense"));
	convertConfig.enable_device_from_file("record.bag");

	pipe->stop();
	pipe->start(convertConfig);
	auto device = pipe->get_active_profile().get_device();
	rs2::playback playback = device.as<rs2::playback>();
	playback.set_real_time(false);
	auto duration = playback.get_duration();
	int progress = 0;
	int frameNumber = 0ULL;

	while (true) {
		auto frameset_c = pipe->wait_for_frames();

		if (frameset_c[0].get_frame_number() == 5 || frameset_c[0].get_frame_number() == 6 || frameset_c[0].get_frame_number() == 7) { // If end of frame, bag loops again, hence get_frame_number resets to 1 and is smaller
			std::for_each(convertToCSV.begin(), convertToCSV.end(), [&frameset_c](std::shared_ptr<rs2::tools::converter::converter_base>& converter) {
				converter->convert(frameset_c);
			});
			std::cout << "\nCONVERTING CSV" << std::endl;
			std::for_each(convertToCSV.begin(), convertToCSV.end(), [](std::shared_ptr<rs2::tools::converter::converter_base>& converter) {
				converter->wait();
			});
			std::cout << "CONVERTING PICTURES" << std::endl;
			std::for_each(convertToPNG.begin(), convertToPNG.end(), [&frameset_c](std::shared_ptr<rs2::tools::converter::converter_base>& converter) {
				converter->convert(frameset_c);
			});
			std::for_each(convertToPNG.begin(), convertToPNG.end(), [](std::shared_ptr<rs2::tools::converter::converter_base>& converter) {
				converter->wait();
			});
			break;
		}

		frameNumber = frameset_c[0].get_frame_number();
	}

	std::cout << "\nDONE!" << std::endl << std::endl; // EXECUTES

	pipe->stop();
	std::cout << "HERE" << std::endl; // DOES NOT EXECUTE
	pipe->start(standardConfig);
			
	convert_bag_button = false;
}
```

[image]

# Reason
I would like to stop the pipeline because the pipeline is with the configuration `convertConfig.enable_device_from_file("record.bag");`, and after this conversion I would like to restart the pipeline with a configuration for streaming Color and/or Depth (`pipe->start(standardConfig);`)

# End
I have tested with just CSV conversion without PNG conversion, with the same results. The pipeline would not stop. Any ideas on how I would fix this is appreciated. Many thanks!

HippoEug, 2018-11-28:
Hello @RealSense-Customer-Engineering , any updates on what I could change to fix this? Many thanks!! 👍 😺

Tuebel, 2019-01-04:
I have experienced the same issue. The problem seems to be this line:
`playback.set_real_time(false);`
If i use `playback.set_real_time(true);` the pipeline stops as expected.

Tuebel, 2019-01-04:
I have found an workaround, simply call 
```cpp
pipe->start();
pipe->stop();
```

HippoEug, 2019-01-07:
Hello @Tuebel , i will give this a shot. Thank you very much!!

HippoEug, 2019-01-07:
This works @Tuebel, thank you very much!! Any idea why this solves it?

RealSense-Customer-Engineering, 2019-01-11:
[Realsense Customer Engineering Team Comment]
HI HippoEug,

Is the workaround suggested by Tuebel working for you?

Thanks!

HippoEug, 2019-01-11:
Hi @RealSense-Customer-Engineering , the workaround works. However, it would be nice to have an explanation of why simply changing this parameter fixes this issue. Is this a bug? Good to know in case there are any side effects (which I have not discovered so far)

mcasl, 2019-01-29:
Similar behavior using Python in this related issue:
https://github.com/IntelRealSense/librealsense/issues/3126

RealSenseCustomerSupport, 2019-01-31:
----------------------------------------------

HI HippoEug,

Wonder if there is any update?

Thanks!

RealSenseCustomerSupport, 2019-02-06:
----------------------------------------------

Hi HippoEug,

Wonder if you have any any update or still need help on this one?

Thanks!

closed 2019-02-26 by RealSenseCustomerSupport

soarwing52, 2019-04-25:
@RealSense-Customer-Engineering @RealSenseCustomerSupport 
I'm using the latest python wrapper but still get the freeze situation.
I can either set_real_time(True) which will skip frames

or enable_device_from_file(filename,False)
and
try:
except RuntimeError:

only this can stop the pipeline and keep the program running
which either is not so ideal.