realsenseai/librealsense #2744: High level of noise on depth images

fwindolf, 2018-11-19:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 (D435, D415) | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |  Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.15.0-36.39~16.04.1-generic 4.15.18 | 
| Platform                                 | PC  |
| SDK Version                            |  2.16.4                        | 
| Language                            |  C                        | 

We made a switch for our application from ASUS Xtion sensors to the Realsense D400 series, using two D415 and one D435 camera for creating 3D models of persons.

After having massive problems with oversmooth models, we tested the depth accuracy and noise behaviour of the D400 vs the Xtion. Compared to the realtively steady depth images of the Xtion, the D400s output extremely noisy images. 

We followed the depth *Best Known Methods for Tuning Intel RealSense Depth Cameras* guide to no avail, using the high accuracy preset with minor modifications on depth table, RSM and census. But even in a stationary setting (which we do not have in our application), post processing was not able to reduce the noise level to anything remotely close to what the Xtion can provide.

Do you have any suggestions on:
* how to reduce noise on the depth images without applying temporal filters?
* how to increase the density without getting interpolation  from fg to bg on large jumps? 

Thanks!

MartyG-RealSense, 2018-11-19:
ASUS Xtion and RealSense 400 Series do not use the same type of sensing.  Xtion (which I believe was based on PrimeSense technology that Kinect 1 used) is a structured light camera.  The RealSense 400 Series uses Stereo type imaging, with a left and right IR imager.  So it is difficult to make direct technical comparisons between results on the Kinect / Xtion and those gained from the D415 / D435.

Also, if the subject that you are scanning is not moving around too fast, it may be advisable to replace the D435 with a third D415.  It is good practice to use all of the same type of camera in a multiple camera setup instead of mixing D415s and D435s.  Also, the D415 has less RMS error (depth noise) over distance.

If you have fluorescent lights such as ceiling strip-lights in the location that you are scanning in, this may also have a detrimental impact on the depth image.  This disruption can be reduced by running the cameras at an FPS closer to the operating frequency of the particular fluorescent lights in that location.  For some people it is 30 FPS, and for others it is 60 FPS.

fwindolf, 2018-11-19:
Thanks for the quick reply.
I understand that there is no direct comparison possible, but after reconstructing (with using gaussian noise prior during integration of frames) there is a huge difference. The Xtion models show very fine detail, where the D400 models tend to be oversmooth with fine details lost. Apparently I am not the only one having this issue:
https://3dscanexpert.com/3d-scan-sensor-shootout-realsense-d415-vs-sr300-vs-orbbec-astra-s/

>  if the subject that you are scanning is not moving around too fast, it may be advisable to replace the D435 with a third D415

It is not, however we do not need super detailed reconstruction of the area the D435 covers, but its FOV is very beneficial to us.

> If you have fluorescent lights such as ceiling strip-lights in the location that you are scanning in, this may also have a detrimental impact on the depth image.

Can you tell us what exactly the effect of fluorescent lights is? Does the stereo matching work worse than with natural light? Or is it more that the depth gets jittery?

MartyG-RealSense, 2018-11-19:
The heated gas inside fluorescent lights causes the light to flicker at a frequency that is hard to see with the human eye and gets picked up on by the camera.

Depth images can also get disrupted if the camera is placed too close to a strong light source such as a ceiling bulb light, manifesting a large white area on the image.

Larger FOVs can be set up by positioning cameras so that their fields of view overlap.  This can also provide more comprehensive depth data.  The more cameras you have, the less blind-spots there are (less holes in the data) and the advantage of having data redundancy (more than one camera capturing the same area).

The 400 Series cameras have the advantage of working better in strong light, whereas in other camera models, strong light usually makes the image worse.

https://communities.intel.com/message/567632#567632

closed 2019-01-10 by RealSense-Customer-Engineering