realsenseai/librealsense #2745: pyrealsense point cloud has no color

Anvom, 2018-11-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

LLDavid, 2018-11-19:
What is the point cloud look like?
I used to have a scaling issue, and the point clouds all look like the same color

LLDavid, 2018-11-20:
@olgn you can try to set a clipping distance for the depth ([ref](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py)). It worked for me.

Anvom, 2018-11-20:
@LLDavid  The point cloud lool like this:
[image]

I am very new to this topic and since the color is more important for me it was the first thing needed to be fixed at the moment. I havent tried to solve any issues concerning the scaling yet. It is still pretty bad.

LLDavid, 2018-11-20:
@Anvom It looks like  not the scaling's prblem. If the input color frame is all right, have you tried other visualizer? like open3D

Anvom, 2018-11-21:
@LLDavid  Thank you. In Open3D I get a colorized point cloud.