realsenseai/librealsense #2746: Using RealSense on Android without rooting

ssakhavi, 2018-11-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 14/16/17)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 |   |
| SDK Version                            |  {  }                          | 
| Language                            |  {Java}                          | 
| Segment			|  {Smartphone }                          | 

### Issue Description
Hi,

Are you planning for the device to be used for Android devices without the need for rooting?

dorodnic, 2018-11-20:
Yes

dorodnic, 2019-02-04:
Please follow #3083 - it is fairly mature at this point

closed 2019-02-04 by dorodnic