realsenseai/librealsense #2747: Cannot access /sys/class/video4linux

bluesky1141, 2018-11-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 5.10.6 | 
| Operating System & Version |   Linux (Ubuntu 16.04.5 LTS)  | 
| Kernel Version (Linux Only)    |  (4.15.0.39)                                         | 
| Platform                                 | laptop   |
| SDK Version                            |  librealsense 2.0                         | 
| Language                            |                         | 
| Segment			|                           | 

### Issue Description
<Describe your issue / question / feature request / etc..>
1) I ran  'rs-enumerate-devices ' on the terminal and a got this error. 

 20/11 10:26:57,330 ERROR [140083742844736] (backend-v4l2.cpp:518) Cannot access /sys/class/video4linux
No device detected. Is it plugged in?

The depth camera is actually connected. 

2) Why do I get the following error when I try to install librealsense on 'https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md' ?

Patched kernels modules were created successfully

Replacing videodev :
Applying the patched module ... modprobe: ERROR: could not insert 'videodev': Required key not available
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version
modprobe: ERROR: could not insert 'videodev': Required key not available

please help me....

dorodnic, 2018-11-20:
Hi @bluesky1141 
The two issues are related - there is no `/sys/class/video4linux` because there was a problem when trying to insert the `uvcvideo` kernel-module using `modprobe`.
I think @ev-mp might have more insight.
For now, you still have the option to build the SDK with CMake flag `-DFORCE_LIBUVC=true` and work with the camera that way.

dorodnic, 2018-11-20:
Sorry, clicked the wrong button

ev-mp, 2018-11-20:
@bluesky1141 , can you post the kernel& installation logs for review?
The appropriate patches script for Ubuntu 16.04.5 is `./scripts/patch-realsense-ubuntu-lts.sh`
Additionally, please check that you've followed the guide and got Uubntu updated to the latest version.

Alternatively, we recommend to set the required kernel patches is by [installing DKMS debian package](https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md#installing-the-packages):
`sudo apt-get install librealsense2-dkms` - this automates most of the burden of following the installation guide, and also automatically reapplies the patches when updating Ubuntu kernel.

jav-solo, 2018-12-21:
I have had a similar issue. Instead of reinstalling Ubuntu, I changed the kernel. I was previously using 4.15.0-43, and I would get this error when trying to see output from the D415.

To do this, I uninstalled all librealsense packages using:
 `dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

Because of an issue related to UEFI secure boot I figure out that I had to go to comment out the line:
`GRUB_HIDDEN_TIMEOUT=0` by placing a "#" in front in the file /etc/default/grub

I then rebooted in GRUB and using the advanced options for Ubuntu I changed the kernel to 4.15.0-42

I reinstalled librealsense using:
`sudo apt-get install liberealsense2-dkms` and `sudo apt-get install liberealsense2-utils`

After that my D415 worked using `realsense-viewer`

RealSense-Customer-Engineering, 2019-01-10:
[Realsense Customer Engineering Team Comment]
Hi bluesky1141,

Wonder if you have followed the suggestions above and got any updates?
Do you still need any help? If not, this one will be closed.

Thanks!

closed 2019-02-18 by bluesky1141

MartyG-RealSense, 2023-03-27:
Hi @jeojeda  Which RealSense SDK version are you using, please?  Official support for D405 was introduced in SDK version **2.51.1**.

jeojeda, 2023-03-28:
hi @MartyG-RealSense Hi have the 2.53.1
[image]

MartyG-RealSense, 2023-03-28:
@jeojeda  Regarding **GRUB_HIDDEN_TIMEOUT=0**, it looks as though the value can be defined by opening an Ubuntu file called **/etc/default/grub**

jeojeda, 2023-03-28:
@MartyG-RealSense I did that part int the grub file. But I still have the same problem with the camera. I still get the
(backend-v4l2.cpp:555) Cannot access /sys/class/video4linux
message

MartyG-RealSense, 2023-03-28:
@jeojeda  Which method did you use to install librealsense?  Did you install from packages with apt-get or build it from source code?

Is the camera detected if you use the instruction `sudo apt-get install realsense-uvcvideo`

jeojeda, 2023-03-28:
I am using a laptop PC.I've just followed this steps:
https://github.com/IntelRealSense/librealsense/blob/ros2debian/doc/distribution_linux.md
and https://github.com/IntelRealSense/librealsense/blob/ros2debian/doc/installation.md