realsenseai/librealsense #2748: license notice

Gallimathias, 2018-11-20:
For topical reasons. I just saw Intel using the Apache license. It would be nice to rename the file COPYING to NOTICE as it is provided in the Apache license. Also to do justice to all licenses in the COPYING file.

dorodnic, 2018-11-20:
Hi @Gallimathias 
Let me look into it. Do you have any link on the subject? In different open-source projects I saw different practices, be it COPYING, Attributions file, or something else. 
Of course we will to do our absolute best to ensure proper attributions and compliance.

Gallimathias, 2018-11-20:
You can see that NOTICE has the advantage that you have to copy it and pass it on. However, the COPYING file is absolutely voluntary and will not be considered.

dorodnic, 2018-11-21:
Thank you, we are looking into the issue

dorodnic, 2018-11-22:
Approved, thank you for raising it

Gallimathias, 2018-11-27:
No Problem ;)

closed 2018-11-27 by Gallimathias