realsenseai/librealsense #2749: Problems in Reading Out the Recorded Intel RealSense .Bag File Video for D435 Camera

XIAY0009, 2018-11-20:
I would like to play the recorded .bag video on a simple window (not as complicated as the RealSense example). 
However I am unable to do so, please see my setup and code below: 

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |I have updated to the newest firmware | 
| Operating System & Version | Ubuntu 16.04 | 
| Platform                                 | PC |
| SDK Version                            | 2 | 
| Language                            | C++  | 
| Segment			| Camera  | 

using namespace std; 

int main()
{

  // Create a simple OpenGL window for rendering:
  //window app(1280, 720, "RealSense Video Playback");

  // Declare two textures on the GPU, one for color and one for depth
  texture depth_image, color_image;

  // Declare depth colorizer for pretty visualization of depth data
  rs2::colorizer color_map;

  rs2::config cfg;
  cfg.enable_device_from_file("color.bag");
  rs2::pipeline pipe;
  pipe.start(cfg);
  while (true)
  {   
    rs2::frameset frames;
    if (pipe.poll_for_frames(&frames))
    {   
     //use frames here
     }   
  }   
  pipe.stop();  
 
 return 0;  

}    

Your help is much appreciated. Thanks!

Best,
Yingyu

MartyG-RealSense, 2018-11-20:
There is a rosbag reader program contributed by a community member that has an MIT licence, making it free to use and adapt.  The download link for the zip file is at the bottom of the opening message on the link below.

https://github.com/IntelRealSense/librealsense/issues/2215

dorodnic, 2018-11-21:
Hi @XIAY0009 
Please checkout the [record-playback](https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback) example, it shows both recording and playback flows.
What specifically is going wrong in your application? I can't see anything immediately wrong with the code

HippoEug, 2018-11-22:
I think the problem is here:

```
rs2::config cfg;
cfg.enable_device_from_file("color.bag");
rs2::pipeline pipe;
pipe.start(cfg);
while (true)
{
    rs2::frameset frames;
    if (pipe.poll_for_frames(&frames))
    {
        //use frames here
    }
}
pipe.stop();
```

Between `pipe.start(cfg)` and `pipe.stop()`, it is too short and nothing can be recorded, hence when playing back the `.bag` file you will get an error. There are 2 easy solutions:

### 1
```
rs2::config cfg;
cfg.enable_device_from_file("color.bag");
rs2::pipeline pipe;
pipe.start(cfg);
std::this_thread::sleep_for(std::chrono::milliseconds(5000)); // Records for 5 seconds
pipe.stop();
```

### 2
```
rs2::config cfg;
cfg.enable_device_from_file("color.bag");
rs2::pipeline pipe;
pipe.start(cfg);

int i = 0;
while (i < 500)
{
    rs2::frameset frames;
    if (pipe.poll_for_frames(&frames))
    {
        // Code to stream frames here. Use OpenCV or OpenGL etc
    }
}
pipe.stop();
```

XIAY0009, 2018-11-25:
@HippoEug Thanks for your reply, I actually managed to record the .bag file. However, the problem is that I do not know how to play it out and display in the window. The code which I shared on 21st November is an attempt to play the recorded file color.bag. Your help is much appreciated. 

In fact I even tried to study ros within the week, and tried to play it with rosbag, but I have a fatal error. Your help is much appreciated. 

Thanks
Best,
Yingyu

HippoEug, 2018-11-26:
I will try to post a code when I can. I think the parts to strip out are anything that you see `ImGui::`, those removes the buttons and sliders and text from `rs-record-playback.cpp`

HippoEug, 2018-11-26:
Basically striped down code is like this:

```
pipe->stop(); // Stop streaming with default configuration
pipe = std::make_shared<rs2::pipeline>();
rs2::config cfg;
cfg.enable_device_from_file("a.bag");
pipe->start(cfg); //File will be opened in read mode at this point
device = pipe->get_active_profile().get_device();

rs2::playback playback = device.as<rs2::playback>();
if (pipe->poll_for_frames(&frames)) // Check if new frames are ready
{
    depth = color_map.process(frames.get_depth_frame()); // Find and colorize the depth data for rendering
}

pipe->stop(); // Stop streaming with default configuration
pipe = std::make_shared<rs2::pipeline>();
rs2::config cfg;
cfg.enable_device_from_file("a.bag");
pipe->start(cfg); //File will be opened in read mode at this point
device = pipe->get_active_profile().get_device();
```

You might have to add some for or while loop conditions in the place where you poll for frames, dependent on what you want out of the display

XIAY0009, 2018-11-26:
@HippoEug Thanks so much, this helps. 
Now I could record and playback successfully. 

However, when I tried to record both color camera and depth camera data simultaneously from the RealSense camera, I am unable to do so, and I am not sure why, below is my code for recording. 

Thanks for your help

Best,
Yingyu 

rs2::config cfg1;
cfg1.enable_stream(RS2_STREAM_COLOR, -1, 0, 0, RS2_FORMAT_RGB8, 30);
cfg1.enable_record_to_file("color.bag");
cfg1.enable_stream(RS2_STREAM_DEPTH, -1, 0, 0, RS2_FORMAT_Z16, 30);     
cfg1.enable_record_to_file("depth.bag");

XIAY0009, 2018-11-26:
I managed to solve it :) Basically I should not save it as two different .bag files, I should only save it as one. In order to play both the RGB and depth videos, I did the following and I am able to play both!! 

rs2::frame depth = color_map.process(data.get_depth_frame());
rs2::frame color = data.get_color_frame();

depth_image.render(depth, { 0,               0, app.width() / 2, app.height() }); 
color_image.render(color, { app.width() / 2, 0, app.width() / 2, app.height() });

HippoEug, 2018-11-27:
That is great news! If there is no more questions regarding this particular issue, you can close it! 👍

closed 2018-11-27 by XIAY0009