realsenseai/librealsense #2750: Unable to save recorded data from realsense-viewer to a custom folder

jasjuang, 2018-11-20:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400} | 
| Firmware Version                   | 5.10.6 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC |
| SDK Version                            | 2.16.5 | 
| Language                            |  C++                         | 

### Issue Description
I was trying to use the realsense-viewer and then I realize there isn't a way for me to save the recorded files into a custom folder, after further investigation, I realized it's being saved to `C:\User\Documents` as indicated in https://github.com/IntelRealSense/librealsense/blob/master/common/model-views.cpp#L4940, is there a way to save to a custom folder?

MartyG-RealSense, 2018-11-20:
At this time, the only path that the recorded files will save to from the RealSense Viewer is the user documents folder, unfortunately.

https://communities.intel.com/message/559390#559390

HippoEug, 2018-11-21:
Another way could be using the C++ example, `rs-record-playback.cpp` and changing the directory from there

dorodnic, 2018-11-21:
Hi @jasjuang 
Totally agreed we need to make this configurable.

RealSense-Customer-Engineering, 2019-01-08:
[Realsense Customer Engineering Team Comment]
Hi jasjuang ,

Besides the workaround suggested by HippoEug would work, you may check back our github later or choose to subscribe LibRealSense to get an update on the progress of customizing saving directory for bag files.
Subscribe LRS

I am closing this one for now.
Thank you!

closed 2019-03-20 by dorodnic