realsenseai/librealsense #2751: How to build a python3 development environment in win7

pursuityu, 2018-11-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         | python3.6 D435 win7                                                               |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  win7  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  python3 }                          | 
| Segment			|  others }                          | 

### Issue Description
How to build a python3 development environment in win7

dorodnic, 2018-11-21:
Follow the regular instructions under [python / build from source / windows](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#windows).
If you are building it on Win7, everything should work. If you are building on Win10 for deployment to Win7, there is an extra CMake flag you need to enable - `-DBUILD_FOR_WIN7=true`

pursuityu, 2018-11-21:
i just want it can work on python3,do you have this .pyd file.

pursuityu, 2018-11-21:
i just want it can work on python3,do you have this .pyd file.when i install sdk, i can find .pyd which it can use in python2.7

dorodnic, 2018-11-21:
Did you try loading `pyrealsense2.cp36-win_amd64.pyd` instead? 
(simplest way to test is to copy it to new folder, rename to `pyrealsense2.pyd` and try running `import pyrealsense2` in python

pursuityu, 2018-11-22:
import pyrealsense2 has no problem， but it happend a problem 
Process finished with exit code -1073740940 (0xC0000374)

pursuityu, 2018-11-26:
i have solved this problem.just copy Intel.Realsense.dll，pyrealsense2.cp36-win_amd64.pyd，realsense2.dll into the pyrealsense2.

RealSense-Customer-Engineering, 2018-11-28:
[Realsense Customer Engineering Team Comment]
Hi pursuityu,

Can we close this one if there is nothing else?

Thanks!

closed 2018-11-28 by pursuityu