realsenseai/librealsense #2752: SR300 Expected Point Cloud Distortion

randyyork123, 2018-11-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300
| Firmware Version                   | 3.21.0.0
| Operating System & Version |   Win 10
| Kernel Version (Linux Only)    |  N/A
| Platform                                 | PC  |
| SDK Version                            | 2 or Legacy
| Language                            |  python
| Segment			|  Other 

### Issue Description
Our research group has been using a number of SR300s. I realize there have been many cases of SR300s being uncalibrated and producing highly distorted point clouds and depth maps (such as https://github.com/IntelRealSense/librealsense/issues/565). 

We are having a hard time determining what is reasonable distortion expected. Some of our cameras are a bit old, so we weren't able to exchange them before with the Click folks. But we also recently purchased a new batch of SR300s to compare against and they don't seem to be quite right either. This is having a significant impact on our research.

I post below a few screenshots of a large, flat board about 30-50 cm away and the distortions we see in the Viewer utility. (I can also post the saved .ply files if it helps.) We are hoping to find out if we are missing something. We are grateful for any thoughts!

1. This one seems to be the worst, especially as a flat surface shows up as a bit twisted with severe curling at the edges:
[image]

2. This one has some curling at the edges:
[image]

3. This one seems to be the best out of the batch, but the color stream refuses to display anything (which I think is related to https://github.com/IntelRealSense/librealsense/issues/151)!
[image]

Thanks!

MartyG-RealSense, 2018-11-21:
One of the Intel guys such as Dorodnic can likely provide a better answer on this question than I can.  It is worth looking at the information about how RealSense SDK 2.0 treats the SR300 model compared to the 400 Series models though (such as color images having no distortion applied on SR300).

https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#sr300

dorodnic, 2018-11-21:
Hello @randyyork123 @MartyG-RealSense 
I actually don't have an answer for this, unfortunately, but the issue will automatically get escalated to RealSense support. You can also submit additional ticket on [realsensesupport.intel.com](https://realsensesupport.intel.com) to speed things up.

MartyG-RealSense, 2018-11-21:
Randy's mention of curling prompted a memory of a previous case a year ago of curling with the SR300.  That user posted about it both On Intel Support and the GitHub.

https://communities.intel.com/message/490996#490996

https://github.com/IntelRealSense/librealsense/issues/565

randyyork123, 2018-11-21:
Thank you both very much. My colleague is going to send a ticket to the support folks and report back here.

@MartyG-RealSense, that's exactly right! That is what tipped me off, as linked in the original post.

MartyG-RealSense, 2018-11-21:
@randyyork123   I see the GitHub link now in your original post.  Apologies for not recognizing it was the same as the GitHub link I provided!

RealSense-Customer-Engineering, 2018-11-28:
[Realsense Customer Engineering Team Comment]
@randyyork123 Please use "rs-enumerate -c" to get the calibration data for both good camera and bad camera. We'll do further analysis based on the calibration data.

https://github.com/IntelRealSense/librealsense/tree/master/tools/enumerate-devices

RealSense-Customer-Engineering, 2018-12-05:
[Realsense Customer Engineering Team Comment]
@randyyork123  Could you please update about the issue? Thanks!

RealSense-Customer-Engineering, 2018-12-17:
[Realsense Customer Engineering Team Comment]
@randyyork123  How is it going about the issue? Looking forward to your update. Thanks!

RealSense-Customer-Engineering, 2018-12-24:
[Realsense Customer Engineering Team Comment]
@randyyork123  Didn't get update from you yet. Will close the ticket at this point. If you still need help, please re-open it. Thanks!

closed 2018-12-24 by RealSense-Customer-Engineering