realsenseai/librealsense #2754: How to connect two D415 devices?

HungLearner, 2018-11-21:
| Camera Model                       | D400  | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version | Win (10) | 
| Platform                                 | PC |
| SDK Version                            | 2.16.1 | 
| Language                            | C# | 

### Issue Description
I need to connect two depth camera devices on the same computer.

But the other one camera frames cannot receive.

[image]

HungLearner, 2018-11-22:
Let me try again~
Thanks!

realsens01, 2018-11-27:
I have a similar problem with a pair of D435s. I have not tried v2.16.5, but I have tried adding a separate PCIe USB port and using a powered AmazonBasics hub (as described in the multi-cam whitepaper). More details here: https://communities.intel.com/message/594911

realsens01, 2018-11-29:
I just want to clarify that the reason for adding the PCIe USB card was to test the sort of bandwidth hypothesis that @dorodnic mentioned (i.e., to provide a second USB hub, since Windows Device Manager indicated that all native USB ports used the same hub). I initially thought that this had mitigated or solved the problem, but it seems to be happening more frequently now. It is also my impression -- based on the multi-camera whitepaper -- that bandwidth should not be an issue for just 2 cameras.

It's also worth mentioning that the cameras do sometimes disconnect, or they fail to be detected, so I have not ruled power out as a factor.

In both cases, the issues are not especially consistent, and are therefore challenging to debug.

realsens01, 2018-12-03:
Note: Another user reported the same problem on my Intel Community post (https://communities.intel.com/message/596048#596048), and noted a possible connection to camera metadata.

RealSense-Customer-Engineering, 2018-12-17:
[Realsense Customer Engineering Team Comment]
Hi @realsense01, @hunglearner,

I use the realsense-viewer based on 2.17.0 to run RGB (RGB8 format) + Depth steams at 1280x720/30FPS with two D415 cameras more than 5 minutes on Win10 RS3.

My setup as follows:
Two D415 cameras are connected to two system USB3 ports instead of an external USB3 hub. The firmware is 5.10.13.0 and metadata is enabled.

By the way, how did you verify if the metadata is enabled?

realsens01, 2018-12-17:
Thanks @RealSense-Customer-Engineering . 

* The other user reported disabling metadata by  _running script "realsense_metadata_win10.ps1" with "-op remove_all"_. The user did not report whether or not the effect was verified.
* I followed the [procedure for enabling metadata support on Windows](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md). I verified that metadata was present using the Python library, but it's also visible in the display of Realsense Viewer (iirc, it replaces _System time_ with _Hardware time_ on each stream display). I ran the script with the remove_all option to disable metadata, and verified that it was disabled using the Realsense Viewer display. I did this at least twice.
* I will try 2.17.0, but I'm skeptical that that is the issue.
* Do the USB ports you used go to separate internal hubs? I believe you can check this using the Windows Device Manager.
* Did the recorded files have approximately the same number of depth and color frames?

Thanks again.

HungLearner, 2018-12-25:
> Hi @HungLearner
> Did you try with the newer [v2.16.5](https://github.com/IntelRealSense/librealsense/releases/tag/v2.16.5)?
> If you still see the issue, this might be more related to USB bandwidth rather than power.
> (Power would result in disconnect, not pause in frames). You can try using different USB ports (if you have several root-hubs on your motherboards, for example front and back)

Hello, i have a new question about sdk.
Today, i use 2.17.0 library in the C#, my output have many messages.
Can it close by any function?

RealSense-Customer-Engineering, 2018-12-26:
[Realsense Customer Engineering Team Comment]
Hi HungLearner,

Which C# sample app did you use? Did you also enable Metadata?
Can you run 'usbview' tool and dump to a log file? Like to see your USB connection status.

RealSense-Customer-Engineering, 2019-01-03:
[Realsense Customer Engineering Team Comment]
Hi @realsense01,

From your USBview file, one of D435 USB devices is not enumerated normally. As you can see below, some info is missing. Did you have the chance to try another PC or different OS like Ubuntu to see if the same issue "No Frames Received!" there during two D435 streams with metadata?

Device Power State:               PowerDeviceD0
       ---===>Device Information<===---
*!*ERROR:  No String Descriptor for index 2!
ConnectionStatus:
Current Config Value:              0x01  -> Device Bus Speed: SuperSpeed
Device Address:                    0x1D
idVendor:                        0x8086 = Intel Corporation
idProduct:                       0x0B07
bcdDevice:                       0x50A6
iManufacturer:                     0x01
*!*ERROR:  No String Descriptor for index 1!
iProduct:                          0x02
*!*ERROR:  No String Descriptor for index 2!
iSerialNumber:                     0x03
*!*ERROR:  No String Descriptor for index 3!
bNumConfigurations:                0x01

realsens01, 2019-01-08:
This error disappears (reliably) if Realsense Viewer is active before plugging the cameras into the USB ports. This does not, however, solve the "No frames" error. Files attached (for the case in which Realsense Viewer was active before plugging the cameras in). I believe everything looks normal in these files, but the error occurred after about 2 minutes. Further explanation here: https://forums.intel.com/s/question/0D70P000006B1abSAC

Regarding trying another PC or OS: Are you suggesting that this would help to debug the problem?

[USBViewAll-both_cameras.zip](https://github.com/IntelRealSense/librealsense/files/2735205/USBViewAll-both_cameras.zip)

realsens01, 2019-01-08:
Another observation: I just looked inside a short bag file that I recorded yesterday, right after metadata was enabled. Only one camera. There was no frame error while recording, but there are multiple frame drops in the file. There is one gap that is 0.4s long. Should I expect to see dropped frames frequently? Even for one camera in a standard configuration on a standard computer? Is the "No frames received" error just an indication that dropped frames have passed a threshold?

realsens01, 2019-01-16:
This problem might be related to another problem that we are now experiencing. I've provided detailed information about that problem here: https://github.com/IntelRealSense/librealsense/issues/2216#issuecomment-454654209. That problem occurs in Realsense Viewer, but not when .bag files are recorded via a Python script (even with metadata enabled). Preliminary tests seem to indicate that the same might be true of the "No frames received" error described here. This suggests that the problem is with Realsense Viewer, and not the SDK, the drivers, the OS, or the hardware.

RealSenseCustomerSupport, 2019-04-02:
----------------------------------------------

Please try the latest firmware and librealsense. Not observe this issue. I will close this. If the issue is still there, you can create another ticket for track.

realsens01, 2019-04-02:
Can you please explain the context of "Not observe this issue"? Are you referring to the "No frames" error? What kind of system was tested? Was metadata enabled? What version of firmware / software was tested? Which camera model?

I still consider this to be an issue. I will try the latest firmware and software as soon as possible, and report back here.

closed 2019-04-03 by HungLearner

Lowpassfilter, 2019-08-09:
hi all, I'm new to realsense. May I ask how to enable two devices at the same time in C#? As I understand, I have to create two pipelines. But what is the correct grammer to do it?

Lowpassfilter, 2019-08-09:
Do I need to enable a device using the serial number?