realsenseai/librealsense #2758: Program on Mac OSX Runs, but not on Ubuntu

philipjames44, 2018-11-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Operating System & Version |   Ubuntu 16 / MacOS  | 
| Kernel Version (Linux Only)    |  4.19                                         | 
| Platform                                 | Intel NUC  |
| Language                            |  C++                          | 

My below code works completely fine on MacOSX, however on Ubuntu it gets stuck repeating "nah" meaning it does not receive any data. If I run realsense-viewer, I can stream an record both depth and color without issue. So I am not quite sure what's going on here.

One thing, however, that occurs when I run realsense-viewer on Ubuntu, but not mac, is that I get this warning:
(I'm not sure if it matters though as the viewer can still stream and record data regardless)
```
21/11 15:10:34,713 WARNING [139669595444992] (backend-v4l2.cpp:1248) Pixel format 435a4e49-b602-480f-978c-e4e88ae likely requires patch for fourcc code INZC!
 21/11 15:10:34,714 WARNING [139669595444992] (backend-v4l2.cpp:1248) Pixel format 52494150-8536-4841-b6bf-8fc6ffb likely requires patch for fourcc code PAIR!
```

Code for reference:
```
#include "opencv2/opencv.hpp"
#include <librealsense2/rs.hpp>
#include <sstream>
#include <iostream>
#include <iomanip>
#include <chrono>
#include <string>
using namespace std;
using namespace cv;

typedef std::chrono::high_resolution_clock Clock;

int main(){
	using namespace cv;
	
	const auto window_name1 = "Depth Image";
 	const auto window_name2 = "Color Image";
  	
	namedWindow(window_name1, WINDOW_AUTOSIZE);
	namedWindow(window_name2, WINDOW_AUTOSIZE);
	
	// Declare frameset and frames which will hold the data from the camera
	rs2::frameset frames;
	rs2::frame depth;
	
	// Declare depth colorizer for pretty visualization of depth data
	rs2::colorizer color_map;	
	color_map.set_option(RS2_OPTION_COLOR_SCHEME, 3.f); //Black to White
	
	// Create a shared pointer to a pipeline
	auto pipe = std::make_shared<rs2::pipeline>();
	pipe->start();

	auto pipe2 = std::make_shared<rs2::pipeline>();
	
	// Initialize a shared pointer to a device with the current device on the pipeline
    	rs2::device device = pipe->get_active_profile().get_device();

	pipe->stop(); // Stop the pipeline with the default configuration
	
	//clock for timer
	auto t1 = Clock::now();
	int oldSeconds = 0;
	int seconds = 0;
	int minutes = 0;
	int desiredVideoLengths = 15; //minute lengths of video sequences 
	string filename = "1.bag";
	int fileNum = 1;

	pipe = std::make_shared<rs2::pipeline>();
	pipe2 = std::make_shared<rs2::pipeline>();
   	rs2::config cfg; // Declare a new configuration

	auto depth_sensor = device.first<rs2::depth_sensor>();
        depth_sensor.set_option(RS2_OPTION_VISUAL_PRESET, RS2_SR300_VISUAL_PRESET_SHORT_RANGE);
 
	
	cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_RGB8, 30);
	cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
	cfg.enable_record_to_file(filename);
 	pipe->start(cfg); //File will be opened at this point
	device = pipe->get_active_profile().get_device();	

//	auto depth_sensor = device.first<rs2::depth_sensor>();
//	depth_sensor.set_option(RS2_OPTION_VISUAL_PRESET, RS2_SR300_VISUAL_PRESET_SHORT_RANGE);	

	while(true){
		rs2::frameset data;
		pipe->poll_for_frames(&data);

		if(!data){
			cout << "nah" << endl;
		}
		else{
	        	rs2::frame depth = data.get_depth_frame().apply_filter(color_map);
          		rs2::frame color = data.get_color_frame();
  
          	//	cout << "Depth Frame: " << depth.get_frame_number() << endl;
          	//	cout << "Color Frame: " << color.get_frame_number() << endl;
  
          		// Query depth frame size (width and height)
          		const int wD = depth.as<rs2::video_frame>().get_width();
          		const int hD = depth.as<rs2::video_frame>().get_height();
  
          		// Query depth frame size (width and height)
         		const int wC = color.as<rs2::video_frame>().get_width();
         		const int hC = color.as<rs2::video_frame>().get_height();
  
         	 	// Create OpenCV matrix of size (w,h) from the colorized depth data
         		Mat depthImage(Size(wD, hD), CV_8UC3, (void*)depth.get_data(), Mat::AUTO_STEP);
  
      	 	   // Create OpenCV matrix of size (w,h) from the colorized depth data
        		Mat colorImage(Size(wC, hC), CV_8UC3, (void*)color.get_data(), Mat::AUTO_STEP);
  
       		   // Update the window with new data
        		imshow(window_name1, depthImage);
  
        	  // Update the window with new data
	        	imshow(window_name2, colorImage);
			waitKey(20);
		}

		auto t2 = Clock::now();
		seconds = std::chrono::duration_cast<std::chrono::seconds>(t2 - t1).count();
						
		if(seconds != oldSeconds){
			if(seconds % 60 == 0)
				minutes++;
			cout << seconds / 60 << " minutes or " << endl;			
			cout << seconds << " seconds have elapsed " << endl;
			oldSeconds = seconds;
			
			//update filename
			if(minutes % desiredVideoLengths == 0 && minutes != 0){
				minutes = 0;
				fileNum++;
				filename = to_string(fileNum) + ".bag";
				
				pipe->stop();	
				cfg.enable_record_to_file(filename);	
				pipe->start(cfg); //File will be opened at this point
		         	device = pipe->get_active_profile().get_device();
				cout << "----------------------------------------------------------" << endl;
                                cout << "----------------------------------------------------------" << endl;
			}
		}		
				
	}	
	
	pipe->stop();
 	return EXIT_SUCCESS;
}
```

dorodnic, 2018-11-21:
I can find nothing immediately wrong with the code. Also the warnings are unrelated (they are about two formats that exposed by the device but not used by librealsense). What about the built-in examples like `rs-capture` and `rs-align`? 
Did you install the SDK from apt or built from source? 
Just to rule out more of the setup, you can build the SDK with the CMake flag `-DFORCE_LIBUVC=true`. This will essentially give you the same exact code as on the Mac, so it will rule out any V4L2 and Kernel related shenanigans.
Also, did you try the code without enabling recording?

philipjames44, 2018-11-21:
@dorodnic 
both capture and align work properly. I installed the sdk using https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

However I think I made some progress. Using the code below I can get color streams to stream and record. I can get depth to stream and record for around a second before a segmentation fault. Occasionally, it will run indefinitely, but it seems to be nondeterministic. However, if I apply a color map, it will say frames received no data every time the executable is run. Thoughts? (Pardon the terrible indentation)

```
  #include "opencv2/opencv.hpp"
   #include <iostream>
   #include <librealsense2/rs.hpp>
   using namespace std;
   using namespace cv;
  
   int main(){
   
  using namespace cv;
  const auto window_name = "Display Image";
  namedWindow(window_name, WINDOW_AUTOSIZE);
  
  rs2::config cfg;
  cfg.enable_record_to_file("kms.bag");
	        cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_RGB8, 30);

      cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);

 
 
//cfg.enable_device_from_file("/Users/phil/Documents/20181008_182015.bag");
  rs2::pipeline pipe;
  pipe.start(cfg); //File will be opened in read mode at this point
 
rs2::colorizer color_map;
        color_map.set_option(RS2_OPTION_COLOR_SCHEME, 3.f); //Black to White
 
  
  int num = 0;
  while(true && num < 1000000){
          num++;
   	cout << num << endl;
	       rs2::frameset data;// = pipe.wait_for_frames();
        //  pipe.poll_for_frames(&data);
       // rs2::frame depth = data.first(RS2_STREAM_COLOR);//data.get_color_frame();//.apply_filter(color_map); 
//	if(!depth){
  //                cout << "nah" << endl;
    //      }
          if(pipe.poll_for_frames(&data)){
	rs2::frame depth = data.first(RS2_STREAM_DEPTH);
	depth.get_data();
          // Query frame size (width and height)
          const int w = depth.as<rs2::video_frame>().get_width();
          const int h = depth.as<rs2::video_frame>().get_height();
          // Create OpenCV matrix of size (w,h) from the colorized depth data
          Mat image(Size(w, h), CV_8UC3, (void*)depth.get_data(), Mat::AUTO_STEP);
          // Update the window with new data
          imshow(window_name, image);
          waitKey(20);
          }
  }
  
 pipe.stop();
 return EXIT_SUCCESS;
      }

```

dorodnic, 2018-11-21:
Very strange. I will try to reproduce when I get the chance. What happens if you comment out the opencv lines in the code?

philipjames44, 2018-11-21:
@dorodnic 
The issue gets even stranger! In the more recent code sample, it records when all opencv is removed. The original program, however, does not record when opencv is commented out. Perhaps related, the depth info that the second program records is a bit strange. It seems the depth is almost cutting in and out very very quickly.

HippoEug, 2018-11-22:
Hi @philipjames44,
I only tested your first code in the original question, didn't test on the second one. I tested on VS2017, Windows 10.

#### Your Original Code in Original Question 
```
#include "opencv2/opencv.hpp"
#include <librealsense2/rs.hpp>
#include <sstream>
#include <iostream>
#include <iomanip>
#include <chrono>
#include <string>
using namespace std;
using namespace cv;

typedef std::chrono::high_resolution_clock Clock;

int main() {
	using namespace cv;

	const auto window_name1 = "Depth Image";
	const auto window_name2 = "Color Image";

	namedWindow(window_name1, WINDOW_AUTOSIZE);
	namedWindow(window_name2, WINDOW_AUTOSIZE);

	// Declare frameset and frames which will hold the data from the camera
	rs2::frameset frames;
	rs2::frame depth;

	// Declare depth colorizer for pretty visualization of depth data
	rs2::colorizer color_map;
	color_map.set_option(RS2_OPTION_COLOR_SCHEME, 3.f); //Black to White

	// Create a shared pointer to a pipeline
	auto pipe = std::make_shared<rs2::pipeline>();
	pipe->start();

	auto pipe2 = std::make_shared<rs2::pipeline>();

	// Initialize a shared pointer to a device with the current device on the pipeline
	rs2::device device = pipe->get_active_profile().get_device();

	pipe->stop(); // Stop the pipeline with the default configuration

	//clock for timer
	auto t1 = Clock::now();
	int oldSeconds = 0;
	int seconds = 0;
	int minutes = 0;
	int desiredVideoLengths = 15; //minute lengths of video sequences 
	string filename = "1.bag";
	int fileNum = 1;

	pipe = std::make_shared<rs2::pipeline>();
	pipe2 = std::make_shared<rs2::pipeline>();
	rs2::config cfg; // Declare a new configuration

	auto depth_sensor = device.first<rs2::depth_sensor>();
	depth_sensor.set_option(RS2_OPTION_VISUAL_PRESET, RS2_SR300_VISUAL_PRESET_SHORT_RANGE);

	cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_RGB8, 30);
	cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
	cfg.enable_record_to_file(filename);
	pipe->start(cfg); //File will be opened at this point
	device = pipe->get_active_profile().get_device();

	//	auto depth_sensor = device.first<rs2::depth_sensor>();
	//	depth_sensor.set_option(RS2_OPTION_VISUAL_PRESET, RS2_SR300_VISUAL_PRESET_SHORT_RANGE);	

	while (true) {
		rs2::frameset data;
		pipe->poll_for_frames(&data);

		if (!data) {
			cout << "nah" << endl;
		}
		else {
			rs2::frame depth = data.get_depth_frame().apply_filter(color_map);
			rs2::frame color = data.get_color_frame();

			//	cout << "Depth Frame: " << depth.get_frame_number() << endl;
			//	cout << "Color Frame: " << color.get_frame_number() << endl;

				// Query depth frame size (width and height)
			const int wD = depth.as<rs2::video_frame>().get_width();
			const int hD = depth.as<rs2::video_frame>().get_height();

			// Query depth frame size (width and height)
			const int wC = color.as<rs2::video_frame>().get_width();
			const int hC = color.as<rs2::video_frame>().get_height();

			// Create OpenCV matrix of size (w,h) from the colorized depth data
			Mat depthImage(Size(wD, hD), CV_8UC3, (void*)depth.get_data(), Mat::AUTO_STEP);

			// Create OpenCV matrix of size (w,h) from the colorized depth data
			Mat colorImage(Size(wC, hC), CV_8UC3, (void*)color.get_data(), Mat::AUTO_STEP);

			// Update the window with new data
			imshow(window_name1, depthImage);

			// Update the window with new data
			imshow(window_name2, colorImage);
			waitKey(20);
		}

		auto t2 = Clock::now();
		seconds = std::chrono::duration_cast<std::chrono::seconds>(t2 - t1).count();

		if (seconds != oldSeconds) {
			if (seconds % 60 == 0)
				minutes++;
			cout << seconds / 60 << " minutes or " << endl;
			cout << seconds << " seconds have elapsed " << endl;
			oldSeconds = seconds;

			//update filename
			if (minutes % desiredVideoLengths == 0 && minutes != 0) {
				minutes = 0;
				fileNum++;
				filename = to_string(fileNum) + ".bag";

				pipe->stop();
				cfg.enable_record_to_file(filename);
				pipe->start(cfg); //File will be opened at this point
				device = pipe->get_active_profile().get_device();
				cout << "----------------------------------------------------------" << endl;
				cout << "----------------------------------------------------------" << endl;
			}
		}

	}

	pipe->stop();
	return EXIT_SUCCESS;
}
```

[image]

I still got `"nah"` as an output, but the color and depth frames were coming in. Made no modification to your code, it works on Windows too. But I realized the `pipe->poll_for_frames(&data);` doesn't really work as intended, changed it to `pipe = pipe->wait_for_frames();` on `Line: 71`.

[image]

philipjames44, 2018-11-22:
@HippoEug 
I'm aware it takes 1-3 seconds to initialize. This is not by design but I have no other way of doing instant initialization. I believe it may have to do with how I set the depth sensor to short range. I'm glad to see that it works on windows though! Unfortunately, I have no idea why it doesn't work on Ubuntu though :/ I'll try out the change you suggested, but I doubt that it will fix the root issue.

philipjames44, 2018-11-22:
@HippoEug 
When I try your recommended change I get  `error: no viable overloaded '='` in regards to the pipe =

HippoEug, 2018-11-22:
Just tried out your second example. After modifications, I managed to stream and record to .bag simultaneously. 

Here is my modified code:
```
#include "opencv2/opencv.hpp"
#include <librealsense2/rs.hpp>
#include <sstream>
#include <iostream>
#include <iomanip>
#include <chrono>
#include <string>
using namespace std;
using namespace cv;

int main() {
	using namespace cv;

	const auto window_name1 = "Depth Image";
	const auto window_name2 = "Color Image";

	namedWindow(window_name1, WINDOW_AUTOSIZE);
	namedWindow(window_name2, WINDOW_AUTOSIZE);

	// Declare frameset and frames which will hold the data from the camera
	rs2::frame depth;

	// Declare depth colorizer for pretty visualization of depth data
	rs2::colorizer color_map;
	color_map.set_option(RS2_OPTION_COLOR_SCHEME, 3.f); //Black to White

	// Create a shared pointer to a pipeline
	auto pipe = std::make_shared<rs2::pipeline>();

	rs2::config cfg; // Declare a new configuration
	cfg.enable_record_to_file("kms_x.bag");

	pipe->start(cfg); //File will be opened at this point

	int i = 0;

	while (i < 1000) {
		std::cout << i << std::endl;
		i++;

		rs2::frameset data;
		pipe->poll_for_frames(&data);

		if (!data) {
			cout << "nah" << endl;
		}
		else {
			rs2::frame depth = data.get_depth_frame().apply_filter(color_map);
			rs2::frame color = data.get_color_frame();

			// Query depth frame size (width and height)
			const int wD = depth.as<rs2::video_frame>().get_width();
			const int hD = depth.as<rs2::video_frame>().get_height();

			// Query depth frame size (width and height)
			const int wC = color.as<rs2::video_frame>().get_width();
			const int hC = color.as<rs2::video_frame>().get_height();

			// Create OpenCV matrix of size (w,h) from the colorized depth data
			Mat depthImage(Size(wD, hD), CV_8UC3, (void*)depth.get_data(), Mat::AUTO_STEP);

			// Create OpenCV matrix of size (w,h) from the colorized depth data
			Mat colorImage(Size(wC, hC), CV_8UC3, (void*)color.get_data(), Mat::AUTO_STEP);

			// Update the window with new data
			imshow(window_name1, depthImage);

			// Update the window with new data
			imshow(window_name2, colorImage);
			waitKey(20);
		}
	}

	pipe->stop();
	return EXIT_SUCCESS;
}
```
[image]

[image]

Able to open the `kms_x.bag` on Realsense Viewer afterwards. Please note I am using the D435, so there may be some differences

HippoEug, 2018-11-23:
Intel recommends using CMake, but I have no prior experience to using it so I had to find ways to manually add the files into Visual Studio.

For Realsense, Intel already created a `.props` file with all the configuration done, so upon adding the .props file in Property Manager I could use most, if not all Realsense files.

[image]

As you can see, there are 3 `.props` file for the 3 most commonly used libraries for this project. There used to be only `intel.realsense.props`, they only added the other 2 `.props` very very recently. 

Just adding these should work, most `#include` directories should compile with no warnings.

Here is my own [installation guide](https://github.com/HippoEug/WetlingRealsenseES/blob/master/README.md), with manually added OpenGL/GLFW and OpenCV files I downloaded from websites before adding them to Visual Studio via Project -> Properties -> etc etc. But now this installation guide is outdated, since just adding the `.props` file included in the SDK download should instantly work, much easier.

Hope this is clear enough!

philipjames44, 2018-11-23:
@HippoEug 
Absolutely fantastic! I will try out everything tomorrow and let you know how it goes!

philipjames44, 2018-11-23:
@HippoEug @dorodnic 

So on my Mac, your earlier suggestion of changing line 71 worked well on Mac osx. However, when plugging my camera into my laptop for the first time I will always get one of the following scenarios: my camera color stream is warped and unusable (green lines everywhere, or I get the following `libc++abi.dylib: terminating with uncaught exception of type rs2::error: Frame didn't arrived within 5000
Abort trap: 6`
The program runs fine if I run it again and produces good outputs, however upon unplugging and plugging in my camera again, I will get one of the same above issues consistently on first run of the program without fail.

I have yet to test this change on ubuntu to see if I get any better results, so I will let you know on that front later. But I have a feeling that the frames not working upon the first time I plug the camera may be related to the root issue on my ubuntu system. Thoughts?

philipjames44, 2018-11-24:
Interesting. On ubuntu I only get this error :`terminate called after throwing an instance of 'rs2::error'
  what():  Frame didn't arrived within 5000
Aborted (core dumped)
`

The program will do this each time

philipjames44, 2018-11-24:
Update: It has something to do with the setting of the depth laser strength. Not sure how to fix it yet though.

philipjames44, 2018-11-24:
FIX: For posterity's sake, I got the original code to work by changing the way the code retrieved the device profile and hence how it set the sensor strength. Here is what I used as reference: https://github.com/IntelRealSense/librealsense/blob/master/examples/sensor-control/api_how_to.h#22

and here is the code:
```
#include "opencv2/opencv.hpp"
#include <librealsense2/rs.hpp>
#include <sstream>
#include <iostream>
#include <iomanip>
#include <chrono>
#include <string>
using namespace std;
using namespace cv;

typedef std::chrono::high_resolution_clock Clock;

int main(){
	using namespace cv;
	
	const auto window_name1 = "Depth Image";
 	const auto window_name2 = "Color Image";
  	
	namedWindow(window_name1, WINDOW_AUTOSIZE);
	namedWindow(window_name2, WINDOW_AUTOSIZE);
	
	// Declare frameset and frames which will hold the data from the camera
	rs2::frameset frames;
	rs2::frame depth;
	
	// Declare depth colorizer for pretty visualization of depth data
	rs2::colorizer color_map;	
	color_map.set_option(RS2_OPTION_COLOR_SCHEME, 3.f); //Black to White
	
	// Create a shared pointer to a pipeline
	auto pipe = std::make_shared<rs2::pipeline>();
//	pipe->start();
	
	// Initialize a shared pointer to a device with the current device on the pipeline
	rs2::context ctx;
	rs2::device_list devices = ctx.query_devices();
	
	rs2::device_hub device_hub(ctx);

	rs2::device selected_device;
	selected_device = device_hub.wait_for_device();	
	cout << "Selected Device: " << selected_device.get_info(RS2_CAMERA_INFO_NAME) << endl; 

	auto depth_sensor = selected_device.first<rs2::depth_sensor>();
	depth_sensor.set_option(RS2_OPTION_VISUAL_PRESET, RS2_SR300_VISUAL_PRESET_SHORT_RANGE);

 //   	rs2::device device = pipe->get_active_profile().get_device();

//	pipe->stop(); // Stop the pipeline with the default configuration
	
	//clock for timer
	auto t1 = Clock::now();
	int oldSeconds = 0;
	int seconds = 0;
	int minutes = 0;
	int desiredVideoLengths = 1; //minute lengths of video sequences 
	string filename = "1.bag";
	int fileNum = 1;

	pipe = std::make_shared<rs2::pipeline>();
   	rs2::config cfg; // Declare a new configuration

//	auto depth_sensor = device.first<rs2::depth_sensor>();
//        depth_sensor.set_option(RS2_OPTION_VISUAL_PRESET, RS2_SR300_VISUAL_PRESET_SHORT_RANGE);
 
	
	cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_RGB8, 30);
	cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
	cfg.enable_record_to_file(filename);
 	pipe->start(cfg); //File will be opened at this point
//	device = pipe->get_active_profile().get_device();	

//	auto depth_sensor = device.first<rs2::depth_sensor>();
//	depth_sensor.set_option(RS2_OPTION_VISUAL_PRESET, RS2_SR300_VISUAL_PRESET_SHORT_RANGE);	

	while(true){
		rs2::frameset data;
		pipe->poll_for_frames(&data);
//		waitKey(4000);

		if(!data){
//			cout << "nah" << endl;
		}
		else{
	        	rs2::frame depth = data.get_depth_frame().apply_filter(color_map);
          		rs2::frame color = data.get_color_frame();
  
          	//	cout << "Depth Frame: " << depth.get_frame_number() << endl;
          	//	cout << "Color Frame: " << color.get_frame_number() << endl;
  
          		// Query depth frame size (width and height)
          		const int wD = depth.as<rs2::video_frame>().get_width();
          		const int hD = depth.as<rs2::video_frame>().get_height();
  
          		// Query depth frame size (width and height)
         		const int wC = color.as<rs2::video_frame>().get_width();
         		const int hC = color.as<rs2::video_frame>().get_height();
  
         	 	// Create OpenCV matrix of size (w,h) from the colorized depth data
         		Mat depthImage(Size(wD, hD), CV_8UC3, (void*)depth.get_data(), Mat::AUTO_STEP);
  
      	 	   // Create OpenCV matrix of size (w,h) from the colorized depth data
        		Mat colorImage(Size(wC, hC), CV_8UC3, (void*)color.get_data(), Mat::AUTO_STEP);
  
       		   // Update the window with new data
        		imshow(window_name1, depthImage);
  
        	  // Update the window with new data
	        	imshow(window_name2, colorImage);
			waitKey(20);
		}

		auto t2 = Clock::now();
		seconds = std::chrono::duration_cast<std::chrono::seconds>(t2 - t1).count();
						
		if(seconds != oldSeconds){
			if(seconds % 60 == 0)
				minutes++;
			cout << seconds / 60 << " minutes or " << endl;		
			cout << seconds << " seconds have elapsed " << endl;
			oldSeconds = seconds;
			
			//update filename
			if(minutes % desiredVideoLengths == 0 && minutes != 0){
				minutes = 0;
				fileNum++;
				filename = to_string(fileNum) + ".bag";
				
				pipe->stop();	
				cfg.enable_record_to_file(filename);	
				pipe->start(cfg); //File will be opened at this point
//		         	device = pipe->get_active_profile().get_device();
				cout << "----------------------------------------------------------" << endl;
                                cout << "----------------------------------------------------------" << endl;
			}
		}		
				
	}	
	
	pipe->stop();
 	return EXIT_SUCCESS;
}

```

I believe the root issue stems from the pipe being called to start and stop quickly to get the device profile, but I don't know why this cause so many issues. @HippoEug @dorodnic

closed 2018-11-24 by philipjames44