realsenseai/librealsense #2763: Variable shadowing

SamuelLordBlack, 2018-11-22:
A minor request, but it would be nice if the RealSense header files would compile without warnings. We enable -Wshadow in our projects, and librealsense has several instances of variable shadowing in its header files. Our workaround has been to add **#pragma GCC system_header** to rs.hpp, but this is an additional step we have to remember each time we pull in a new drop, and it can mask actual issues.

For reference, 2.16.5 has the following instances of variable shadowing:

rs_frame.hpp:307:37: declaration of ‘frame_ref’ shadows a member of ‘rs2::frame’
rs_frame.hpp:659:28: declaration of ‘e’ shadows a previous local
rs_frame.hpp:856:28: declaration of ‘e’ shadows a previous local
rs_frame.hpp:871:48: declaration of ‘frame’ shadows a member of ‘rs2::frameset’
rs_frame.hpp:887:18: declaration of ‘frame’ shadows a member of ‘rs2::frameset’
rs_frame.hpp:931:55: declaration of ‘frame’ shadows a member of ‘rs2::frameset’
rs_sensor.hpp:17:9: declaration of ‘notification’ shadows a member of ‘rs2::notification’
rs_record_playback.hpp:214:9: declaration of ‘device’ shadows a member of ‘rs2::recorder’
rs_pipeline.hpp:323:52: declaration of ‘config’ shadows a member of ‘rs2::config’

dorodnic, 2018-11-22:
Hi @SamuelLordBlack 
So basically variables with names already taken, right?
Sure, we will look into it and once it is done add `-Wshadow` to Travis

SamuelLordBlack, 2018-11-22:
Yes, this warning is generated if a variable already exists in that scope, so the new variable masks the previous one.

dorodnic, 2018-11-22:
Got it

dorodnic, 2018-11-29:
@SamuelLordBlack - Should be resolved in [2.17.0](https://github.com/IntelRealSense/librealsense/releases/tag/v2.17.0)

closed 2018-12-10 by dorodnic