realsenseai/librealsense #2764: Can not build SDK2.0 from source on MacOS Mojave

darrenjiayinxie, 2018-11-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |   2.16.5                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

Use Xcode 10 to build, however there appears error:
ld: library not found for -lusb-1.0
clang: error: linker command failed with exit code 1 (use -v to see invocation)
<Describe your issue / question / feature request / etc..>

HippoEug, 2018-11-23:
Hi @darrenjiayinxie , AFAIK there is not a complete version that supports MacOS yet.

From [Intel's documentation updated May 2018](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Intel-RealSense-SDK2-Github-Guide.pdf), it says:

[image]

How did you compile your code with? What camera are you using?

darrenjiayinxie, 2018-11-23:
I follow the tutorial
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md
I am using D415

sleepy-maker, 2018-11-27:
We can use pre-build version until the team solve the issue.

```
(we can install the latest version)
$ brew install librealsense
```
We can use viewer etc, but unfortunately not fully functional.

darrenjiayinxie, 2018-11-27:
Thanks! Now I switch to the ubuntu 16.04 and it works well. Hope the Mac
version can work soon.

thomasliu623, 2018-11-28:
@dorodnic Hi, I'm running into the same `ld: library not found for -lusb-1.0` bug when trying to run on XCode, and I attempted to fall-back to 2.15.0 based on your previous comment, but am still seeing the same issue. What were your precise steps of falling back? Did you just clone the `2.15.0` release branch and then follow the steps outlined here: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md?

GitwellAnyohub, 2018-11-30:
bump for @thomasliu623 question

bryanrtboy, 2018-12-01:
I am encountering this same error: ld: library not found for -lusb-1.0 when trying to compile with Xcode.
Taking @sleepy-maker advice and Installing with brew allowed me to open the examples, but I get black screens with no indication the D415 is recognized or active.

bryanrtboy, 2018-12-02:
I have this same error when trying to build with Xcode: ld: library not found for -lusb-1.0 
@sleepy-maker suggestion works for testing the camera.   Starting by running /usr/local/bin/rs-sensor-control, I can see the device and then start to use the other examples. I am trying to run in Unity, and found a .dll but could really use a fix!

sharner, 2018-12-10:
I have the same issue building from source on macOS 10.14.2
```
ld: library not found for -lusb-1.0
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```
Has anyone had any luck with this?  I can verify that the library is here:
```
/usr/local/Cellar/libusb/1.0.22/lib/libusb-1.0.dylib
```
The library path has /usr/local/Cellar/libusb/1.0.22/lib/

I can run rs-sensor-control just fine.  Has anyone solved this?

GitwellAnyohub, 2018-12-10:
@sharner when you use make you may have to specify the Library where libusb is stored.

```
make LIBRARY_PATH=/usr/local/lib
```

I had trouble building from source on mac and here is what I did to get it to work:

https://github.com/GitwellAnyohub/Real_Sensible_Moseq/blob/master/bash_and_install/install_realsense_sdk_mac_os.sh

sharner, 2018-12-10:
@GitwellAnyohub thanks, your script worked for me.  I was never able to get Xcode working, as in the instructions.

GitwellAnyohub, 2018-12-10:
Glad it worked!

RealSense-Customer-Engineering, 2018-12-17:
[Realsense Customer Engineering Team Comment]
Submit issue as solved.

dorodnic, 2019-01-12:
We need to add this to documentation or find a way to resolved via Cmake

peymon, 2019-01-23:
In [another thread](https://github.com/IntelRealSense/librealsense/issues/3119) @kvbogdanov fixed this by running the following before opening xcode:
`/bin/launchctl setenv LIBRARY_PATH /usr/local/lib`

Can confirm it worked.

jackjansen, 2019-01-23:
And as another workaround: after opening the project in Xcode, add `/usr/local/lib` to the library search path in the per-project settings. This has to be done for at least the realsense2 dylib. And for some reason adding that library at the toplevel project settings does not work (which may actually be the reason behind this not working in the first place).

closed 2024-03-29 by MartyG-RealSense