realsenseai/librealsense #2767: Librealsense cannot Cmake for building files

ljc19800331, 2018-11-23:
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { librealsense2 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hello,

When I want to CMAKE in the librealsense source folder, I got the following error:
-- The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Info: REALSENSE_VERSION_STRING=2.16.5
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Setting Unix configurations
-- Checking internet connection...
-- Internet connection identified, enabling BUILD_WITH_TM2
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for one of the modules 'libusb-1.0'
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Checking for module 'gtk+-3.0'
--   Found gtk+-3.0, version 3.18.9
-- Building with TM2
-- ----------------------------------------------------------------------------
-- Product versions:
-- - HOST 0.19.3.1175 (Default from versions.cmake)
-- - Remote FW 0.0.18.3452 (Default from versions.cmake)
-- - Remote CENTRAL APP 2.0.19.269 (Default from versions.cmake)
-- - Remote CENTRAL BL 1.0.1.112 (Default from versions.cmake)
-- --------------------------------------------------------------------------------------------------------------------------------------------------------------
-- Downloading FW 0.0.18.3452 from 'http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/target/0.0.18.3452/target-0.0.18.3452.mvcmd'
^[[A^[[A^[[B^[[B-- Downloading FW 0.0.18.3452 completed - status 28;"Timeout was reached"
CMake Error at third-party/libtm/resources/CMakeLists.txt:92 (message):
  Download error
  http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/target/0.0.18.3452/target-0.0.18.3452.mvcmd
-- Configuring incomplete, errors occurred!
See also "/home/mgs/librealsense/build/CMakeFiles/CMakeOutput.log"

I checked the Internet setting and everything is fine. Thanks.

closed 2018-11-23 by ljc19800331

ljc19800331, 2019-01-18:
hey @Yaosirrr @. I followed the guidance #2767 issue to solve this problem. Hope it helps.

xiao-keeplearning, 2019-11-18:
> hey @Yaosirrr @. I followed the guidance #2767 issue to solve this problem. Hope it helps.

What should I do? #2767 is just this issue.