realsenseai/librealsense #2771: Unable to stream depth at 6 or 60 fps

VasuAgrawal, 2018-11-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 (4x D435) | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |   Linux for Tegra (based on Ubuntu 18)  | 
| Kernel Version (Linux Only)    |  4.9.108-tegra                                         | 
| Platform                                 | Jetson AGX Xavier Devkit  |
| SDK Version                            |  2.16.3                         | 
| Language                            |  C++                          | 
| Segment			|  Robot                          | 

### Issue Description

I have 4 RealSense D435 cameras that I'm trying to run at the same time, streaming both depth and RGB information. I have the 4 cameras plugged into a [USB hub](https://www.newegg.com/Product/Product.aspx?Item=9SIA6ZP81W9102&Description=startech%20hb31c4ab&cm_re=startech_hb31c4ab-_-17-707-457-_-Product), which is then plugged in to one of the USB C ports on the Xavier.

I was trying to determine the maximum frame rates and resolutions I could run the 4 cameras at, and started by testing a single camera. It seems that there's a bug _somewhere_ which is preventing the cameras from generating the correct depth information when the RGB camera is activated, at both 6 and 60 fps. At 15 and 30 FPS, there doesn't seem to be any problems.

This bug makes a lot more sense with some photos / videos, so I took a [screen recording of the procedure](https://drive.google.com/file/d/1_jaW4Z6sGrJ8kInnoEzWjKLS607zMMIx/view?usp=sharing) of determining which configurations of depth and RGB work and which don't (with a single camera). Here are the results:

  | 6 fps | 15 fps | 30 fps | 60 fps
-- | -- | -- | -- | --
424 x 240 | NO | YES | YES | NO
640 x 360 | NO | YES | YES | NO
640 x 480 | NO | YES | YES | NO
848 x 480 | NO | YES | YES | NO
1280 x 720 | YES | YES | YES | N/A

While streaming 4 cameras at 60 FPS is impossible given USB 3 bandwidth limits, I do have a use case for streaming the cameras at high resolutions at 6 FPS. Does anyone have any idea what could be causing the failures at 6 and 60 FPS?

To see whether I could stream 4 cameras at 6 FPS at 720p, I repeated the same experiments as above, except with 4 cameras (testing only the working configurations). Unfortunately, 4x 720p seems to be just slightly too much bandwidth, and the 848 x 480p configuration doesn't work (per above).

  | 6 fps | 15 fps | 30 fps | 60 fps
-- | -- | -- | -- | --
424 x 240 |   | YES | YES |  
640 x 360 |   | YES | NO |  
640 x 480 |   | YES |   |  
848 x 480 |   | NO (barely) |   |  
1280 x 720 | NO (barely) |   |   |  

You can see the [procedure here](https://drive.google.com/file/d/1bAncbY-WEghcGRy9WX0FeuVBY3_U-H1Q/view?usp=sharing).

Perhaps this is also the place to bundle in a related feature request - would it be possible to add additional framerates? Specifically, I would very much like to run the 4 camera 640 x 360 configuration at a functional frame rate between 15 and 30, e.g. 25.

MartyG-RealSense, 2018-11-26:
I do not have an answer for why you are having problems at 6 and 60 FPS specifically (one of the Intel guys may be able to help you with that).  I did want to fill in some background information that may be helpful to you though.

Intel's white paper on multiple cameras has a chart for approximate bandwidth use on a 4-port USB hub without hardware sync.  It shows that 90 FPS may be possible with a 4-camera setup under certain settings.  In the paper, Intel tested with a 4-port AmazonBasics hub.

[image]

The specification of the computing hardware can have a bearing on performance too.  Intel's September webinar session on multiple cameras recommended that the computing device's specification be increased as the number of cameras added is increased (for example, an i7 processor may be appropriate for a 4-camera setup).  Jetson Xavier (the board you use) is a powerful board though, so that may not be the cause of your processing lag. 

Intel said in the webinar that at a future date, x3 compression of data will be enabled.   This will allow more channels and more cameras to be supported.  A new 'Modified Huffman' lossless compression will enable the streaming of higher resolution and FPS on USB2, and more channels in multi-camera configurations.

 Regarding your request for a 25 FPS framerate to be supported, this is a request that other users have made too, as it is seen as being ideal for using the camera with movie-related work.  I do not have any information on whether there are currently plans to implement support for this FPS though.  For now, 25 FPS is supported only in the Y16 IR mode, which is an unrectified channel meant for use in camera calibration rather than everyday use in applications.  This mode also only supports a couple of resolutions.

https://communities.intel.com/message/557408#557408

RealSense-Customer-Engineering, 2018-12-12:
[Realsense Customer Engineering Team Comment]
Hi @VasuAgrawal,

It looks Nvidia AGX Xavier platform used. So far, I don't have that to have a further check. Per my previous understanding on Jetson TX2, is there a script like "jetson_clocks.sh" to disable auto-frequency adjustment and see how it works?

RealSense-Customer-Engineering, 2019-01-11:
[Realsense Customer Engineering Team Comment]
Hi @VasuAgrawal,
Were you able to test the 4 camera for 640x360 at 30 fps using a powered USB hub?

closed 2019-02-02 by RealSenseCustomerSupport