realsenseai/librealsense #2772: depth image problem

SunshineDou, 2018-11-26:
hello 
I have a Realsense D435, I want to make datasets using it which can run in ElasticFusion,  I record the video and get bag files, but the depth image in the bag files is only 16KB or so, which is so small, and the depth image is very dark, but when I only have a snapshot,  the depth image has 77KB or so. what's the problem? what I need is the "fixed, black to white" depth image sequences, how can I get it?
Thank you very much

HippoEug, 2018-11-26:
Hi @SunshineDou, how are you performing the `.bag` recording? Through the Realsense Viewer, or `rs-convert-playback.cpp`? Can you elaborate on the depth image in the bag files? You played the `.bag` file to see the depth image, or did you use `rs-convert.cpp` to get the depth image which is 77KB?

SunshineDou, 2018-11-26:
Thank you very much for your replying. I run the rs-convert.cpp to convert the bag files into rgb and depth images, in which the depth image is very dark and only has 29KB or 18KB or so, I get the 77KB image by run the realsense-viewer, and set the depth visual preset to "fixed" , the color scheme to "Black to white" and then save a snapshot.  what can I do to get a sequences of normal "fixed, Black to white" depth image?

SunshineDou, 2018-11-26:
Thanks again for your instant replying. I record a video of about 18s using rs-playback-record.cpp in librealsense, and then I convert the recorded a.bag into depth and color image using rs-convert.cpp. I use the following convert command:
./rs-convert -p /home/cindy/librealsense-master/build/tools/convert/a/ -i a.bag 
the color images in a.bag are all about 300KB or so, but the depth images are 20KB or so. what's the problem?

dorodnic, 2018-11-26:
Hi @SunshineDou 
Can you see valid depth in the RealSense Viewer?

SunshineDou, 2018-11-26:
yeah, I can. actually I can get normal depth image about 170KB or so, but it is in "jet" color scheme, as I want to get " fixed, Black to White" depth image. How can I get it ? some tine ago, I can get white to black depth images, but it has only 20KB.

RealSense-Customer-Engineering, 2018-11-30:
[Realsense Customer Engineering Team Comment]
@Sunshinedou What's the FW version and librealsense version you're using when get this problem?

RealSense-Customer-Engineering, 2018-11-30:
[Realsense Customer Engineering Team Comment]
@sunshinedou  Do you use the default resolution of rs-playback-record.cpp?  I just tried the default resolution using D435 with FW 5.10.6 and librealsense 2.16.3, and I can get the depth raw file about 1800KB by "rs-covert.exe -i a.bag -r "

SunshineDou, 2018-12-04:
@RealSense-Customer-Engineering Thank you very much for your replying
I use the resolution of 640x80 with librealsense2, I don't know the firmware version, how to check it

HippoEug, 2018-12-04:
@SunshineDou One way is to use Realsense Viewer, click on info

[image]

RealSense-Customer-Engineering, 2018-12-04:
[Realsense Customer Engineering Team Comment]
@sunshinedou So I understood you're using 640x480. If you're using USB3.0 connection, the default resolution should be 1280x720. Could you please check if you're connecting with USB2.0 port or cable?  I just tried USB2.0 here with librealsense v2.16.3 and got crash. So if you can use USB3.0, please try USB3.0 first. Thanks!

HippoEug, 2018-12-04:
@RealSense-Customer-Engineering did that change recently in the newer SDK versions? I remember previouly by default, my RGB's resolution is 1280x720 with depth being 640x480

RealSense-Customer-Engineering, 2018-12-04:
[Realsense Customer Engineering Team Comment]
@HippoEug Here I used the default build of rs-playback-record and rs-covert, not viewer either.

HippoEug, 2018-12-04:
`rs-capture.cpp` !

SunshineDou, 2018-12-06:
@HippoEug @RealSense-Customer-Engineering 
Thank you very much for answering. my firmware is as follows:
[image]
my usb type is 2.1, default resolution is 640x480. I think I should try usb3.0 later. Thank you very much

closed 2018-12-13 by SunshineDou