realsenseai/librealsense #2774: Add Attribute Table to appear: System.IO.FileNotFoundException (Exception from HRESULT: 0x8007007E)”

laolaolulu, 2018-11-26:
Add Attribute Table to appear: System.IO.FileNotFoundException (Exception from HRESULT: 0x8007007E)”  

[image]
[image]

laolaolulu, 2018-11-26:
windows 10 UWP c++
[image]

MartyG-RealSense, 2018-11-26:
An Intel employee over at the Intel Support website gave the advice below about the 0x8007007E error:

"One way to handle this is to open in Visual Studio one of the SDK sample projects that most closely matches what you are trying to do and copy exactly the Project Properties to your own project.  Pay special attention to the include dirs, linked libraries, etc".

laolaolulu, 2018-11-27:
I have no problem creating Windows console applications, and I have a problem creating Windows Universal (UWP).

closed 2018-12-03 by laolaolulu