realsenseai/librealsense #2779: DCM

SunshineDou, 2018-11-27:
hello
I want to install dcm for realsense D435, but I cannot find it, I can only find dcm for F , SR and R series. can you help me? where is the dcm for realsense D435?
Thank you very much

HippoEug, 2018-11-27:
Pardon my ignorance, but what does DCM stand for?

SunshineDou, 2018-11-27:
DCM is "Intel realsense depth camera manager". you can find the DCM for F, R and SR series in the Downloads part in the link below 
https://software.intel.com/en-us/realsense-sdk-windows-eol

HippoEug, 2018-11-27:
In the link you provided, didn't it say the DCM aka SDK1.0 will be discontinued and would not be updated for D435? 

The current DCM replacement for D400 series would probably be SDK 2.0, the one that is installed with Intel Realsense Viewer etc

SunshineDou, 2018-11-27:
oh, I missed it, Thank you. as I want to run a project but it's lack of "pxcsensemanager.h" file, which maybe in DCM, so I want to install it

RealSense-Customer-Engineering, 2018-11-28:
[Realsense Customer Engineering Team Comment]
Hi @SunshineDou,

"pxcsensemanager.h" file comes from the previous RSSDK which is not maintained. Since you use D435, you need to use SDK 2.0 instead. And, DCM is not used anymore for D400 series camera.

RealSense-Customer-Engineering, 2018-12-05:
[Realsense Customer Engineering Team Comment]
Hi @SunshineDou,

The librealsense SDK 2.0 for D435 is totally different with the previous RSSDK.

SunshineDou, 2018-12-06:
@RealSense-Customer-Engineering 
Ok, I have got it, Thank you for your replying

closed 2018-12-11 by RealSense-Customer-Engineering