realsenseai/librealsense #2781: Announcement: RealSense 400 Series data sheet updated to November 2018 edition (D435i added)

MartyG-RealSense, 2018-11-27:
Hi everyone,

The RealSense 400 Series data sheet has been updated to the November 2018 edition.  It includes details of the new D435i camera model with IMU.

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Intel-RealSense-D400-Series-Datasheet.pdf

HippoEug, 2018-11-28:
The link does not open for me. Hope it'll be fixed soon!

MartyG-RealSense, 2018-11-28:
I just tested the link.   It works in Chrome and Edge but not in Firefox.  It seems to work in Firefox if you launch the PDF from this page:

https://www.intel.co.uk/content/www/uk/en/support/articles/000026827/emerging-technologies/intel-realsense-technology.html

RealSense-Customer-Engineering, 2018-11-30:
[Realsense Customer Engineering Team Comment]
Hi

I'm also seeing the links work in Chrome.  If anyone is still having issues with the links please let us know what browsers are having issues.

Thanks

kzobov, 2018-12-01:
seems to me that cache cases this behavior. try cleaning it guys as Marty's initial link works in all browsers for me...

closed 2018-12-03 by RealSense-Customer-Engineering